brgl/libgpiod #1: gpioctld: system daemon controlling GPIOs

brgl, 2017-01-14:
- create gpioctld - a daemon controlling the GPIOs so that we can
  request and then keep the line reserved until we want it released
- create gpioctl - a program controlling the daemon

ZeekHuge, 2017-01-20:
Hi ! I would like to help here. Can you please guide me ?

brgl, 2017-01-20:
Hi @ZeekHuge,

I have already started working on that. I currently need to get it to the point where it has at least basic functionalities, then I'll push a development branch (features/daemon probably) on which collaborative work can start.

What I have now is epoll-based daemon listening on multiple sockets (client connections, gpio events, signals etc.). I'm using libconfig for configuration files and libprotobuf-c for communication between the client application and the daemon.

clemensg, 2017-01-25:
Hi @brgl,

I have a similar use case. I current use a hash table with a pointer to each gpiod_line I need to manage from a daemon.
The chips need to stay open during the runtime of the daemon for this to work (see my PR #11). I close them at the end. With this, I can open the chips and request the lines at daemon startup act and react to various events, get and set gpios by name during the daemon runtime and close them at daemon exit. Other userspace programs communicate with the daemon through SEQPACKET unix domain sockets.

Maybe PR #11 could be useful for your features/daemon branch too? Or do you have an alternative approach in mind to avoid opening and closing the chips for each action?

closed 2017-10-29 by brgl