brgl/libgpiod #3: RFE: Non-interactive operation options

macemoneta, 2017-01-15:
Thanks for these utilities; I switched our embedded home automation systems from sysfs pretty easily using them.

I did make two changes:

In gpioset.c, I removed the getchar() in function wait_for_enter. Since the command is called from a script, there was no need for the interaction. My script maintains the state of the GPIO pins used so that they can be restored across reboots, and there is no concurrent use. 

In gpiomon.c, I changed the initialization of do_run from true to false. My scripts use a poll loop, so I just need to see the state change:

   # Clear any pending state updates
   state="$(/usr/local/bin/gpiomon gpiochip0 230)"
   while [ "$state" != "" ]; do
      state="$(/usr/local/bin/gpiomon gpiochip0 230)"
   done

   # Wait for state change
   state="$(/usr/local/bin/gpiomon gpiochip0 230)"
   while [ "$state" == "" ]; do
      state="$(/usr/local/bin/gpiomon gpiochip0 230)"
   done

I think both of those changes might be generically useful, and options on the commands to change their operation would make it easier than patching.

brgl, 2017-01-15:
Hi macemoneta,

I'm planning on doing the first release next week. Right now the project is subject to major changes. I will also extend the tools significantly so that they're more useful and configurable. I'll consider your including the changes you proposed.

brgl, 2017-01-16:
> My script maintains the state of the GPIO pins used so that they can be restored across reboots, and there is no concurrent use.

BTW how do you maintain the state? The whole point of this new interface is to restore the default state when the file descriptor of the gpiochip is closed. The "state-leaks" were a one of the biggest issues with the previous sysfs-based API.

macemoneta, 2017-01-16:
From the script's perspective, that's trivial. Since the script is changing the state, it simply writes the current state to a file.
```
echo 0 > "${stateFile}-${reqPin}"
/usr/local/bin/gpioset gpiochip0 $reqPin 0
```
At the points where I need to restore the state, I can just:
`/usr/local/bin/gpioset gpiochip0 $reqPin $(/bin/cat "${statefile}-${reqPin}")`
I did the same with the sysfs interface, (including direction and pull up/down) so there was no change in that regard. That function really shouldn't be part of the tools, in my opinion.

brgl, 2017-01-16:
My point was: when gpioset closes the gpiochip, the GPIO line state returns to the kernel-defined default. This is why I'm planning on creating a daemon that would keep the device files open if the user needs to keep certain lines asserted.

macemoneta, 2017-01-16:
Oh, when I removed the getchar() in function wait_for_enter of gpioset, the state of the GPIO pin remained unchanged at exit. I did notice if I later tried to use gpioget it would revert, so I just kept my code and didn't use gpioget.

brgl, 2017-01-16:
Hmm this is strange, it doesn't behave like this on my setup. What kernel version are you using? Any chance of posting an strace output from you gpioset invocation?

macemoneta, 2017-01-16:
I'm using 4.9.3 (Fedora 25 kernel) soon to be 4.9.4 on a couple of Linksprite PCDuino3 Nano Lite (Allwinner A20).

Strace attached.

[gpioset-strace.txt](https://github.com/brgl/libgpiod/files/708524/gpioset-strace.txt)

brgl, 2017-01-16:
I don't know this board, but if I understand correctly: you set the value of GPIO line 231 to 0 and it remains at 0 when gpioset exits? If you ran 'gpioget gpiochip 0 231' before gpioset would it give you 1? Can you also run 'gpioinfo gpiochip0' after gpioset to make sure the line is no longer exported?

macemoneta, 2017-01-16:
> you set the value of GPIO line 231 to 0 and it remains at 0 when gpioset exits?

Yes

> If you ran 'gpioget gpiochip 0 231' before gpioset would it give you 1? 

Yes:
```
gpioget gpiochip0 231
1

```

> Can you also run 'gpioinfo gpiochip0' after gpioset to make sure the line is no longer exported?

gpioset gpiochip0 231 0
gpioinfo gpiochip0 | grep 231
        line 231: unnamed unused output active-high

brgl, 2017-01-16:
The new framework should clean-up the resources on close/exit/crash. I need to consult the author of the kernel part on this one.

brgl, 2017-01-16:
I now see that the pins I was testing on are pulled down. Is the pin you're driving floating by any chance? That would explain the behavior.

brgl, 2017-01-16:
Anyway - while waiting for a single state change in gpiomon is a good idea and I will surely incorporate it, your gpioset use case is not how it should work. I will extend gpioset for sure with things like waiting a given number of seconds before returning etc. but the idea behind the new framework is to have the kernel clean up as soon as we close the file descriptor.

macemoneta, 2017-01-16:
The output pins are floating, though they are attached to a relay board that might have an external pull-down. I use a pull-down on the input pin that I have configured.

For gpioset, how about an optional configurable delay (in ms) and exit. That could be used as a toggle in your case, and would allow my use case without requiring a locally maintained patch.

macemoneta, 2017-01-16:
Also, for gpioset, could you make the wait optionally indefinite until killed (e.g., '0')? That way a script could background it and kill the process when the state should be reverted.

brgl, 2017-01-16:
> The output pins are floating, though they are attached to a relay board that might have an external pull-down. I use a pull-down on the input pin that I have configured.

Just to verify - could you use the old sysfs framework, export the same GPIO, set its value to 1 and then unexport it? This would be equivalent to what you're doing here. If the behavior is consistent then the library is good.

> Sorry, just noted your updated comment. Gpioset with a delay would be perfect. Milliseconds would be better.

Will do, but I'm not sure if I'll have time for it before releasing 0.1 which I want to do soon for people to test the library.

> Also, for gpioset, could you make the wait optionally indefinite until killed (e.g., '0')? That way a script could background it and kill the process when the state should be reverted.

You can always kill -9 the current version once it stops after detaching from the terminal. Eventually we'll have a daemon that will keep the state in the background.

macemoneta, 2017-01-16:
There's no urgency on changes; the code I have works for my use case.

Unfortunately, as of the 4.9.2 kernel, Fedora no longer configures the GPIO sysfs. That's what prompted me to hunt down a replacement tool. However, once exported, I would never unexport the GPIO. It's just exported once per boot. Doing otherwise makes no sense to me for a control / input process.

Kill -9 is just ugly to me; a kill allowing the process to clean up nicely seems more production-worthy.

brgl, 2017-01-16:
> Unfortunately, as of the 4.9.2 kernel, Fedora no longer configures the GPIO sysfs.

This is a strange move given the recency of the new framework and lack of command-line tools.

> However, once exported, I would never unexport the GPIO. It's just exported once per boot. Doing otherwise makes no sense to me for a control / input process.

Except that with the character device, closing the device file is equivalent to unexporting the GPIO with sysfs. Keeping the file open guarantees that nobody else will fiddle with your lines.

macemoneta, 2017-01-16:
> This is a strange move given the recency of the new framework and lack of command-line tools.

That's the Fedora bleeding edge; the change forces people to wake up to deprecated functionality and look at the alternatives. Otherwise, everyone will just sit on their butts waiting for it to be removed from upstream. Then panic.  :)

> Keeping the file open guarantees that nobody else will fiddle with your lines.

In an embedded application, there is no one else. Ever. If the application is controlling a garage door or listening for a doorbell button press (for example), there is no one else that will be doing that.

If you need multiple control points sending commands to an embedded app that handles the GPIO (e.g., multiple people requesting a light be turned on/off concurrently), you serialize control with flock. The GPIO is physically connected to a single device, so I'm having a hard time imagining another way this concurrency could occur.

Like I said, the code, as is, works for me. If you have a different direction you want to take it I can maintain the local copies.

brgl, 2017-01-17:
> In an embedded application, there is no one else. Ever. If the application is controlling a garage door or listening for a doorbell button press (for example), there is no one else that will be doing that.

Sure, but this library is made with the goal of being useful for everybody - not only closed off, embedded systems. For example: I'm thinking about integrating it with libiio.

brgl, 2017-01-17:
Hi Mace,

please check out the new features of gpioset and gpiomon (refer to the help text and commit messages for info). Hope that helps.

macemoneta, 2017-01-17:
Thanks, will do!

macemoneta, 2017-01-17:
I updated to use the new versions, gpioset with '-m exit' and gpiomon with '--num-events=1 --rising-edge --silent' and it's all working great here. Thanks!

brgl, 2017-01-17:
Cool, let me close the issue then.

closed 2017-01-17 by brgl