brgl/libgpiod #22: Compiling on Ubuntu

chaoticmachinery, 2017-12-07:
When I try configuring this under Ubuntu I get the following:

checking linux/gpio.h usability... no
checking linux/gpio.h presence... no
checking for linux/gpio.h... no
configure: error: linux/gpio.h header not found (needed to build the library)

I have confirmed that the linux headers are installed.
/usr/src/linux-headers-4.10.0-40/include/linux/gpio.h

Is there something else I need to give configure?

Thank you,
CM

brgl, 2017-12-07:
This file should be in /usr/include/linux/gpio,h. On my debian 9 system it's provided by linux-libc-dev. Alternatively you can pass CFLAGS="-I/usr/src/linux-headers-4.10.0-40/include" to the configure script to let it know the location of the header.

timur-tabi, 2018-03-07:
Still doesn't work:
```
$ ./autogen.sh CFLAGS="-I/usr/src/linux-headers-`uname -r`/include"
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'autostuff'.
libtoolize: copying file 'autostuff/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:33: installing 'autostuff/compile'
configure.ac:21: installing 'autostuff/missing'
bindings/cxx/Makefile.am: installing 'autostuff/depcomp'
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... no
checking whether make supports nested variables... no
checking whether make supports nested variables... (cached) no
checking for style of include used by make... none
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... none
checking for ar... ar
checking the archiver (ar) interface... ar
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) none
checking for g++... no
checking for c++... no
checking for gpp... no
checking for aCC... no
checking for CC... no
checking for cxx... no
checking for cc++... no
checking for cl.exe... no
checking for FCC... no
checking for KCC... no
checking for RCC... no
checking for xlC_r... no
checking for xlC... no
checking whether we are using the GNU C++ compiler... no
checking whether g++ accepts -g... no
checking dependency style of g++... none
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
[... 49 lines cut ...]
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for ANSI C header files... (cached) yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking for ioctl... yes
checking for asprintf... yes
checking for scandir... yes
checking for alphasort... yes
checking for ppoll... yes
checking getopt.h usability... yes
checking getopt.h presence... yes
checking for getopt.h... yes
checking dirent.h usability... yes
checking dirent.h presence... yes
checking for dirent.h... yes
checking sys/poll.h usability... yes
checking sys/poll.h presence... yes
checking for sys/poll.h... yes
checking linux/gpio.h usability... no
checking linux/gpio.h presence... no
checking for linux/gpio.h... no
configure: error: linux/gpio.h header not found (needed to build the library)
```

InvalidExcepti0n, 2018-05-01:
try this

git clone https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git
cd libgpiod
mkdir -p include/linux
cp /usr/src/linux-headers-$(uname -r)/include/linux/compiler_types.h include/linux/.
./autogen.sh --enable-tools=yes --prefix=/usr/local CFLAGS="-I/usr/src/linux-headers-$(uname -r)/include/uapi -Iinclude"
make
sudo make install
sudo ldconfig

let me know if works for you as work for me on Mint 18.3

romainreignier, 2018-07-17:
@InvalidExcepti0n It works for me with warnings because I was using kernel headers. Thanks.

I was Ubuntu 16.04 with kernel 4.10 and the `compiler_types.h` file was not there so I have copied `compiler*` files.

gustavojcorrea, 2018-08-21:
@romainreignier  I also don't have the compiler_types.h file. Could you please provide a bit more information on your process to solve this issue? Thank you

romainreignier, 2018-08-22:
@gustavojcorrea I have followed @InvalidExcepti0n instructions except that I replaced this line:
```sh
cp /usr/src/linux-headers-$(uname -r)/include/linux/compiler_types.h include/linux/.
```
by this one
```
cp /usr/src/linux-headers-$(uname -r)/include/linux/compiler* include/linux/.
```

gustavojcorrea, 2018-08-23:
@romainreignier When I use `cp /usr/src/linux-headers-$(uname -r)/include/linux/compiler* include/linux/.` I get the following `cp: cannot stat '/usr/src/linux-headers-4.9.54-ti-r68/include/linux/compiler*': No such file or directory`

brgl, 2018-09-24:
@chaoticmachinery can I close this? I can't really help you, I don't use ubuntu. I think this question belongs on a ubuntu support forum.

closed 2018-09-25 by brgl