brgl/libgpiod #23: Compiling C++ bindings with buildroot fails: ERROR: unsafe header/library path used in cross-compilation: '-L/usr/lib'

jktjkt, 2018-02-23:
I patched my local buildroot to enable the C++ bindings:
```diff
diff --git a/package/libgpiod/libgpiod.mk b/package/libgpiod/libgpiod.mk
index 598adcc599..68b1166a5d 100644
--- a/package/libgpiod/libgpiod.mk
+++ b/package/libgpiod/libgpiod.mk
@@ -19,4 +19,8 @@ else
 LIBGPIOD_CONF_OPTS += --disable-tools
 endif
 
+ifeq ($(BR2_INSTALL_LIBSTDCPP),y)
+LIBGPIOD_CONF_OPTS += --enable-bindings-cxx
+endif
+
 $(eval $(autotools-package))
```
 and asked it to use my local checkout (commit a5ee3d7c4fe58dc838cdf3cfb5fabde1ddf2c910, current master) via `local.mk`:
```
builddir $ grep GPIO local.mk
LIBGPIOD_OVERRIDE_SRCDIR = /path/to/my/checkout/of/libgpiod
```
I also had to run `./autogen.sh` in the git checkout prior to letting Buildroot work there. That's expected.

The build fails:
```
jkt@kolibrik ~/work/prog/_build/br-cfb $ rm -rf build/libgpiod*
jkt@kolibrik ~/work/prog/_build/br-cfb $ make libgpiod
umask 0022 && make -C /home/jkt/work/cesnet/gerrit/github/buildroot/buildroot O=/home/jkt/work/prog/_build/br-cfb/. libgpiod
>>> libgpiod custom Syncing from source dir /home/jkt/work/cesnet/gerrit/github/brgl/libgpiod
rsync -au --chmod=u=rwX,go=rX --exclude .svn --exclude .git --exclude .hg --exclude .bzr --exclude CVS /home/jkt/work/cesnet/gerrit/github/brgl/libgpiod/ /home/jkt/work/prog/_build/br-cfb/build/libgpiod-custom
>>> libgpiod custom Configuring
(cd /home/jkt/work/prog/_build/br-cfb/build/libgpiod-custom/ && rm -rf config.cache && PATH="/home/jkt/work/prog/_build/br-cfb/host/bin:/home/jkt/work/prog/_build/br-cfb/host/sbin:/home/jkt/.local/bin:/home/jkt/bin/:/opt/qtc/bin:/opt/darktable/bin:/home/jkt/.local/bin:/home/jkt/bin/:/opt/qtc/bin:/opt/darktable/bin:/usr/local/bin:/usr/bin:/bin:/opt/bin:/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.3:/usr/lib/llvm/4/bin:/usr/games/bin" AR="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ar" AS="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-as" LD="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ld" NM="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-nm" CC="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc" GCC="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc" CPP="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-cpp" CXX="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-g++" FC="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gfortran" F77="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gfortran" RANLIB="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ranlib" READELF="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-readelf" STRIP="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-strip" OBJCOPY="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-objcopy" OBJDUMP="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/usr/bin/gcc" GCC_FOR_BUILD="/usr/bin/gcc" CXX_FOR_BUILD="/usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/home/jkt/work/prog/_build/br-cfb/host/include" CFLAGS_FOR_BUILD="-O2 -I/home/jkt/work/prog/_build/br-cfb/host/include" CXXFLAGS_FOR_BUILD="-O2 -I/home/jkt/work/prog/_build/br-cfb/host/include" LDFLAGS_FOR_BUILD="-L/home/jkt/work/prog/_build/br-cfb/host/lib -Wl,-rpath,/home/jkt/work/prog/_build/br-cfb/host/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-as" DEFAULT_LINKER="/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ld" CPPFLAGS="-fstack-protector-strong -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-fstack-protector-strong -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os " CXXFLAGS="-fstack-protector-strong -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os " LDFLAGS="" FCFLAGS=" -Os " FFLAGS=" -Os " PKG_CONFIG="/home/jkt/work/prog/_build/br-cfb/host/bin/pkg-config" STAGING_DIR="/home/jkt/work/prog/_build/br-cfb/host/arm-buildroot-linux-gnueabihf/sysroot" INTLTOOL_PERL=/usr/bin/perl ac_cv_lbl_unaligned_fail=yes ac_cv_func_mmap_fixed_mapped=yes ac_cv_func_memcmp_working=yes ac_cv_have_decl_malloc=yes gl_cv_func_malloc_0_nonnull=yes ac_cv_func_malloc_0_nonnull=yes ac_cv_func_calloc_0_nonnull=yes ac_cv_func_realloc_0_nonnull=yes lt_cv_sys_lib_search_path_spec="" ac_cv_c_bigendian=no   CONFIG_SITE=/dev/null ./configure --target=arm-buildroot-linux-gnueabihf --host=arm-buildroot-linux-gnueabihf --build=x86_64-pc-linux-gnu --prefix=/usr --exec-prefix=/usr --sysconfdir=/etc --localstatedir=/var --program-prefix="" --disable-gtk-doc --disable-gtk-doc-html --disable-doc --disable-docs --disable-documentation --with-xmlto=no --with-fop=no  --enable-ipv6 --disable-nls --disable-static --enable-shared  --enable-tools --enable-bindings-cxx )
configure: WARNING: unrecognized options: --disable-gtk-doc, --disable-gtk-doc-html, --disable-doc, --disable-docs, --disable-documentation, --with-xmlto, --with-fop, --enable-ipv6, --disable-nls
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-buildroot-linux-gnueabihf-strip... /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for arm-buildroot-linux-gnueabihf-gcc... /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc accepts -g... yes
checking for /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc option to accept ISO C89... none needed
checking whether /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc understands -c and -o together... yes
checking dependency style of /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc... gcc3
checking for arm-buildroot-linux-gnueabihf-ar... /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ar
checking the archiver (/home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-ar) interface... ar
checking for arm-buildroot-linux-gnueabihf-gcc... (cached) /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc accepts -g... (cached) yes
checking for /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc option to accept ISO C89... (cached) none needed
checking whether /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc understands -c and -o together... (cached) yes
checking dependency style of /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-gcc... (cached) gcc3
checking whether we are using the GNU C++ compiler... yes
checking whether /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-g++ accepts -g... yes
checking dependency style of /home/jkt/work/prog/_build/br-cfb/host/bin/arm-linux-gnueabihf-g++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-buildroot-linux-gnueabihf
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
[... 188 lines cut ...]
Making install in cxx
Making install in .
 /bin/mkdir -p '/home/jkt/work/prog/_build/br-cfb/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib'
 /bin/mkdir -p '/home/jkt/work/prog/_build/br-cfb/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include'
 /bin/sh ../../libtool   --mode=install /usr/bin/install -c   libgpiodcxx.la '/home/jkt/work/prog/_build/br-cfb/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib'
[... cut ...]
Makefile:374: recipe for target 'install-recursive' failed
make[3]: *** [install-recursive] Error 1
Makefile:486: recipe for target 'install-recursive' failed
make[2]: *** [install-recursive] Error 1
package/pkg-generic.mk:286: recipe for target '/home/jkt/work/prog/_build/br-cfb/build/libgpiod-custom/.stamp_staging_installed' failed
make[1]: *** [/home/jkt/work/prog/_build/br-cfb/build/libgpiod-custom/.stamp_staging_installed] Error 2
Makefile:16: recipe for target '_all' failed
make: *** [_all] Error 2
```

brgl, 2018-02-23:
This isn't part of any release yet, it's still WIP. I will surely address any build issues before releasing 1.1.

jktjkt, 2018-02-23:
Yeah, sure, I understand that. I just wanted to make myself known as an interested user, and to report back what I have. I'm very interested in C++ bindings because that would help me nuke some of my code :).

brgl, 2018-02-23:
Cool, please don't hesitate to let me know should you find a solution before I do.

jktjkt, 2018-02-26:
A small update -- at first, I was running `./autogen.sh`, which also runs `./configure`. That is obviously wrong. However, even just initializing the dir by running `git clean -dxf; autoreconf --force --install --verbose` doesn't change anything.

brgl, 2018-02-26:
For buildroot, I'd advise to first run make dist from the repo, unpack the tarball somewhere and then use the local SITE_METHOD to fetch the packaged source.

brgl, 2018-03-09:
@jktjkt interesting - I can't build the C++ bindings in buildroot either and am getting the same error, but the yocto SDK builds it fine. I guess there's something in the buildroot environment.

heinemml, 2018-03-29:
Just a observation by me. In my Buildroot it compiles.

By comparison of the failing command I see that I'm not using the `-fstack-protector-strong` parameter.
So your ERROR is a WARNING in my build and runs through. Maybe this is the same reason why it works with yocto?

heinemml, 2018-04-05:
Found this Buildroot issue, which pretty much resembles this problem: http://lists.busybox.net/pipermail/buildroot/2016-September/172127.html

In short: Buildroot requires a patched libtool to handle the different path for crosscompilation.
So with this `libgpiod.mk` I can build from the local git clone. (running `autoreconf --force --install --verbose` before building).

Note, this also adds the dependency on `autoconf-archive` for the host toolchain which is in the git version.

```
################################################################################
#
# libgpiod
#
################################################################################

LIBGPIOD_VERSION = 0.3.2
LIBGPIOD_SOURCE = libgpiod-$(LIBGPIOD_VERSION).tar.xz
LIBGPIOD_SITE = https://www.kernel.org/pub/software/libs/libgpiod
LIBGPIOD_LICENSE = LGPL-2.1+
LIBGPIOD_LICENSE_FILES = COPYING
LIBGPIOD_INSTALL_STAGING = YES
LIBGPIOD_AUTORECONF = YES
LIBGPIOD_AUTORECONF_OPTS += -I $(HOST_DIR)/usr/share/autoconf-archive

LIBGPIOD_DEPENDENCIES = host-pkgconf \
                        host-autoconf-archive

ifeq ($(BR2_PACKAGE_LIBGPIOD_TOOLS),y)
LIBGPIOD_CONF_OPTS += --enable-tools
else
LIBGPIOD_CONF_OPTS += --disable-tools
endif

ifeq ($(BR2_INSTALL_LIBSTDCPP),y)
LIBGPIOD_CONF_OPTS += --enable-bindings-cxx
endif

$(eval $(autotools-package))
```

brgl, 2018-04-09:
@heinemml thanks! Can confirm - it builds for me with the two AUTORECONF options set.

brgl, 2018-09-24:
Bindings are now built properly in buildroot and yocto.

closed 2018-09-24 by brgl