brgl/libgpiod #24: Please revert the API/ABI change in #783ff2e

smurfix, 2018-03-11:
Hi,

while it's true that starting enums with 1 might help with debugging, this is a major API change which is quite unfortunate for libgpiod binding in other languages, or any other code which needs to maintain ABI compatibility. At the very least this requires a soname change.

IMHO the advantages aren't worth the hassle … not at this stage.

brgl, 2018-03-12:
Hi @smurfix 

I already responded to your private e-mail but let me rephrase:

The current stable release is v1.0 and your CFFI bindings should be build against it. Current master branch contains changes scheduled for v1.1. Neither API nor ABI compatibility will be broken in the v1.0.x stable branch to which only bug-fixes for v1.0 will be applied. On the other hand v1.1 release will include changes that keep the source code backward compatible but may break the binary compatibility. You will need to rebuild your bindings once it is out. 

This change does not require an soname change as it's only the case if the backward compatibility is broken in terms of source code. You should stick to the v1.0.x branch if you care about binary compatibility.

I hope that helps.

brgl, 2018-03-12:
Let me quote wikipedia on that:

> If the application programming interface (API) of a library changes in a backward-incompatible way, the soname would be incremented, e.g. from libX.so.5 to libX.so.6.

This change doesn't break the API compatibility. ABI is a different thing and it can be broken if we increment the minor version number.

smurfix, 2018-03-12:
You're right insofar as there is no API change. However, the interface between two pieces of compiled code, such as a program and the shared library it uses, is governed by the ABI. I have corrected the Wikipedia entry.

Looking at the ABI, your change is neither forward nor backward compatible. Specifically, I can't run a program compiled with 1.0 with version 1.1 of the shared library. Thus you can't update the library without also upgrading the programs using it. Therefore the shared library needs a new soname.

NB, the soname version doesn't have to be the same as the library's "official" version number; for instance, the mysqlclient library was at version 16 last time I looked, while the public API version is 5.whatever.

brgl, 2018-03-12:
Ugh, sources on that are confusing to say the least, but you are right and I was wrong - soname should be increased on ABI changes and doesn't have to be linked in any way to the library release version. I want to keep this change, so I will extend the build files and bump the soname.

smurfix, 2018-03-12:
Thank you. I admit it's not easy keeping that stuff straight. I've suffered through the Linux libc 4>5>6 transitions (the 4>5 step was ugly as hell) and sometimes I wish I wouldn't know as much about this topic …

brgl, 2018-03-12:
I'm not seeing any switches in automake/libtool that would allow to easily set the soname of a library. Libtool seems to be setting the name automatically - even if I add -Wl,-soname,<soname> to the LDFLAGS they are ignored, because the build system adds another -soname switch at the end of the gcc command line. Do you have any idea?

brgl, 2018-03-12:
I'm also seeing that on my debian stretch system, there's not a single shared library whose SONAME would be different than its release name.

brgl, 2018-03-12:
Cmake has VERSION and SOVERSION but I can't find an autotools equivalent.

smurfix, 2018-03-13:
https://autotools.io/libtool/version.html

smurfix, 2018-03-13:
admittedly the Cmake way of handling that makes a lot more sense. Autotools is geared for ancient systems where you sometimes had to jump through strange hoops to get that stuff straight.

brgl, 2018-03-13:
@smurfix Please take a look at the latest commit in branch topic/libtool-ABI-versioning and tell me if you think it's good.

smurfix, 2018-03-13:
Looks sane to me.

brgl, 2018-03-14:
Merged, thanks.

closed 2018-03-14 by brgl