brgl/libgpiod #26: Calling line.event_wait in thread hangs main thread

sgjava, 2018-04-21:
[Code](https://github.com/sgjava/userspaceio/blob/1.0/libgpiod/python/src/buttonthread.py) I have using CFFI/Python 3 and libgpiod 1.0 line.event_wait work fine in a thread. Using your new Python bindings this [code](https://github.com/sgjava/userspaceio/blob/master/libgpiod/python/src/buttonthread.py) hangs until timeout or event fires.

brgl, 2018-04-21:
That's interesting, I'm curious if it's not something related to the Big Interpreter Lock in python. I'll take a look.

brgl, 2018-04-21:
Seems like I will need to use Py_BEGIN_ALLOW_THREADS and Py_END_ALLOW_THREADS before and after the internal call to gpiod_line_event_wait() respectively.

sgjava, 2018-04-21:
Let me know when you commit changes and I'll test, thanks.

brgl, 2018-04-21:
@sgjava I don't really have time to test it today, but I pushed a fix to topic/python-locking branch that should theoretically fix it. I should probably add these macros around all blocking operations in the module.

sgjava, 2018-04-21:
I just tested event_wait and it works as expected in a thread. Let me know when you commit changes to master. I'll finish up converting the rest of my example code to the new bindings.

sgjava, 2018-04-23:
I just built and tested my threading example. Looks good, thanks.