brgl/libgpiod #27: RE: Running on a BBB (BeagleBone Black) for GPIOd/Hello!

silver2row, 2018-09-04:
Hello,

Seth here. I am having trouble configuring my BBB (Linux Debian Distro) 4.14.x.

    configure: error: Package requirements (libkmod >= 18) were not met:

    No package 'libkmod' found

    Consider adjusting the PKG_CONFIG_PATH environment variable if you
    installed software in a non-standard prefix.

    Alternatively, you may set the environment variables KMOD_CFLAGS
    and KMOD_LIBS to avoid the need to call pkg-config.
    See the pkg-config man page for more details.

I have been trying to compile at this stage:

    ~/libgpiod$ sudo ./configure --enable-tests

...

I have gotten nowhere so far. I know this is not easy and I should have researched my failure further but...

Do you have any ideas on this situation?

Seth

P.S. I am currently trying to learn how to label my BBB w/ userspaceIO and libgpiod (w/ respective tool cmds).

silver2row, 2018-09-04:
Hello Once More,

Oh and sir...

...

    I am currently, since I found this pin, using gpiochip0 24 as my pin but I do not know which pin this is 
    on the BBB (BeagleBone Black).

Seth

silver2row, 2018-09-04:
Hello,

I just tried to perform this cmd:

    sudo apt install kmod

...

The machine is already up to the newest version of kmod. 

Seth

brgl, 2018-09-04:
You need libkmod-dev and libudev-dev to build tests.

silver2row, 2018-09-04:
Hello,

Aw! Thank you, sir. I will update the Linux Distro on my BBB and then test from there w/ the new libs.

Seth

P.S. I really appreciate the quick reply.

silver2row, 2018-09-06:
Hello Again Sir,

Seth here. You were right. I needed those two, specific libraries. Thank you.

Seth

closed 2018-09-06 by silver2row