brgl/libgpiod #28: Read value of ouput GPIO pin?

hongquan, 2018-09-16:
Hi,

I'm trying to learn how to use _gpiod_ on BeagleBone with Debian Stretch.
When I make a GPIO pin output, I want to read its state to know if it is 0 or 1. But doing so, I always get this error:

> gpioget: error reading GPIO values: Device or resource busy

Is there any way to do so?

brgl, 2018-09-20:
I can't help you if you don't share any code.

pdp7, 2018-09-20:
@hongquan For BeagleBone support, it is very helpful if you can run this script that @robertcnelson includes on the BeagleBoard.org Debian images:
```sudo /opt/scripts/tools/version.sh```

Please paste the output in your reply.

Also, I would recommend using the most [recent BeagleBoard.org Debian image](http://beagleboard.org/latest-images):
[Stretch IoT (without graphical desktop) for BeagleBone and PocketBeagle via microSD card](https://debian.beagleboard.org/images/bone-debian-9.5-iot-armhf-2018-08-30-4gb.img.xz)

brgl, 2018-09-20:
@hongquan you should always at least paste the exact commands you're running and any output they emit.

In your case: gpioset requests given gpio so you can't request it again by gpioget. You can see that if you run gpioset and then gpioinfo.

hongquan, 2018-09-20:
@brgl Tomorrow when I have access to the board, I will send you the exact command.

> gpioset requests given gpio so you can't request it again by gpioget.

Yes, it seems to be the cause. I noticed that, if the GPIO pin is exported to _sysfs_, `gpioget` also cannot read value.

I also remember that after `gpioset` returns, the pin immediately become "input" pin and `gpioget` can read its value.

But this is not very practical. I'm building an IoT application. A GPIO pin is used to switch on/off a pump. My application needs to set that pin to 1 for a while (to keep the pump ON). Meanwhile, the application also need to show on a web UI to let user know if the pump is ON or OFF.

Do you have a resolution for this yet? When the pin is being hold by a process, another process can check if the pin is `0` or `1`.

brgl, 2018-09-20:
But if you're driving the pin then you know best if it's high or low.

mhei, 2018-09-20:
I tend to disagree: imagine a board which receive via network the desired value but does not store it in an extra memory location. Then the web gui should be able to visualize the current state by looking at the register which has exact this information. Actually, this is the cleaner approach instead of keeping a second state variable around - from my point of view...

brgl, 2018-09-20:
It's impossible to have two processes request the same GPIO line due to how the kernel part works. I would suggest the following: use python bindings and write a simple script that would do what you need instead of shell tools. You can have a single python script read and write values of GPIO lines while holding its ownership. I'm also working on a dbus interface for the GPIO character device. Once it's ready you could probably use some command-line dbus tools for that. I assume I'll release it before November.

hongquan, 2018-09-21:
I'm closing issue when waiting for the daemon. I will continue the _sysfs_ approach for now.

Debian Stretch (for BeagleBone) comes with libgpiod v1.0, which doesn't have Python binding.

closed 2018-09-21 by hongquan