brgl/libgpiod #29: Pin direction is wrong when listed in `gpioinfo` output

andy-shev, 2018-09-24:
Consider the following sequence of operations:

    x=DIG7_PU_PD

    gpioinfo | grep $x
        line   7: "DIG7_PU_PD"       unused   input  active-high 

    gpiofind $x
    gpiochip2 7

    gpioset gpiochip2 7=0
    gpioget gpiochip2 7
    0

    gpioset gpiochip2 7=1
    gpioget gpiochip2 7
    1

    gpioinfo | grep $x
        line   7: "DIG7_PU_PD"       unused   input  active-high 

I admit that line is requested only for an operation to be made, but the state of it apparently left output. The last info dump is simple wrong.

It might be kernel GPIO library error, though I don't have time right now to go deeper into it.

andy-shev, 2018-09-24:
...Also this adds an atomic operations support on GPIO line(s).

brgl, 2018-09-24:
Hmm I initially thought that this is caused by the fact that we don't reread the line info after releasing it, but in this case these are separate processes. Your suggestion about gpiorequest and gpiorelease won't work because if one process requests a line, others will receive -EBUSY when calling the linehandle ioctl.

andy-shev, 2018-09-25:
It seems that GPIO handling should be a daemon then. Or some other possibilities to open and share file descriptors (UNIX domain sockets? Sorry, I'm not to user space quite long time, don't know the details)

brgl, 2018-09-25:
Well you can use python bindings for a script that would keep the descriptor open. And yes - a dbus daemon is coming soon. Maybe we'd also need a simpler daemon without all the dbus machinery. TBD

andy-shev, 2018-09-25:
Python is not a solution at all. What I'm working on is an embedded system with as small image as I would like to have. Python is not a part of that image and quite unlikely to be. So, daemon w/o D-Bus sounds plausible for such cases.

So, I guess the proper label to this more likely "enhancement" than a bug.

brgl, 2018-09-25:
Unfortunately this is something we can't overcome. The legacy sysfs interface - with all its shortcoming - served as a kind of GPIO daemon because it kept global state. A character device by its very design will have a state tied to a process. And the issue you reported must come from the gpiolib in the kernel precisely because of that fact.

htot, 2018-10-05:
If I understand correct gpioset changes direction to output, while gpioget changes direction to input:
```
root@edison:~# gpioset gpiochip0 75=0
root@edison:~# gpioinfo | grep 75
        line  75:      unnamed       unused  output  active-high 
root@edison:~# gpioget gpiochip0 75  
0
root@edison:~# gpioinfo | grep 75
        line  75:      unnamed       unused   input  active-high 
```
If so, the direction is shown by gpioinfo is correct. 

But the state "active-high" doesn't make sense to me. I would be expecting low.

andy-shev, 2019-03-26:
So, v1.3 clarifies that gpioset (as a shell tool) is a premature tool.

We really need some knobs to emulated old and very nice behaviour when user can esport (request) pin and do things till it will be unexported. Now, there is no solution provided in the shell. This is no good.

brgl, 2019-03-26:
Yes, we need a daemon. I'm trying to write the dbus daemon first, but as usual, the time is scarce. Especially since we changed the gpio-mockup interface in v5.1 and I need to adjust the testing framework urgently before it's released.

Next I plan to implement a lightweight (as in without other dependencies than glibc) unix-socket based daemon with a client application but unfortunately this will take even more time.

So yeah, I agree, but there's no way of making gpioset work any differently - this is how file descriptors work.

closed 2024-01-12 by brgl

htot, 2024-01-12:
I'm doing this [acpi-tables-load](https://github.com/edison-fw/meta-acpi/blob/mickledore/recipes-bsp/acpi-tables/files/edison/acpi-tables-load)

The intention (and currently seems to work) is:
- load an acpi table that defines a certain GPIO
-  toggle it `gpioset -t 1,0  "TRI_STATE_ALL"=0`
-  unload the table
-  load another table
- set the pin high and keep it high `gpioset -z "TRI_STATE_ALL"=1`

Would that be as intended? 

Of course daemonizing gpioset to keep the line high mean I need to kill it first if I need to toggle it again. Not the best.