brgl/libgpiod #30: Thread safety and the libgpiod C libaray

fivdi, 2018-10-07:
Are the C functions provided by the libgpiod C libaray thread safe?

My current assumption is that the functions are not thread safe because functions like [gpiod_line_update](https://github.com/brgl/libgpiod/blob/master/src/lib/core.c#L291-L320) could be modifying data such as the string pointed to by `line->consumer` in one thread with [this code](https://github.com/brgl/libgpiod/blob/master/src/lib/core.c#L315) ...
```c
	strncpy(line->consumer, info.consumer, sizeof(line->consumer));
```
... while a second thread could be reading the same string using the `const char *` to `line->consumer` returned by [gpiod_line_consumer](https://github.com/brgl/libgpiod/blob/master/src/lib/core.c#L256-L259). The second thread may retrieve an incorrect consumer name.

On the other hand [gpiod_line_event_wait_bulk](https://github.com/brgl/libgpiod/blob/master/src/lib/core.c#L675-L722) is a blocking call which makes it a good candidate for calling in a separate thread. This however would require thread safety.

brgl, 2018-10-07:
Hi @fivdi 

libgpiod is *not* thread safe. More on why here: https://git.kernel.org/pub/scm/linux/kernel/git/kay/libabc.git/plain/README in the "Zero global state" section.

brgl, 2018-10-14:
Yes, there's no global state so that's precisely how it works. Closing.

closed 2018-10-14 by brgl