brgl/libgpiod #32: Beaglebone: gpioset turns pin off and unexports it

shirriff, 2018-12-03:
Problem: gpioset turns a pin off when it should turn the pin on.

Expected behavior: `gpioset gpiochip2 9=1` should turn on gpio2[9]
Observed behavior: `gpioset gpiochip2 9=1` turns off gpio2[9] and unexports it

I'm new to libgpiod, so maybe I'm misunderstanding the API?

Details:
```
$ config-pin P8_44 1 # This turns P8_44 on as expected
$ gpioset gpiochip2 9=1 # Should turn on gpio2[9], i.e. P8_44, but turns it off.
$ config-pin P8_44 1 # Now config-pin can't access the pin???
WARNING: GPIO pin not exported, cannot set direction or value!
$ gpioset --mode=wait gpiochip2 9=1 # This turns the pin on until I hit ^C ?
```

This isn't specific to P8_44; for instance `gpioset gpiochip2 7=1` will mess up P8_46 (gpio2[7]). This also happens even if I don't use config-pin, so it's not config-pin messing things up.

Version: Linux beaglebone 4.14.71-ti-r80 #1 SMP PREEMPT Fri Oct 5 23:50:11 UTC 2018 armv7l GNU/Linux

brgl, 2018-12-04:
Can you post the output of `gpioinfo gpiochip2`?

shirriff, 2018-12-04:
To simplify my issue to the essentials:
```
gpioset gpiochip2 9=1
```
Expected behavior: gpio2[9] (P8_44) is turned on.
Observed behavior: gpio2[9] is not turned on.

```
$ gpioinfo gpiochip2
gpiochip2 - 32 lines:
      line   0:  "GPMC_CSN3"       unused   input  active-high 
      line   1:   "GPMC_CLK"      "P8_18"   input  active-high [used]
      line   2: "GPMC_ADVN_ALE" "P8_07" input active-high [used]
      line   3: "GPMC_OEN_REN" "P8_08" input active-high [used]
      line   4:   "GPMC_WEN"      "P8_10"   input  active-high [used]
      line   5: "GPMC_BE0N_CLE" "P8_09" input active-high [used]
      line   6:  "LCD_DATA0"      "P8_45"   input  active-high [used]
      line   7:  "LCD_DATA1"       unused  output  active-high [used]
      line   8:  "LCD_DATA2"      "P8_43"   input  active-high [used]
      line   9:  "LCD_DATA3"       unused  output  active-high [used]
      line  10:  "LCD_DATA4"      "P8_41"   input  active-high [used]
      line  11:  "LCD_DATA5"      "P8_42"  output  active-high [used]
      line  12:  "LCD_DATA6"      "P8_39"   input  active-high [used]
      line  13:  "LCD_DATA7"      "P8_40"   input  active-high [used]
      line  14:  "LCD_DATA8"      "P8_37"   input  active-high [used]
      line  15:  "LCD_DATA9"      "P8_38"   input  active-high [used]
      line  16: "LCD_DATA10"      "P8_36"   input  active-high [used]
      line  17: "LCD_DATA11"      "P8_34"   input  active-high [used]
      line  18: "GMII1_RXD3"       unused   input  active-high 
      line  19: "GMII1_RXD2"       unused   input  active-high 
      line  20: "GMII1_RXD1"       unused   input  active-high 
      line  21: "GMII1_RXD0"       unused   input  active-high 
      line  22:  "LCD_VSYNC"      "P8_27"   input  active-high [used]
      line  23:  "LCD_HSYNC"      "P8_29"   input  active-high [used]
      line  24:   "LCD_PCLK"      "P8_28"   input  active-high [used]
      line  25: "LCD_AC_BIAS_EN" "P8_30" input active-high [used]
      line  26:  "MMC0_DAT3"       unused   input  active-high 
      line  27:  "MMC0_DAT2"       unused   input  active-high 
      line  28:  "MMC0_DAT1"       unused   input  active-high 
      line  29:  "MMC0_DAT0"       unused   input  active-high 
      line  30:   "MMC0_CLK"       unused   input  active-high 
      line  31:   "MMC0_CMD"       unused   input  active-high 
```

(I don't know why `gpioinfo` says that line 9 is both unused and used.)

`config-pin` is a standard Beaglebone command that uses the character device API to turn a pin on and off and control pin configuration. You can ignore `config-pin` in my report, as `gpioset` doesn't turn the pin on regardless of the use of `config-pin`. (I encourage you to look at the functionality that `config-pin` provides, however, if you want `gpioset` to replace it on the Beaglebone.)

By "unexporting the GPIO", I mean that `gpioset` causes the pin's directory (`gpio73`) to disappear from `/sys/class/gpio/`. You can ignore that too, if use of the character device is incompatible with sysfs.

brgl, 2018-12-05:
> To simplify my issue to the essentials:
> 
> ```
> gpioset gpiochip2 9=1
> ```
> Expected behavior: gpio2[9] (P8_44) is turned on.
> Observed behavior: gpio2[9] is not turned on.
> 

By on I assume you mean high? How do you verify it? If the pin in question is has a pull-down resistor on the input then once gpioset exists, it will go low again. You can test it by running `gpioset --mode=wait gpiochip2 9=1` and then observing the value.

> ```
> $ gpioinfo gpiochip2
> gpiochip2 - 32 lines:
>       line   0:  "GPMC_CSN3"       unused   input  active-high
>       line   1:   "GPMC_CLK"      "P8_18"   input  active-high [used]
>       line   2: [...]

The `[used]` flag means that someone exported the line. The `unused` string means said user didn't set the user name on this line, which is strange... Any idea who may have exported it? Is it used from the kernel space?

> `config-pin` is a standard Beaglebone command that uses the character device API to turn a pin on and off and control pin configuration. You can ignore `config-pin` in my report, as `gpioset` [...]

If this: https://github.com/beagleboard/bb.org-overlays/blob/master/tools/beaglebone-universal-io/config-pin is the config-pin you're talking about then it's not using the character device, it's using the legacy sysfs API.

> By "unexporting the GPIO", I mean that `gpioset` causes the pin's directory (`gpio73`) to disappear from `/sys/class/gpio/`. You can ignore that too, if use of the character device is incompatible with sysfs.

This is impossible, if you export a line in sysfs it will appear as `[used]` and the user string will be `"sysfs"`. If you try to run gpioset on it, then it will bail-out with the following error message: `error setting the GPIO line values: Device or resource busy`.

shirriff, 2018-12-06:
> By on I assume you mean high? How do you verify it? If the pin in question is has a pull-down resistor on the input then once gpioset exists, it will go low again. You can test it by running gpioset --mode=wait gpiochip2 9=1 and then observing the value.

I tried "--mode=wait"; it turns on the output (and attached LED) until I hit ^C. I checked with an oscilloscope and found that gpioset without the wait turns on the output for 35μs. Is this the proper behavior - that it just turns on the pin momentarily? My expectation was that gpioset would let me turn a pin on or off and leave it on or off. Maybe gpioset is the wrong tool for this?

>This is impossible, if you export a line in sysfs it will appear as [used] and the user string will be "sysfs". If you try to run gpioset on it, then it will bail-out with the following error message: error setting the GPIO line values: Device or resource busy.

Well, I pride myself on doing the impossible.
```
$ echo 73 > /sys/class/gpio/export
$ ls -l /sys/class/gpio/gpio73
lrwxrwxrwx 1 root root 0 Dec  6 02:37 /sys/class/gpio/gpio73 -> ../../devices/platform/ocp/481ac000.gpio/gpiochip2/gpio/gpio73
$ gpioinfo gpiochip2 | grep ' 9'
	line   9:  "LCD_DATA3"      "sysfs"   input  active-high [used]
$ gpioset gpiochip2 9=1
$ gpioinfo gpiochip2 | grep ' 9'
	line   9:  "LCD_DATA3"       unused  output  active-high [used]
$ ls -l /sys/class/gpio/gpio73
ls: cannot access '/sys/class/gpio/gpio73': No such file or directory
```

In any case, if gpioset is only intended for transient changes to GPIOs, then you can close this issue.

brgl, 2018-12-06:
> I tried "--mode=wait"; it turns on the output (and attached LED) until I hit ^C. I checked with an oscilloscope and found that gpioset without the wait turns on the output for 35μs. Is this the [...]

There's a significant difference between how sysfs and the character device work. With cdev the state of the GPIO is tied to a process - once a process exits, the state returns to default. So yes, this is expected behavior and is the reason why we have modes for waiting in gpioset.

I'm also working on a dbus daemon that will live at all times and be able to keep a line in requested state.

> Well, I pride myself on doing the impossible.
>
> ```
> $ echo 73 > /sys/class/gpio/export
> $ ls -l /sys/class/gpio/gpio73
> lrwxrwxrwx 1 root root 0 Dec  6 02:37 /sys/class/gpio/gpio73 -> [...]

Hmm this looks like a bug in the kernel. Any chance you could check the same behavior on current mainline?

> In any case, if gpioset is only intended for transient changes to GPIOs, then you can close this issue.

It's not - you can keep it alive to keep the line driven high.

DaStoned, 2019-01-02:
> There's a significant difference between how sysfs and the character device work. With cdev the state of the GPIO is tied to a process - once a process exits, the state returns to default. So yes, this is expected behavior and is the reason why we have modes for waiting in gpioset.

Having just gone through the exact same problem, may I suggest that you add the above explanation to the gpioset documentation, README or some other visible location. Without knowing anything about character devices, this behaviour not something I would expect from a utility which manipulates the state of GPIO outputs. The included note below does not really bring it home to a person who is used to 'sticky-state' sysfs and register interfaces:

```  -m, --mode=[exit|wait|time|signal] (defaults to 'exit'):
                tell the program what to do after setting values```

brgl, 2019-01-02:
@DaStoned sounds good, I'll add it to the usage message displayed by gpioset --help.

closed 2019-02-18 by brgl