brgl/libgpiod #34: gpioget: error reading GPIO values: Device or resource busy

QB4-dev, 2019-01-30:
Hello

More likely it's not libgpiod fault itself. 
I'm using Yocto on iMX6ULL single board computer and I wanted to create  default gpio configuration inside device-tree according this doc:
https://www.kernel.org/doc/Documentation/devicetree/bindings/gpio/gpio.txt

```devicetree
&gpio1{
	bat {
		gpio-hog;
		gpios = <5 GPIO_ACTIVE_HIGH>;
		input;
		line-name = "BAT";
	};
	chrg_stop {
		gpio-hog;
		gpios = <12 GPIO_ACTIVE_HIGH>;
		output-high;
		line-name = "CHARGE_STOP";
	};
};
```

When i use gpioinfo everything looks fine, exept that this configured gpio lines are marked as already used(this leds, and button are driven by gpio-leds and gpio-keys kernel modules so it is the reason why are marked as used):
```devicetree
gpioinfo /dev/gpiochip0
gpiochip0 - 32 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed        "BAT"   input  active-high [used]
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       "BTN_OK"   input  active-high [used]
        line  10:      unnamed      "LED0"  output  active-high [used]
        line  11:      unnamed      "LED1"  output  active-high [used]
        line  12:      unnamed    "CHARGE_STOP" output active-high [used]
```
Reading kernel debug file as mentioned here:
https://stackoverflow.com/questions/16033748/how-do-can-i-find-out-which-linux-driver-is-hogging-my-gpio
gives:
```devicetree
root@visionsom6ull:/sys/kernel/debug# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-31, parent: platform/209c000.gpio, 209c000.gpio:
 gpio-5   (                    |BAT                 ) in  lo
 gpio-9   (                    |BTN_OK          ) in  hi
 gpio-10  (                    |LED0              ) out lo
 gpio-11  (                    |LED1              ) out lo
 gpio-12  (                    |CHARGE_STOP         ) out hi
```
I can't figure out which one component is using those named lines.
Is gpio manipulation on pins like that even possible from userspace?

All I need to do is to read one signal, and set another. 
It is too easy task to write special kernel module.

brgl, 2019-01-30:
The `gpio-hog;` property makes the kernel take over the line right after the chip is registered. Drop it and you should see the lines as unused.

QB4-dev, 2019-01-30:
Thank You for instant reply. Indeed when I dropped `gpio-hog;` property the lines are available, but they lost initial configuration:
```devicetree
output-high;
line-name = "CHARGE-STOP";
```
now gpioinfo result looks like:
```devicetree
gpiochip0 - 32 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed       unused   input  active-high
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       "BTN_OK"   input  active-high [used]
        line  10:      unnamed      "LED0"  output  active-high [used]
        line  11:      unnamed      "LED1"  output  active-high [used]
        line  12:       unnamed       unused   input  active-high
```
And I can use them by gpioget and gpioset.

It seems that without `gpio-hog` no action is taken during boot, otherwise kernel is configuring gpios during boot, but takes control to kernelspace.
This is also mentioned here:
https://stackoverflow.com/questions/48325367/purpose-and-usage-of-gpio-hog-declaration

So it looks like it is impossible to setup initial state of  gpio in device tree and after that use it from userspace.

closed 2019-02-18 by brgl