brgl/libgpiod #35: If there is no /dev/gpiochip-device, cxx-binding is crashing

pboettch, 2019-02-01:
I'm simply running gpiodetectcxx.cpp on my PC, where there is no gpiochip in /dev. It quits with a segfault.

The problem is, that the chip_iter is creating a `chip` with a `gpiod_chip` set to `NULL` and the deleter is calling `::gpiod_chip_close(chip);` which crashes because it dereferences a NULL-pointer.

I don't know how to fix it properly. I added locally a NULL-check inside the gpiod_chip_close-function.

brgl, 2019-02-01:
Thanks for the report! I'll look into it because it's strange - the shared_ptr deleter shouldn't be called if the pointer is not referencing any object.

pboettch, 2019-02-01:
```C++
#include <iostream>
#include <memory>

void deleter(int *i)
{
	std::cerr << "deleting " << i << "\n";
}

int main(void)
{
	std::shared_ptr<int> ptr(nullptr, deleter);
	return 0;
}
```

prints 'deleting 0'.

pboettch, 2019-02-01:
Very interesting. https://stackoverflow.com/questions/11164354/does-the-standard-behavior-for-deleters-differ-between-shared-ptr-and-unique-ptr

The first answer says 

> The important detail here is the expression "owns an object p". 20.7.2.2/1 says that "a shared_ptr object is empty if it does not own a pointer". 20.7.2.2.1/9 (the relevant constructor) says that it "constructs a shared_ptr object that owns the object p and the deleter d".

A shared_ptr owns a pointer if set, even if it is nullptr.

If this is correct, the right thing is to check for nullptr before calling `::gpiod_chip_close` in the deleter.

pboettch, 2019-02-01:
I happily make a pull-request for this ;-).

brgl, 2019-02-01:
Nice catch! I have some comments regarding the PR though. Let me answer inline.

brgl, 2019-02-18:
No fixed.

closed 2019-02-18 by brgl