brgl/libgpiod #37: Nice to have: online API docs

nettings, 2019-02-10:
Hey guys, love libgpiod, but it would make things a lot easier if the API documentation of the current release could be built automatically and hosted somewhere on the web, whether here or on kernel.org.

brgl, 2019-02-18:
Using github pages has been on my list for a long time, but I'm not sure if it's possible to have it generated automatically from the same repo without adding any additional files that don't belong in the main repo on kernel.org. Any hints?

nettings, 2019-02-18:
Not really. I'm currently figuring out more and more github features, but I haven't seen anything yet that would match this problem. I know you can somehow attach release files to your project without cluttering the repo with generated files, but I don't exactly know how, and if that can only be a single tarball or a whole hierarchy of doxygen stuff. Also it would have to be displayed "raw"... if I find something out, I'll let you know.

closed 2024-01-12 by brgl