brgl/libgpiod #38: libgpiod v.1.0.1: gpiod_ctxless_event_loop_multiple() segfaults when numlines=0

nettings, 2019-03-05:
Hi, apologies for reporting a bug in a now deprecated function of an old library version, but it's what I have available on the latest Raspbian stretch.
When you call gpiod_ctxless_event_loop_multiple with no lines set (offsets is initialized to an array of zeroes, numlines is 0), it segfaults.
Would be nice if the user didn't have to add checks before calling, since it's just a degenerate case of setting up an event handler that should be legal.
Sorry if this is already fixed in the current version, right now I have no way of upgrading.

brgl, 2019-03-06:
It's still there in current master. While segfaulting in this case is wrong, I can't quite seem why it should be legal? You can't watch zero lines, can you? Even calling the relevant ioctl() wouldn't be possible. The fix for that should be returning -1 and setting errno to -EINVAL.

brgl, 2019-03-06:
Please check current master.

nettings, 2019-03-11:
Rebuilt my program with current master, removed my guard against empty offsets[] and nlines == 0, got
```
../gpiod_process.c:287	start_GPIOD():	gpiod_ctxless_event_loop_multple: err = -1, errno = 22 (Invalid argument).
```
instead of segfault. Program runs happily ever after. Thanks!

closed 2019-03-11 by nettings