brgl/libgpiod #39: doxygen generation broken on master

nettings, 2019-03-09:
Doxygen generation is currently broken. When you browse to libgpiod/doc/html/modules.html, the list of all modules is only this: '>common</strong> Common helper macros | Commonly used utility macros'

It looks like some garbled syntax somewhere. These are the build errors I get:
```
nettings@hoppetosse:~/github/libgpiod> make doc
/home/nettings/github/libgpiod/include/gpiod.h:79: warning: group strong: ignoring title ">high_level</strong> High-level API" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:386: warning: group strong: ignoring title ">chips</strong> GPIO chip operations" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:523: warning: group strong: ignoring title ">lines</strong> GPIO line operations" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:528: warning: group strong: ignoring title ">line_bulk</strong> Operating on multiple lines" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:636: warning: group strong: ignoring title ">line_info</strong> Line info" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:754: warning: group strong: ignoring title ">line_request</strong> Line requests" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:1073: warning: group strong: ignoring title ">line_value</strong> Reading & setting line values" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:1124: warning: group strong: ignoring title ">line_event</strong> Line events handling" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:1208: warning: group strong: ignoring title ">line_misc</strong> Misc line functions" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:1259: warning: group strong: ignoring title ">iterators</strong> Iterators for GPIO chips and lines" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:1383: warning: group strong: ignoring title ">misc</strong> Stuff that didn't fit anywhere else" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/bindings/cxx/gpiod.hpp:26: warning: group strong: ignoring title ">gpiod_cxx</strong> C++ bindings" that does not match old title ">common</strong> Common helper macros"

/home/nettings/github/libgpiod/include/gpiod.h:528: warning: Refusing to add group strong to itself
/home/nettings/github/libgpiod/include/gpiod.h:636: warning: Refusing to add group strong to itself
/home/nettings/github/libgpiod/include/gpiod.h:754: warning: Refusing to add group strong to itself
/home/nettings/github/libgpiod/include/gpiod.h:1073: warning: Refusing to add group strong to itself
/home/nettings/github/libgpiod/include/gpiod.h:1124: warning: Refusing to add group strong to itself
/home/nettings/github/libgpiod/include/gpiod.h:1208: warning: Refusing to add group strong to itself
```

Probably a simple syntax error somewhere, but I'm not too familiar with doxygen. Will see if I can find it myself, if not: now you know.

brgl, 2019-03-11:
I'm not seeing it - what version of doxygen are you using?

nettings, 2019-03-11:
1.8.15, the default installation that is part of openSUSE Tumbleweed. I must confess this is the first time I've tried building doxygen documentation on this distro, so it may be a problem on my side.
Was looking into it on the train, but couldn't track it down - it looked like something trivial as a forgotten closing tag in the markup, but it seems that this problem is somewhere in doxygen, not in libgpiod... Will report back if I find something out.