brgl/libgpiod #40: gpiod_ctxless_set_value_cb() doesn't keep value on pin after returning

caseygum, 2019-03-14:
I'm a bit of a noob with embedded linux, so if there's an obvious reason for this behavior, please let me know.

I've got a BeagleBone Black and am experimenting with this library in C.  Prior to using this library or the associated tools, if I wanted to set an output pin high, I would do this:

cd /sys/class/gpio
(if gpioXX directory not present, then echo XX > export)
cd gpioXX
echo 1 > value

This would keep the value high indefinitely.

When I call gpiod_ctxless_set_value("gpiochipX", Y, 1, false, NULL, NULL), the value sets high and immediately goes back to low.  What's the best way of setting the value and having it stick until the next time a change it?
Thanks!

brgl, 2019-03-15:
This is what the callback pointer is for. This function opens the character device, sets values and calls the callback function. Once it returns, the values revert to defaults because the character device file descriptor is closed. This is precisely how we want it to work - if the process holding the line crashes, we know that the state will not be undefined.

caseygum, 2019-03-15:
Ah, okay - that makes sense.  Since the callback runs synchronously, it blocks the caller until the callback is complete.  That was kind of screwing me up.  I guess I'll just need to spin up a thread for each output line I want to write to.  Thanks for getting back to me!

closed 2019-06-27 by brgl