brgl/libgpiod #41: Faster line direction change

rodarima, 2019-03-22:
Hi,

I currently trying to read the DHT11 sensor, connected via GPIO to a Raspberry Pi 3 B. The sensor requires the line to be set to 0 for at least 18ms and then 1 again. Then, the sensor will wait for 50us to pull the line to 0 again, then up after 50 us, and start sending data.

I would like to catch the two edge events, but I'm not able to switch the line direction faster enough, and I'm missing the first edge. The code I'm using is the following:

```c
	if(gpiod_line_request_output(line, "temp output", 0) < 0)
	{
		perror("gpiod_line_request_output()");
		return 1;
	}

	nanosleep(&dt, NULL);

	clock_gettime(CLOCK_MONOTONIC, &tic);

	gpiod_line_release(line);

	if(gpiod_line_request_input(line, "temp reader"))
	{
		perror("gpiod_line_request_input()");
		return 1;
	}

	clock_gettime(CLOCK_MONOTONIC, &toc);

	if(toc.tv_sec > tic.tv_sec)
	{
		toc.tv_sec -= tic.tv_sec;
		toc.tv_nsec += 1000000000;
	}
	toc.tv_nsec -= tic.tv_nsec;

	printf("Change direction took %ld ns\n", toc.tv_nsec);
```
Which leads to the following measurements:

```
% repeat 10 ./temp gpiochip0
Change direction took 61353 ns
Change direction took 65208 ns
Change direction took 63280 ns
Change direction took 69374 ns
Change direction took 63802 ns
Change direction took 86145 ns
Change direction took 72603 ns
Change direction took 58957 ns
Change direction took 70103 ns
Change direction took 64322 ns
```

Is there a faster way to change the line direction using the current exposed API calls? And if not, is this a useful use case to be considered maybe in something like (of an already requested line):

```c
gpiod_line_set_direction(struct gpiod_line *line, int direction, int default_val)
```

Best regards,
Rodrigo.

brgl, 2019-03-25:
The low-level API uses the fastest possible code paths - that is the underlying ioctl() calls. There's really very low overhead over what the kernel offers, so I'm afraid it'll be difficult to speed it up. I'm also not sure you can achieve the kind of performance you need from user space - for every ioctl() you have two context switches.

mhei, 2019-03-25:
I doubt that userspace is sufficient, too.
But there is a mainline kernel driver for DHT11: https://github.com/torvalds/linux/blob/master/drivers/iio/humidity/dht11.c Maybe it's possible to give it a try?

brgl, 2019-03-25:
I'm not sure what you mean by giving it a try.

mhei, 2019-03-25:
Sorry for my unclear wording: when a kernel level driver exists for this sensor, maybe it's possible to use this (and the iio interface) instead of creating a libgpiod-based userspace driver for the sensor?

rodarima, 2019-03-31:
Hi,
I was reading the kernel GPIO driver documentation, as well as the kernel driver for DHT11 pointed out by @mhei (thanks!).

I currently already have access to the sensor, but I would like to tinker with it. In the DHT11 driver, the line is first set to input, and then the IRQ is requested.

```c
                ret = gpio_direction_output(dht11->gpio, 0);
                if (ret)
                        goto err;
                usleep_range(DHT11_START_TRANSMISSION_MIN,
                             DHT11_START_TRANSMISSION_MAX);
                ret = gpio_direction_input(dht11->gpio);
                if (ret)
                        goto err;
                ret = request_irq(dht11->irq, dht11_handle_irq,
                                  IRQF_TRIGGER_RISING | IRQF_TRIGGER_FALLING,
                                  iio_dev->name, iio_dev);
                if (ret)
                        goto err;
```

This leads to the posibility were some events may be missing, while the IRQ is not set up. It should be faster as two context switches are avoided, but it doesn't seem like a good design. I was wondering if the interrupt could be requested *before* setting the line to input, so it should not miss any edge (hopefully).

Interestingly, it seems it was [posible before](https://github.com/raspberrypi/linux/issues/808) but I didn't try now yet. I will try to get it working as soon as I have some spare time, but for now, I would say that the issue can be closed, as is not a limitation of libgpiod.

closed 2019-04-01 by brgl