brgl/libgpiod #47: Implement DBus API

brgl, 2024-01-15:
This ticket tracks the development of the DBus daemon. Current development branch can be found [here](https://github.com/brgl/libgpiod-private/tree/topic/dbus).

The API is not complete yet but some parts are functional. The daemon, command-line client and tests can be built with: `./autogen.sh --enable-bindings-glib --enable-dbus --enable-tests && make`. The daemon tests can be run with `sudo GPIODBUS_TEST_DAEMON_PATH=dbus/manager/gpio-manager ./dbus/tests/gpiodbus-test` after `dbus/data/io.gpiod1.conf` is installed to `/etc/dbus-1/system.d/`. Command-line client tests can be run with `sudo PATH=<path to libgpiod>/tests/bash/:$PATH ./dbus/client/gpiocli-test.bash`. In order to run these, the daemon must be started in the background first.

brgl, 2024-03-28:
A little update: the DBus API is getting close to being functional. The development branch was updated. A big part of the command-line client are ready and can be run with the daemon in the background. I expect to post the series for review in around 2 weeks if nothing happens.

brgl, 2024-05-10:
An RFC with the DBus API has been posted to linux-gpio: https://lore.kernel.org/linux-gpio/20240412122804.109323-1-brgl@bgdev.pl/

o-alquimista, 2024-05-28:
> This ticket tracks the development of the DBus daemon. Current development branch can be found [here](https://github.com/brgl/libgpiod-private/tree/topic/dbus).

It seems the branch was renamed and this URL is broken. Is [b4/dbus](https://github.com/brgl/libgpiod-private/tree/b4/dbus) the correct branch? It's the only branch right now.

brgl, 2024-05-28:
Yes! I started using `b4` for libgpiod and enrolled the old `topic/dbus` branch into a managed branch. Sorry.

o-alquimista, 2024-05-28:
```
$ ./autogen.sh --enable-bindings-glib --enable-dbus --enab
le-tests && make

(...)
checking for mount >= 2.33.1... no
configure: error: Package requirements (mount >= 2.33.1) were not met:

Package 'mount' not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables MOUNT_CFLAGS
and MOUNT_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

I'm on Alpine Linux 3.20.

```
$ mount --version
mount from util-linux 2.40.1 (libmount 2.40.1: btrfs, namespaces, idmapping, fd-based-mount, statx, assert, debug)
```

Don't know what else to try.

mhei, 2024-05-28:
Looks like some kind of "devel" package for mount is missing. Probably you must run something similar to "apk add ..." But I'm not familiar with Alpine and the available packages etc.

o-alquimista, 2024-05-28:
Can't find any.

My last attempt: https://pkgs.alpinelinux.org/packages?page=1&name=%2amount%2a&branch=edge

Tried `mount-dev` (the alpine convention), `libmount` (already installed), `mount-*`, `mount*`...

brgl, 2024-05-29:
@o-alquimista do you have the `pkgconfig/mount.pc` file anywhere in your system?

o-alquimista, 2024-05-29:
> @o-alquimista do you have the `pkgconfig/mount.pc` file anywhere in your system?

Tried `# find / -type f -name "mount.pc"` and got nothing. I'll ask the alpine people on IRC.

o-alquimista, 2024-05-29:
It's `util-linux-dev`. I only found it through this form: https://pkgs.alpinelinux.org/contents?file=mount.pc&path=&name=&branch=edge

o-alquimista, 2024-05-29:
Manager won't start:

```
$ doas gpio-manager -d
gpio-manager-Message: 15:46:55.368: initializing gpio-manager
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.368: initializing GPIO DBus daemon
gpio-manager-Message: 15:46:55.370: gpio-manager started
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.373: DBus connection acquired
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.375: exporting object for GPIO chip: '/io/gpiod1/chips/gpiochip0'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.376: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line0'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.376: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line1'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.377: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line2'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.378: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line3'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.379: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line4'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.380: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line5'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.381: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line6'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.381: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line7'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.382: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line8'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.383: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line9'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.384: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line10'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.384: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line11'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.385: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line12'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.386: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line13'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.386: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line14'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.387: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line15'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.388: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line16'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.388: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line17'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.389: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line18'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.389: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line19'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.390: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line20'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.391: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line21'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.391: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line22'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.392: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line23'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.392: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line24'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.393: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line25'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.393: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line26'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.394: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line27'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.394: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line28'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.395: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line29'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.396: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line30'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.396: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line31'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.397: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line32'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.397: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line33'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.398: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line34'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.398: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line35'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.399: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line36'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.400: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line37'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.400: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line38'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.401: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip0/line39'
[... cut ...]
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.414: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip1/line6'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.415: exporting object for GPIO line: '/io/gpiod1/chips/gpiochip1/line7'
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.415: GPIO daemon now listening
(gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.416: DBus name lost: 'io.gpiod1'

(gpio-manager:20381): gpio-manager-ERROR **: 15:46:55.416: name 'io.gpiod1' lost on the bus
Trace/breakpoint trap
```

Service for system dbus is running.

brgl, 2024-05-29:
Can you run it through both strace and gdb and post results here? The entire strace log and gdb stack trace?

honzamzr, 2024-05-31:
> Manager won't start:
>
> ```
> $ doas gpio-manager -d
> gpio-manager-Message: 15:46:55.368: initializing gpio-manager
> (gpio-manager:20381): gpio-manager-DEBUG: 15:46:55.368: initializing GPIO DBus daemon [...]

I had same issue. My io.gpiod1.conf was stored on bad place because of --prefix=/usr. I fixed that with --sysconfdir=/etc

brgl, 2024-05-31:
Ah! So the autotools docs say:

```
(...) by default sysconfdir will be ‘$(prefix)/etc’, because this is what the GNU Standards require. When such a package is installed on an FHS compliant system, the installer will have to set ‘--sysconfdir=/etc’. As the maintainer of the package you should not be concerned by such site policies: use the appropriate standard directory variable to install your files so that the installer can easily redefine these variables to match their site conventions.
```

So I didn't want to hard-code the path to `/etc/`.

honzamzr, 2024-05-31:
I understand, I do not want change this behaviour. I only wanted to say @o-alquimista 
 what could be reason for his problem.

brgl, 2024-05-31:
Sure, thank you!

o-alquimista, 2024-05-31:
> Can you run it through both strace and gdb and post results here? The entire strace log and gdb stack trace?

[strace-gpio-manager.log](https://github.com/user-attachments/files/15515570/strace-gpio-manager.log)

```
#0  __restore_sigs (set=set@entry=0x7ffffff5a0)
    at ./arch/aarch64/syscall_arch.h:48
#1  0x0000007ff7f8f088 in raise (sig=<optimized out>) at src/signal/raise.c:11
#2  0x0000007ff7b7937c in g_log_default_handler () at /usr/lib/libglib-2.0.so.0
#3  0x0000007ff7b79658 in g_logv () at /usr/lib/libglib-2.0.so.0
#4  0x0000007ff7b79870 in g_log () at /usr/lib/libglib-2.0.so.0
#5  0x00000055555574b8 in on_name_lost
    (con=0x7ff7fe6940, name=0x7ff7c75b30 "io.gpiod1", data=<optimized out>)
    at gpio-manager.c:68
#6  0x0000007ff7e26930 in ??? () at /usr/lib/libgio-2.0.so.0
#7  0x0000007ff7e273a0 in ??? () at /usr/lib/libgio-2.0.so.0
#8  0x0000007ff7db8f00 in ??? () at /usr/lib/libgio-2.0.so.0
#9  0x0000007ff7db9dec in ??? () at /usr/lib/libgio-2.0.so.0
#10 0x0000007ff7e1c8a0 in ??? () at /usr/lib/libgio-2.0.so.0
#11 0x0000007ff7db8f00 in ??? () at /usr/lib/libgio-2.0.so.0
#12 0x0000007ff7db8f44 in ??? () at /usr/lib/libgio-2.0.so.0
#13 0x0000007ff7b6d1e8 in ??? () at /usr/lib/libglib-2.0.so.0
#14 0x0000007ff7b70a10 in ??? () at /usr/lib/libglib-2.0.so.0
#15 0x0000007ff7b7141c in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#16 0x00000055555547e8 in main (argc=<optimized out>, argv=<optimized out>)
    at gpio-manager.c:160
```

brgl, 2024-05-31:
Did @honzamzr 's suggestion help? Is the io.gpiod1.conf in the right place under `/etc/dbus-1/system.d/`?

o-alquimista, 2024-05-31:
> This ticket tracks the development of the DBus daemon. Current development branch can be found [here](https://github.com/brgl/libgpiod-private/tree/topic/dbus).
>
> The API is not complete yet but some [...]

## Automated tests
I had difficulty running the commands with `doas` instead of `sudo`.

For example, `doas GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager ./dbus/tests/gpiodbus-test` doesn't work.
> doas: GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager: command not found

And if you move `doas` after the env var: `GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager doas ./dbus/tests/gpiodbus-test`.

```
TAP version 13
# random seed: R02S5f216aed3b4f07a478ff9eef17a40abc
# gpiod-test-DEBUG: running libgpiod test suite
# gpiod-test-DEBUG: 7 tests registered
# gpiod-test-DEBUG: checking linux kernel version
# gpiod-test-DEBUG: kernel release is v6.6.31 - ok to run tests
1..7
# Start of gpiod tests
# Start of gpiodbus tests
# Start of chip tests
**
gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 1 /gpiod/gpiodbus/chip/read_chip_info
not ok /gpiod/gpiodbus/chip/chip_added - gpiodbus-test-FATAL-ERROR: GPIODBUS_TEST_DAEMON_PATH environment variable must be set
Bail out!
Trace/breakpoint trap
```

It fails, as the GPIODBUS_TEST_DAEMON_PATH variable is not set.

So I just executed everything under `su -l`.

### Daemon tests

```
# GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager ./dbus/tests/gpiodbus-test

TAP version 13
# random seed: R02S8df58825df16303ff054fe6dddf44ee9
# gpiod-test-DEBUG: running libgpiod test suite
# gpiod-test-DEBUG: 7 tests registered
# gpiod-test-DEBUG: checking linux kernel version
# gpiod-test-DEBUG: kernel release is v6.6.31 - ok to run tests
1..7
# Start of gpiod tests
# Start of gpiodbus tests
# Start of chip tests
**
gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 1 /gpiod/gpiodbus/chip/read_chip_info
**
gpiodbus-test:ERROR:tests-chip.c:76:_gpiod_test_func_chip_added: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-chip.c:76:_gpiod_test_func_chip_added: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 2 /gpiod/gpiodbus/chip/chip_added
**
gpiodbus-test:ERROR:tests-chip.c:96:_gpiod_test_func_chip_removed: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-chip.c:96:_gpiod_test_func_chip_removed: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 3 /gpiod/gpiodbus/chip/chip_removed
# End of chip tests
# Start of line tests
**
gpiodbus-test:ERROR:tests-line.c:49:_gpiod_test_func_read_line_properties: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-line.c:49:_gpiod_test_func_read_line_properties: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 4 /gpiod/gpiodbus/line/read_line_properties
**
gpiodbus-test:ERROR:tests-line.c:163:_gpiod_test_func_properties_changed: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-line.c:163:_gpiod_test_func_properties_changed: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 5 /gpiod/gpiodbus/line/properties_changed
# End of line tests
# Start of request tests
**
gpiodbus-test:ERROR:tests-request.c:66:_gpiod_test_func_request_input_lines: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-request.c:66:_gpiod_test_func_request_input_lines: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 6 /gpiod/gpiodbus/request/request_input_lines
**
gpiodbus-test:ERROR:tests-request.c:91:_gpiod_test_func_release_request: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
# gpiodbus-test:ERROR:tests-request.c:91:_gpiod_test_func_release_request: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
not ok 7 /gpiod/gpiodbus/request/release_request
# End of request tests
# End of gpiodbus tests
# End of gpiod tests
```

### Command-line tests
```
# PATH=<path to libgpiod>/tests/bash/:$PATH ./dbus/client/gpiocli-test.bash
```

I could not find `bash` in `<libgpiod-git-repo>/tests/bash`:

And in any case, I'm not using bash, I'm using Busybox "ash".

brgl, 2024-05-31:
> > This ticket tracks the development of the DBus daemon. Current development branch can be found [here](https://github.com/brgl/libgpiod-private/tree/topic/dbus).
> > The API is not complete yet but some parts are functional. The daemon, command-line client and tests can be built with: `./autogen.sh --enable-bindings-glib --enable-dbus --enable-tests && make`. The daemon tests can be run with `sudo GPIODBUS_TEST_DAEMON_PATH=dbus/manager/gpio-manager ./dbus/tests/gpiodbus-test` after `dbus/data/io.gpiod1.conf` is installed to `/etc/dbus-1/system.d/`. Command-line client tests can be run with `sudo PATH=<path to libgpiod>/tests/bash/:$PATH ./dbus/client/gpiocli-test.bash`. In order to run these, the daemon must be started in the background first.
> 
> ## Manual tests

Thanks a lot for testing it and reporting back, really appreciated!

> `gpiocli` commands `detect` and `info` are working. They don't need to be run as root.
> 
> A line must be requested before it can be used in `get` and `set`.
> 
> > gpiocli get: Line 'GPIO18' not managed by gpio-manager, must be requested first
> 
> `gpiocli request` must be run as root, or it fails with "access denied".
> 

This is a feature. It's the default, strict config. It can be changed according to your needs, just modify the io.gpiod.conf file.

> ```
> $ gpiocli request -c 0 GPIO18
> gpiocli request: failed to request lines from chip '0': GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 1 matched rules; type="method_call", sender=":1.39" (uid=1000 pid=29317 comm="gpiocli request -c 0 GPIO18") interface="io.gpiod1.Chip" member="RequestLines" error name="(unset)" requested_reply="0" destination=":1.0" (uid=0 pid=28991 comm="gpio-manager")
> ```

Ah, I need to make this error message more readable.

> 
> Requesting a line:
> 
> ```
> $ doas gpiocli request -c 0 GPIO18
> request0
> ```
> 
> List requested lines:
> 
> ```
> $ gpiocli requests
> request0 (gpiochip0) Offsets: [18]
> ```
> 
> `gpiocli get` also needs root.
> 
> The help text says `gpiocli get: either at least one line or the request must be specified`. But if I specify `request0` (the request name), it doesn't work:
> 
> ```
> $ doas gpiocli get request0
> gpiocli get: Line not found: request0
> ```

Because you need to run it as `gpiocli get -r request0`. Otherwise the program assumes `request0` is a name of one of the lines.

> 
> This works though:
> 
> ```
> $ doas gpiocli get GPIO18
> "GPIO18"=active
> ```
> 
> I also tested `gpiocli set`, and it works as expected.
> 
> ## Automated tests
> The cloned repository is on branch `b4/dbus`. The command I used to configure the project: `./autogen.sh --enable-bindings-glib --enable-dbus --enable-tools --enable-tests && make`
> 
> I had difficulty running the commands with `doas` instead of `sudo`.
> 

I'm not familiar with doas.

> For example, `doas GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager ./dbus/tests/gpiodbus-test` doesn't work.
> 
> > doas: GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager: command not found
> 
> And if you move `doas` after the env var: `GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager doas ./dbus/tests/gpiodbus-test`.
> 
> ```
> not ok /gpiod/gpiodbus/chip/chip_added - gpiodbus-test-FATAL-ERROR: GPIODBUS_TEST_DAEMON_PATH environment variable must be set
> Bail out!
> ```
> 
> So I just executed everything under `su -l`.
> 
> ### Daemon tests
> ```
> # GPIODBUS_TEST_DAEMON_PATH=/usr/local/bin/gpio-manager ./dbus/tests/gpiodbus-test
> 
> TAP version 13
> # random seed: R02S8df58825df16303ff054fe6dddf44ee9
> # gpiod-test-DEBUG: running libgpiod test suite
> # gpiod-test-DEBUG: 7 tests registered
> # gpiod-test-DEBUG: checking linux kernel version
> # gpiod-test-DEBUG: kernel release is v6.6.31 - ok to run tests
> 1..7
> # Start of gpiod tests
> # Start of gpiodbus tests
> # Start of chip tests
> **
> gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)

You kernel most likely doesn't have the `gpio-sim` module compiled?

> # gpiodbus-test:ERROR:tests-chip.c:18:_gpiod_test_func_read_chip_info: assertion failed (_err == NULL): Unable to initialize libgpiosim: No such file or directory (g-gpiosim-error, 1)
> not ok 1 [...]

Bash is a requirement for running the tests. We just use COPROC and other bashisms a lot. It's not needed for anything else though.

Bart

o-alquimista, 2024-09-02:
> DBus support can be built by passing --enable-dbus to configure.

Shouldn't the README also tell us to pass `--enable-bindings-glib`, given that the configure script fails if you don't?

> configure: error: GLib bindings are required to build the dbus daemon - use --enable-bindings-glib

brgl, 2024-09-02:
I sent a patch to the mailing list: https://lore.kernel.org/linux-gpio/20240902165636.43268-1-brgl@bgdev.pl/