brgl/libgpiod #48: gpiodget failure with inappropriate ioctl for device

JoseGuilhermeCR, 2024-01-22:
Hello there!

First, I'd like to thank all who contribute to libgpiod which is really handy! :D

I'm currently developing a GPIO expander module for Linux 6.1.70 which runs on a Raspberry PI.  Basic functionality should be implemented as a `gpio_chip` is created and provides `get_direction`, `direction_input`, `direction_output`, `get` and `set` functions.

For quick testing purposes I'm using a manually built libgpiod at commit `fc2f2feaa99dce8aab6381168051f2725b537621` which is from january 16, so fairly recent. This is so I can make sure the executables are using libgpiod V2 (the one that comes with Raspberry PI's OS is the V1).

After loading my module, I successfully create a gpiochip. In my system, /dev/gpiochip2. This can be checked with `gpiodetect`:
```
wedev@raspberrypi:~/Downloads/libgpiod $ ./tools/gpiodetect
gpiochip0 [pinctrl-bcm2711] (58 lines)
gpiochip1 [raspberrypi-exp-gpio] (8 lines)
gpiochip2 [pi4ioe5v6416] (16 lines)
```

and `gpioinfo 2`:
```
gpiochip2 - 16 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused   input  active-high 
        line   3:      unnamed       unused   input  active-high 
        line   4:      unnamed       unused   input  active-high 
        line   5:      unnamed       unused  output  active-high 
        line   6:      unnamed       unused   input  active-high 
        line   7:      unnamed       unused   input  active-high 
        line   8:      unnamed       unused   input  active-high 
        line   9:      unnamed       unused   input  active-high 
        line  10:      unnamed       unused   input  active-high 
        line  11:      unnamed       unused   input  active-high 
        line  12:      unnamed       unused   input  active-high 
        line  13:      unnamed       unused   input  active-high 
        line  14:      unnamed       unused   input  active-high 
        line  15:      unnamed       unused   input  active-high 
```

The problem I'm facing is that `gpioget` is complaining about an invalid IOCTL when I try to use it, even though `gpioset` works just fine.
```
wedev@raspberrypi:~/Downloads/libgpiod $ ./tools/gpioget -c 2 0
/home/wedev/Downloads/libgpiod/tools/.libs/gpioget: unable to read GPIO line values: Inappropriate ioctl for device
```

brgl, 2024-01-23:
It is quite unusual that the returned fd is 0. This could only happen if you previously closed all your stdin which doesn't appear to be the case. Can you confirm that this is the value returned by the call to `get_unused_fd_flags()` in `linehandle_create()` in `drivers/gpio/gpiolib-cdev.c` in the kernel?

JoseGuilhermeCR, 2024-01-23:
Yes, that is what I thought. Since I don't close `stdin`, what happens at the end is that the IOCTL is indeed wrong... because I did call GPIOD IOCTL on `stdin`.  Either way, I got it working after trying to debug the return value you've specified.

I placed the following line at `linehandle_create()` just after `get_unused_fd_flags()`:

```
diff --git a/drivers/gpio/gpiolib-cdev.c b/drivers/gpio/gpiolib-cdev.c
index e40c93f0960b..a6394fc25e79 100644
--- a/drivers/gpio/gpiolib-cdev.c
+++ b/drivers/gpio/gpiolib-cdev.c
@@ -426,6 +426,8 @@ static int linehandle_create(struct gpio_device *gdev, void __user *ip)
                goto out_free_lh;
        }
 
+       dev_info(&gdev->dev, "get_unused_fd_flags() returned %i", fd);
+
        file = anon_inode_getfile("gpio-linehandle",
                                  &linehandle_fileops,
                                  lh,
```
There's actually a catch here: `linehandle_create()` is only used in API V1, so I used libgpiod that comes bundled with the Raspberry PI's distribution. Either way, after using `gpioget` I noticed that when I called a get on `gpiochip0` it was logged in `dmesg`, however when using my `gpiochip2` no message could be seen.

`linehandle_create()` might not get to the `get_unused_fd_flags()` call though...
```c
                /*
		 * Lines have to be requested explicitly for input
		 * or output, else the line will be treated "as is".
		 */
		if (lflags & GPIOHANDLE_REQUEST_OUTPUT) {
			int val = !!handlereq.default_values[i];

			ret = gpiod_direction_output(desc, val);
			if (ret)
				goto out_free_lh;
		} else if (lflags & GPIOHANDLE_REQUEST_INPUT) {
			ret = gpiod_direction_input(desc);
			if (ret)
				goto out_free_lh;
		}
```
 
These gpiod calls are forwarded to my `gpiochip`... and it ends up I was incorrectly returning a positive value in it. I'm using I2C transfers, and the return of the function is how many transfers I've done so by mistake the code looked like this:

```c
        struct pi4io16_data *pi4io16;
	int ret;
	u16 word;

	pi4io16 = gpiochip_get_data(chip);

	ret = pi4io16_i2c_read_word(pi4io16->client,
				    PI4IO16_REG_CONFIGURATION_0, &word);
	if (ret < 0) {
		dev_err(&pi4io16->client->dev,
			"pi4io16_i2c_read_word() failed: %i", ret);
		return ret;
	}

	if (direction == GPIO_LINE_DIRECTION_IN) {
		word |= BIT(offset);
	} else {
		word &= ~BIT(offset);
	}

	ret = pi4io16_i2c_write_word(pi4io16->client,
				     PI4IO16_REG_CONFIGURATION_0, word);
	if (ret < 0) {
		dev_err(&pi4io16->client->dev,
			"pi4io16_i2c_write_word() failed: %i", ret);
		return ret;
	}

	return ret;
```

When `pi4io16_i2c_write_word()` succeeds, it's return value will be greater than 0 and when this value is propagated to `linehandle_create()`  it causes a jump to `out_free_lh`  label because it uses `if (ret)`, not `if (ret < 0)`, which returns the positive value from the IOCTL. This next part is a mix of stuff because I have the source for the libgpiod v2, but in `lib/gpiod.c:gpiod_chip_request_lines` the IOCTL return code is only checked as whether it's negative:
```c
 ret = ioctl(chip->fd, GPIO_V2_GET_LINE_IOCTL, &uapi_req);
        if (ret < 0)
                return NULL;
```

Since the IOCTL failed but returned 1, the library will then think everything is ok and use the previously zeroed `fd` field of the structure and try to perform the next IOCTL on `stdin`

I went ahead and fixed the functions on my `gpiochip` so that they return proper 0 on success, and now everything seems to be working.

By taking a look at `struct gpio_chip`'s description I can see that the function pointers `direction_input` and `direction_output` have the following description:
```
 * @direction_input: configures signal "offset" as input, or returns error
 *	This can be omitted on input-only or output-only gpio chips.
 * @direction_output: configures signal "offset" as output, or returns error
 *	This can be omitted on input-only or output-only gpio chips.
```

A few other functions have it written `negative error`, so I didn't notice that here it's only `error`. It might be a good idea to specify that a non-zero value is an error in this case? Or maybe the return code verification on `gpiolib-cdev` should be `if (ret < 0)` instead of `if (ret)`? I kinda like the former better because I don't think the IOCTL should allow the positive value return in this case.

I also didn't look `GPIO_V2_GET_LINE_IOCTL` on the kernel side, but the positive return value I was mistakenly returning on my gpiochip functions were  causing it to return a positive value from the IOCTL as well.

brgl, 2024-01-24:
Thanks for the detailed investigation and report. It's a bug in core GPIO code in the kernel where we don't sanitize (or enforce) return values for GPIO provider callbacks. This will need to be addressed. But since older kernels are also out there, we need to fix that in libgpiod too. I will Cc you on the patches.

JoseGuilhermeCR, 2024-01-24:
I'm glad I could be of help to the kernel for once! Thank you for your attention, I'll be reading the patches so that I'm aware of updates.

JoseGuilhermeCR, 2024-03-15:
I guess I can consider this as closed!

closed 2024-03-15 by JoseGuilhermeCR