brgl/libgpiod #50: libgpiod-dev v2.1 is missing

DonavanMartin, 2024-01-29:
Hi,

I'm attempting to update my C++ code to align with the new version of libgpiod (> 2.0). However, I'm facing difficulties installing libgpiod-dev using the provided command:

```
sudo apt-get install libgpiod-dev
```

The package available at https://packages.debian.org/unstable/libgpiod-dev is currently at version 1.6.3-1+b3.

Is it feasible to include a new package for version 2.0 or later?

Best regards.

Ablu, 2024-01-29:
debian seems to have a request for it already: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1055129

DonavanMartin, 2024-01-29:
This issue related to  [#1055129](mailto:1055129@bugs.debian.org)
libgpiod: Please upload v2.x to Debian

(https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1055129)

closed 2024-01-29 by DonavanMartin