brgl/libgpiod #53: autoreconf error, 2.69

JohnWulff, 2024-02-18:
**I have cloned libgpiod from this web site (brgl/libgpiod). I am having problems building libgpiod.** 
I am running the software on a Raspberry PI 3B+ using Raspbian Buster OS. 
I have successfully used **sysfs** in the run-time of my event-drive language [immediate C](https://www.immediatec.net/) for over 8 years to set GPIO outputs and generate interrupts from GPIO inputs and read them. Since sysfs is deprecated I want to migrate my code to libgpiod.

I installed the following:
`apt install autoconf`
`apt install autoconf-archive`

I initially executed:
`sudo apt install gpiod libgpiod-dev libgpiod-doc`
Trying to compile examples from that install showed that the gpiod.. C functions were not present. I then cloned libgpiod.
Following the Build instructions I executed:
`./autogen.sh --enable-tools=yes --prefix=/usr/local`
This output the following error messages:
`autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
bindings/cxx/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
bindings/cxx/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
bindings/cxx/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
bindings/cxx/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
bindings/cxx/Makefile.am:4:   its definition is in aclocal's search path.
lib/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
lib/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
lib/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
lib/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
lib/Makefile.am:4:   its definition is in aclocal's search path.
tests/gpiosim/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
tests/gpiosim/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
tests/gpiosim/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
tests/gpiosim/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
tests/gpiosim/Makefile.am:4:   its definition is in aclocal's search path.
tools/Makefile.am:7: error: Libtool library used but 'LIBTOOL' is undefined
tools/Makefile.am:7:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
tools/Makefile.am:7:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
tools/Makefile.am:7:   If 'LT_INIT' is in 'configure.ac', make sure
tools/Makefile.am:7:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1`

I do not know to what path or where exactly to set LT_INIT

Ablu, 2024-02-18:
> autoreconf: configure.ac: not using Libtool

Could you check whether you have `libtool` installed?

JohnWulff, 2024-02-18:
No libtool found.
I have now done
`sudo apt install libtool`
This adds the directory `/usr/share/libtool`

`./autogen.sh --enable-tools=yes --prefix=/usr/local` seems to be working except for the following warning:
`autoreconf: running: aclocal --force -I m4`
`aclocal: warning: couldn't open directory 'm4': No such file or directory`

make OK
Vielen Dank, John

JohnWulff, 2024-02-18:
`make install` fails because it requires super-user privileges.
`sudo make install` OK

You may want to modify the Build instructions in the README.

closed 2024-10-11 by brgl