brgl/libgpiod #54: [python] LineRequest.reconfigure_lines problems

dex6, 2024-02-22:
Hello,

HW: RPi 3B, OS: ArchLinux, Kernel: 6.1.69-1-rpi-ARCH, Python: 3.11, gpiod: 2.1.3 from pypi.org

I'm porting some of my python scripts to libgpiod and I've run into a problem with request.reconfigure_lines method. Instead of succeeding, it raises ValueError with `(22, 'Invalid argument')`.

The script basically opens 7 lines (some inputs, some outputs) in one request, does some work, and at the end wants to switch all output lines back to inputs so HW stops driving them. Both config dictionaries (first for request_lines, second for reconfigure_lines) are constructed dynamically from different metadata, and what I believe to be important, are likely to be in different order. When I pass the same dict to both methods, there's no exception (but obviously nothing changes).

Toy reproducer attached - it works reliably cause nowadays python dicts remember insertion order. Before Python 3.6 (or 3.7?) it would fail randomly. I've also included a third dict in the reproducer - with just one line to be reconfigured - this also fails, which I find counterintuitive, cause other functions like get/set values can operate on a subset of requested lines.

Can this be fixed inside gpiod or is the user expected to take care of sorting config dicts somehow?

[test.py.txt](https://github.com/brgl/libgpiod/files/14379743/test.py.txt)

brgl, 2024-02-23:
Hi! Can you try the following patch?

```
diff --git a/gpiod/chip.py b/gpiod/chip.py
index b3d8e61..e2b7f2a 100644
--- a/gpiod/chip.py
+++ b/gpiod/chip.py
@@ -260,7 +260,7 @@ class Chip:
                     lambda t: [
                         j for i in (t) for j in (i if isinstance(i, tuple) else (i,))
                     ]
-                )(tuple(config.keys()))
+                )(tuple(sorted(config.keys())))
             ]
         ).items():
             if count != 1:
diff --git a/gpiod/line_request.py b/gpiod/line_request.py
index cde298f..afcde14 100644
--- a/gpiod/line_request.py
+++ b/gpiod/line_request.py
@@ -158,7 +158,7 @@ class LineRequest:
 
         line_cfg = _ext.LineConfig()
 
-        for lines, settings in config.items():
+        for lines, settings in sorted(config.items()):
             if isinstance(lines, int) or isinstance(lines, str):
                 lines = [lines]
```

dex6, 2024-02-23:
Yeah, that patch does not make sense; and it makes things even worse. In my case 4 lines initially have identical configuration, so my request dict actually mixes integer and tuple keys. But unlike py2, python 3 loudly refuses to sort objects of different types... So the only way would probably require sorting resulting _ext.LineConfig that goes to the C library/kernel, but that probably will break other things. Agreed improving docs is best what can be done in gpiod here.

I'll switch my script to integer-keys-only dicts, sorted before passing to gpiod. Hmm... Leaving original request_lines config dict around, and then updating it in-place before passing again to reconfigure_lines, also seems like a possible option.

Thanks for quick help and your work on the gpiod library!

warthog618, 2024-06-18:
As I see it, the root cause is that gpiod_line_request_reconfigure_lines() in the C library requires the configuration to contain the offsets in the same order as they were requested.  That is not strictly necessary - while the configuration must contain the same set of offsets, the order is not relevant at this layer - libgpiod can readily reorder to match the request order during the translation to the uAPI call.

I'm putting together a patch series to address this and will submit it through the list shortly.

warthog618, 2024-06-19:
Ugggh, as I'm writing the cover letter for that patch series I realise the approach I used is fatally flawed.  I had assumed  the global output values in the config are in the same order the lines are added to the configuration, and performed the sort on the translated uAPI struct, but I find this in the documentation for gpiod_line_config_set_output_values() that populates them:

```
 * Each value must be associated with the line identified by the corresponding
 * entry in the offset array filled by
 * ::gpiod_line_request_get_requested_offsets.
```

which means the sort would need to be performed before those globals are applied in gpiod_line_config_to_uapi().

Worse still, gpiod_line_config_get_line_settings() conflicts with allowing misordering, as it assumes the line configs are in request order, and there is no request offset ordering available to it, so configs having to be in request order is effectively baked into the C API.  Yay.

Without breaking the C API, a solution would have to be in the Python bindings themselves.

brgl, 2024-06-20:
Is this a problem though? It does what the documentation says it does. I'm not against doing it in Python but I'm wondering if this is a bug or a feature.

warthog618, 2024-06-20:
Only in that fixing it in the C library seemed more correct and more generally useful.  I just can't work out how to do that given the gpiod_line_config_set_output_values() helper.

Strictly speaking, I'm not sure that gpiod_line_config_get_line_settings() does what it says - if you have added the lines in an order other than the request order and use gpiod_line_config_set_output_values() as described then it can return the wrong output value.  There is an implicit assumption in the implementation that the line_configs are in request order.

brgl, 2024-06-20:
It's not that implicit though - if you try to use this config to reconfigure the request, it will fail. Sounds quite explicit to me. :)

warthog618, 2024-06-20:
Ok, looking at fixing it in the Python bindings, and making sure the fix complies with the API docs, I read:

```
        Reconfigure requested lines.

        Args:
          config
            Dictionary mapping offsets or names (or tuples thereof) to
            LineSettings. If None is passed as the value of the mapping,
            default settings are used.
```

So I add a test for the None case.  But the existing bindings fails with:
```
ERROR: test_reconfigure_with_default (tests.tests_line_request.ReconfigureRequestedLines.test_reconfigure_with_default)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/libgpiod/bindings/python/tests/tests_line_request.py", line 514, in test_reconfigure_with_default
    self.req.reconfigure_lines({
  File "/home/pi/libgpiod/bindings/python/gpiod/line_request.py", line 170, in reconfigure_lines
    line_cfg.add_line_settings(offsets, _line_settings_to_ext(settings))
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/libgpiod/bindings/python/gpiod/line_settings.py", line 56, in _line_settings_to_ext
    direction=settings.direction.value,
              ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'direction'
```
So, just to confirm, that is another bug and it should use a default LineSettings if None is passed?
Should it also do that for lines with no config specified at all?  That would handle the third case in the OP's example.

brgl, 2024-06-20:
Yeah, I can confirm, looks like a bug, especially when `chip.request_lines()` works fine with None.

warthog618, 2024-06-20:
What about the second point - what if no config is provided for a line - treat that like None?  Or require that all lines have config?

brgl, 2024-06-20:
You mean something like `req.reconfigure_lines(config=None)`? That doesn't make any sense though, right?

warthog618, 2024-06-20:
No, what I mean is what if there is no key in the config dict for a particular line.  In the C API we require all lines to be configured.  The case above is config={X: None, ...}.  So what of there is no entry in the config dict for a particular line? i.e. an implicit None.

brgl, 2024-06-20:
I see. This is python so I wouldn't be too strict. I would even go as far as allowing `req.reconfigure_lines(config=gpiod.LineSettings())` where in case of not specifying the offsets, we'd just reconfigure all lines in this request. Would that work or am I not seeing something?

warthog618, 2024-06-20:
That would be an extension to the existing config definition, which I quoted above, and corresponding reconfigure_lines() signature - it is supposed to be a dict from tuples of line names/offsets to config.

If you want to handle that case then use an empty tuple for the key, i.e.
`req.reconfigure_lines(config={(): gpiod.LineSettings(...), ...})`

You could use the empty tuple config as the "default config" for all the unspecified lines.  In your example that would be all lines.

brgl, 2024-06-20:
> That would be an extension to the existing config definition, which I quoted above, and corresponding reconfigure_lines() signature - it is supposed to be a dict from tuples of line names/offsets to config.
> 

That's alright, it doesn't break existing users.

> If you want to handle that case then use an empty tuple for the key, i.e. `req.reconfigure_lines(config={(): gpiod.LineSettings(...), ...})`
> 
> You could use the empty tuple config as the "default config" for all the unspecified lines. In your example that would be all lines.
> 
> (btw, ellipsis used in examples here mean "more stuff to be optionally filled in", not a literal Python ellipsis)
> 
> On second thoughts, a None key for the default might be safer - in case sets of lines are being collected programatically - you don't want an accidental empty set overriding the default. So I revise my suggestion to `req.reconfigure_lines(config={None: gpiod.LineSettings(...), ...})`.

I'm even fine with both. It's Python. Make it do anything logical in most reasonable cases.

warthog618, 2024-06-21:
I'm not suggesting adding that as part of this fix - being able to override the default config is a step too far for this issue IMHO, but I need to know how to handle the missing offset case as it is one of the paths this fix can't avoid due to the sorting.
I'll send in a series with what I've got - that fixes all the cases the OP asked for.

We can always add overriding the default later - let the idea stew for a while.

Oh, and I'm thinking that the kernel should ignore SET_CONFIG on lines that don't have a direction set.  i.e. that are being configured "as-is" with the default config.  Some of the flags don't have "as-is" states (active_low, drive, event_clock, edge_detection), so they get reset to default (though we don't change any hardware config). Even for bias, which does have an "as-is" state, we lose knowledge of what it was set to - even if the hardware doesn't get reconfigured.  Debounce avoids the whole mess as it already doesn't get reconfigured unless direction is set.
But the flags could end up being set unexpectedly, including inverting the polarity of an output (on the next SET_VALUES), which seems bad.
While the edge detection flags can get reset, the edge detector itself is only reconfigured if direction is set, so that's not good either.
I don't see a case where a user would want to do any of this intentionally.
All up it seems best to not alter ANY config unless a direction is set.
Does that make sense to you?

brgl, 2024-06-21:
> I'm not suggesting adding that as part of this fix - being able to override the default config is a step too far for this issue IMHO, but I need to know how to handle the missing offset case as it [...]

Sounds good.

> Oh, and I'm thinking that the kernel should ignore SET_CONFIG on lines that don't have a direction set. i.e. that are being configured "as-is" with the default config. Some of the flags don't have "as-is" states (active_low, drive, event_clock, 

Wait, the kernel doesn't really know the "as-is" setting - the line is *always* an input or output already.

edge_detection), so they get reset to default (though we don't change any hardware config). Even for bias, which does have an "as-is" state, we lose knowledge of what it was set to - even if the hardware doesn't get reconfigured. Debounce avoids the whole mess as it already doesn't get reconfigured unless direction is set. But the flags could end up being set unexpectedly, including inverting the polarity of an output (on the next SET_VALUES), which seems bad. While the edge detection flags can get reset, the edge detector itself is only reconfigured if direction is set, so that's not good either. I don't see a case where a user would want to do any of this intentionally. All up it seems best to not alter ANY config unless a direction is set. Does that make sense to you?

Doesn't that change the current behavior though? And what would happen to a line that you request "as-is" from user-space? It ends up requested but none of the config is actually applied? Is there any use for that? (Not that there's any actual use for 'as-is' though outside of some kind of early user-space hogging).

warthog618, 2024-06-22:
The kernel has a definite input or output, but a line can be requested with direction as-is too.  In that case only the active_low flag can be set.  You can use it to peek at the existing value of an output line, driver willing.  How the request/reconfigure is treated is dependent on the flags being requested, not the existing state.

But what I'm referring to here is strictly the reconfigure, i.e. linereq_set_config() and linehandle_set_config().  At the moment a line can also be reconfigured with direction as-is, and that can do odd things to the other flags, particularly if it was requested (not reconfigured) as input or output.
I'm not suggesting any change to the request behaviour.  I'm pretty certain this would be fixing reconfigure behaviour that no one realised is there.  Even more certain now.

closed 2024-07-08 by brgl