brgl/libgpiod #60: Edge event timestamp latency/jitter/accuracy

oysstu, 2024-02-29:
I've looked around for any tests or experiences related to the latency, jitter, and accuracy related to the edge event timstamps in libgpiod, but have been unable to find anything. How does it compare to the timestamps set by the pps-gpio kernel module, which is based around interrupts?

Another point of comparison is the [pps-gpio-poll](https://github.com/mlichvar/pps-gpio-poll) out-of-tree kernel module. It provides a PPS device based on busy-waiting and polling of the GPIO. According to the author, which is the maintainer of chrony, this approach is more accurate and have less jitter than the interrupt-based pps-gpio (albeit with more CPU usage).

Any insight is appreciated, even if not backed by real data.

warthog618, 2024-03-10:
The edge events are timestamped in the kernel irq handler, so very similar to `pps-gpio`.

The resulting latency and jitter depends on your platform and what is running on it at the time, so there is no substitute for testing it yourself.  Not sure what you mean by "accuracy".

oysstu, 2024-03-11:
I see, thanks for the input. As you point out, there's too many variables to make absolute claims without testing (resolution of a jiffie, the underlying clock, PREEMPT_RT etc). I was interested in the conceptual difference between the different approaches (e,g, interrupt/polling). I see now that both the pps-gpio driver and libgpiod interfaces the linux irq handler.

I suppose this behavior can also change depending on the underlying gpio driver. I.e., how/if the underlying gpio chip implements interrupts.

closed 2024-03-12 by oysstu