brgl/libgpiod #63: Functions missing from the doxygen documentation

Shulyaka, 2024-03-10:
The documentation is very incomplete. Not only it lacks some quick start guide (e.g. get the chip - get the line info - create a line request, etc) or a migration guide from 1.x versions, but also some of the functions are missing. For the example, check the [Line info section](https://libgpiod.readthedocs.io/en/latest/group__line__info.html). There are number of function listed that operate on `struct gpiod_line_info *`, but not a single function to obtain this struct. The header file has the [gpiod_chip_get_line_info](https://github.com/brgl/libgpiod/blob/6d5d8afc9c7e9a7b47d54fe51f4cb8d7077cdb8b/include/gpiod.h#L209) function but it is not documented.

As a result, it is virtually impossible to deduce from the documentation how the API should be used.

brgl, 2024-03-10:
Yeah, it's unfortunate that a project I'm developing in my free time and which made me so far precisely 0€ does not offer documentation that's on par with commercial software. Feel free to improve the docs and send the patches to the linux-gpio mailing list, I'll happily accept them.

> The header file has the [gpiod_chip_get_line_info](https://github.com/brgl/libgpiod/blob/6d5d8afc9c7e9a7b47d54fe51f4cb8d7077cdb8b/include/gpiod.h#L209) function but it is not documented.

Your very link points to the documentation, I'm not sure what you mean.

Shulyaka, 2024-03-10:
> Yeah, it's unfortunate that a project I'm developing in my free time and which made me so far precisely 0€ does not offer documentation that's on par with commercial software.

Sorry for the miscommunication, I didn't mean to offense, it's a great project. But there is a room for improvement, so I opened the issue to help with that.

> Your very link points to the documentation, I'm not sure what you mean.

Not exactly. My link points to the header file, but the function is missing in the doxygen documentation.

warthog618, 2024-03-11:
> > Your very link points to the documentation, I'm not sure what you mean.
> 
> Not exactly. My link points to the header file, but the function is missing in the doxygen documentation.

Note sure what you mean - it is there in the [doxygen documentation](https://libgpiod.readthedocs.io/en/latest/group__chips.html).
It may be brief, and perhaps you can suggest how to expand it or even provide a patch, but it is there.
Btw, there are also [examples](https://github.com/brgl/libgpiod/tree/master/examples) provided for many common use cases, including getting line info.