brgl/libgpiod #67: gpiod_line_request_release --> application hangs

schoyswr, 2024-03-26:
Hi,

I followed the example `toggle_line_value.c`.  
Basically, the example works, but it must be mentioned that the method `gpiod_line_request_release()` is never called, because the application can only be exited by Ctrl+C.

My own application, based on the example, crashes, as soon as I call `gpiod_line_request_release()`. I just want to toggle a power supply pin at the beginning and want to do some further stuff afterwards. 
Can someone review or adapt the mentioned example, so that also the `gpiod_line_request_release()` is called?
That would be great!

brgl, 2024-03-26:
I need more info on *why* it crashes. Would you mind sharing a gdb stack trace?

warthog618, 2024-03-27:
Updating `toggle_line_value.c` such that the for loop exits after a number of cycles works fine for me:

```
	int i;
	for (i = 0; i < 5; i++) {
...
```

Have you tried anything similar yourself?  Please provide example code that is failing for you.

rsm47-aut, 2024-03-27:
Sorry,
I found the issue - caused by myself. Everything is fine with `gpiod_line_request_release()`.  
Sorry for that!

closed 2024-03-27 by brgl