brgl/libgpiod #69: Obsolete libgpiod package on PyPI

warthog618, 2024-03-27:
There is an old **libgpiod** package on PyPI, currently at 2.0.1, which is confusing since the official package name is now **gpiod** and is at 2.1.3.
Can the old package be removed, or redirected either explicitly or via an updated README to **gpiod**?

brgl, 2024-03-27:
I removed it a bit too hastily. Turns out the nanbield branch of yocto still uses it. I cannot undo it either. I need to patch meta-openembedded now. :(

warthog618, 2024-10-11:
Not sure where you are at with the yocto build, but it is gone from PyPI, so I'm satisfied this issue is resolved.

closed 2024-10-11 by warthog618

brgl, 2024-10-11:
All is good: `kirkstone` is still on libgpiod v1.6.x and the oldest still supported release `scarthgap` pulls the "correct" `gpiod` package from pypi.