brgl/libgpiod #72: C++ bindings under LGPL-3.0

wlozano0collabora, 2024-04-10:
I understand that in https://github.com/brgl/libgpiod/commit/ea84f882d5d39b734a5756b3a6950da4896b9a0a the C++ bindings were relicensed to be LGPL-3 to overcome some difficulties create by projects that use templates, default implementations, etc.

However, this change can cause issues for projects which have strong license restrictions, such as avoid LGPL-3.0 code, which were relying on these bindings.

I understand that a better solution would be to have these bindings under a dual license like LGPL-2.1 or LGPL-3.0 and let each project decide which license to use.

brgl, 2024-04-10:
Can you educate me on why someone would avoid LGPL-3.0 code?

wlozano0collabora, 2024-04-10:
There are projects which have strict license restrictions, for instance [Apertis](https://www.apertis.org) which is an open source distribution based on Debian has the following [license expectations](https://www.apertis.org/policies/license-expectations/).

From the website

[Version 3 of the GPL license](https://spdx.org/licenses/GPL-3.0-or-later.html) was created to address the concern of users who were prevented from running modified code on their device, when the device was shipped with open source software. A common method for preventing users to run their own code is by using signature verification. This practice is known as [Tivoization](https://en.wikipedia.org/wiki/Tivoization). Those licensing rules are a constraint because in some application domains, it is a regulatory (or safety) requirement to ensure that the hardware runs verified software.

brgl, 2024-04-10:
But this doesn't really answer my question, does it? Restricting companies from being able to block users from running modified versions of open-source software sounds like a good thing, no?

wlozano0collabora, 2024-04-10:
Yes, in general, I would agree with your statement that "restricting companies from being able to block users from running modified versions of open-source software sounds like a good thing". However, there are some specific areas where there are regulations for safety, like medical devices, cars, etc which enforce these companies to give confidence to the end user what the software running is reliable.

From my understanding, since the majority of the project is licensed under LGPL-2.1, same as previous version of C++ bindings, it makes sense to provide both LGPL-2.1 and LGPL-3.0 as options, to avoid breaking license compatibility, which I consider is a win-win solution.

closed 2024-04-22 by brgl

wlozano0collabora, 2024-04-22:
Thank you for the fix in such a short time!

wlozano0collabora, 2024-04-25:
Additional question, can we consider that this re-licensing is "retroactive", in the sense that we can say that libgpiod 2.1 (already released) can be considered LGPL-2.1-or-later?
If you agree, that will allow me to backport the patch with the relicensing with a reference to your statement.

Thanks in advance!

wlozano0collabora, 2024-04-25:
Thank you very much for your promt reply!

ceggers-arri, 2024-04-29:
Hi Bart,

thanks for relicensing under LGPL-2.1.

> But this doesn't really answer my question, does it? Restricting companies from being able to block users from running modified versions of open-source software sounds like a good thing, no?

Companies have different reasons (regulatory, product safety, protection against plagiarism) for not allowing users to install modified versions of software on embedded devices. This is something the GPLv3 is fighting against, but for what a price? In most cases manufacturers of commercial products cannot change their policy, so they have to switch to alternative software (e.g. busybox vs. toybox) instead or stick at older versions (prior relicensing, see libsigc++ [1] for example).

That being said, I would be happy seeing the next v2.1.x release in OpenEmbedded Scarthgap.

[1] https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-core/libsigc++-2.0