brgl/libgpiod #74: How to set/toggle GPIO to input or output?

TheRaven500, 2024-05-10:
It is more a question than an issue:
Is there a way to toggle the GPIO as output/input?
If yes: How?
Sorry for this maybe dumb question, but have nothing found in the documentation.

brgl, 2024-05-10:
How exactly did you go through the docs? What did you look at? Because if this is not covered in a user-friendly way, then I'd like to fix it.

TheRaven500, 2024-05-10:
Thank you for the very fast answer! The first thing i have tried is "gpioset --help" to see what options are available. Then i checked the "README" on the main page. After that i searched trough the internet. But i found no command to set GPIO as a output or input. During "try and error" i managed to set the GPIO to output. But i cant remember how.

brgl, 2024-05-10:
So the help output for gpioset says:

```
The line output state is maintained until the process exits, but after that is not guaranteed.
```

Which means: running gpioset sets line to output. Once gpioset exits, the direction (nor value for that matter) is guaranteed. I'm not sure how we can better improve it, do you have an idea?

TheRaven500, 2024-05-10:
Ah ok, now it is a bit more clear to me. The thing is that i worked before with sysfs. There you could do something like "echo "out" > /sys/class/gpio/gpioXX/direction" and the GPIO was set to out (or "in" for input). And after that you could read the value of that GPIO with "cat /sys/class/gpio/gpioXX/value". I used this for many bash scripts. So because of that i thought it is also possible with "gpioset" to switch the status of the GPIO (to in or out) and after that read it with "gpioget". Is this possible with libgpiod? How to set a GPIO until next reboot for example?

brgl, 2024-05-10:
This is not possible not because of libgpiod but because how the kernel character device ABI works. However I have recently posted a series of patches adding a DBus API to the project, this should have what you need: https://lore.kernel.org/linux-gpio/20240412122804.109323-1-brgl@bgdev.pl/

TheRaven500, 2024-05-10:
Ok, strange. Because it looks like it is working with raspi-gpio or the new pinctrl tool. But i am only a user. Not sure if we talk about the same thing.

pelwell, 2024-05-11:
Both raspi-gpio and pinctrl look directly add the hardware from userspace - they don't play by the libgpiod rules.

The way to make a GPIO an input, using the gpiod utilities, is `gpioget`. Using the v1.x libgpiod and tools (which is current in Bookworm), there is no way to query the level of a GPIO without making it an input. With v2.x you can pass the `--as-is` parameter to stop it from being changed from an output.

In the Raspberry Pi version of Linux we deliberately disable the returning of GPIOs to inputs when gpioset exits. This differs from upstream Linux, but both versions have (or will soon have) a module parameter to reverse the behaviour.

warthog618, 2024-05-13:
@pelwell On that subject, have you guys made any progress on packaging **libgpiod v2** for Raspberry Pi OS?

pelwell, 2024-05-13:
@XECdesign can give you a more authoritative answer, but unless there is a good reason to diverge we ship the upstream Debian package. Sid (Debian's development branch) is still on v1.6.3 (https://packages.debian.org/sid/main/gpiod), so that is what we are likely to be on for a while unless there are specific requirements to change that.

warthog618, 2024-05-13:
In [this thread](https://forums.raspberrypi.com/viewtopic.php?t=361735&hilit=quality+of+libgpiod#p2170681) @6by9 indicated that @XECDesign was indeed looking into that, so I'm wondering if that is going anywhere.

The issue is that **v2** didn't make Bookworm, and so wont go into Debian for years.  In the meantime there is confusion amongst Pi users as to which **libgpiod** version they should use, and every time I suggest Pi users should be moving to **v2** which is better but has a very different API, and **v1** is a dead-end and you don't want to have to migrate twice, I get the "but it isn't packaged" push-back.

XECDesign, 2024-05-13:
Updating the existing package would break packages in the repo which were built for 1.6.3.

There is "right way" to do the update that ensures v2 is available and older software still works, but that takes a little more effort. Last time I looked at it, the maintainer of the package of the package was starting to do the work and now they have an experimental 2.1 package.

I'd prefer to wait until it makes it at least as far as 'testing' and then backport it to bookworm then.

warthog618, 2024-05-13:
I always expected this to require new packages.  Sorry, I thought that was obvious.

Waiting for 'testing' is fair enough.  Any idea how far away that would be?

brgl, 2024-05-13:
@XECDesign out of curiosity: which packages in debian are linked against libgpiod?

mhei, 2024-05-13:
We have a similar chicken-egg problem in OpenWrt (see https://github.com/openwrt/packages/pull/24123)
And FYI: work for openocd is ongoing, see also [my comment here](https://github.com/brgl/libgpiod/discussions/73#discussioncomment-9244021).

closed 2024-10-11 by brgl

pelwell, 2024-10-11:
FYI From above: 
> In the Raspberry Pi version of Linux we deliberately disable the returning of GPIOs to inputs when gpioset exits. This differs from upstream Linux, but both versions have (or will soon have) a module parameter to reverse the behaviour.

In the Raspberry Pi downstream kernel, maintaining the previous value is the default (`dtparam -h strict_gpiod`), while in upstream kernels, reverting to inputs is the default.