brgl/libgpiod #75: Increase GPIO read/write frequency

MosiQt, 2024-05-14:
Hello, I'm utilizing the gpiod library on my Yocto Linux system, specifically on an i.MX8MM board. After measuring the read/write frequency speed in a simple code, I've observed it to be 500Hz (2ms). However, for my device, I require higher rates. Are there any configuration options available to achieve this, or perhaps has this limitation been addressed in newer versions of the library?

[image]

MosiQt, 2024-05-14:
This is the simple code I ran the tests:

[image]

```

int main(int argc, char** argv)
{
    const std::string chipName = "gpiochip5";

    gpiod_chip* chip;

    if (chip = gpiod_chip_open_by_name(chipName.c_str()); !chip) {
        std::cerr << "ERROR: No chip";
        return 1;
    }

    std::vector<gpiod_line*> gpLines;
    gpiod_line* line;

    if (line = gpiod_chip_get_line(chip, 10); !line) {
        std::cerr << "ERROR: No line";
        release(chip, nullptr);
        return 1;
    }

    if (gpiod_line_request_output(line, "Consumer", 1) < 0) {
        std::cerr << "ERROR: Request failed";
        release(chip, line);
        return 1;
    }

    int value = 0;
    for (int i = 0; i < 20000; ++i) {

        gpiod_line_set_value(line, value);
        value = !value;
    }
    release(chip, line);
    std::cout << "Finished" << std::endl;

    return 0;
}
```

brgl, 2024-05-14:
The code looks right so I suppose this is simply what the combination of the user-space to kernel context switching plus going through several abstraction layers from your program, through the character device, GPIOLIB and finally the low-level GPIO driver results in. You may want to write a kernel driver for whatever you're trying to do and disable preemption for duration of GPIO banging but I don't know enough to be able to help you precisely.