brgl/libgpiod #81: Operation not permitted for set_value()

ActiveTutorial, 2024-06-10:
For this script:
                                                                     
import gpiod

chipname = "gpiochip0"
line_offset_16 = 16
line_offset_26 = 26

chip = gpiod.Chip(chipname)

line_16 = chip.get_line(line_offset_16)

line_26 = chip.get_line(line_offset_26)

try:
    events = line_26.request(consumer="test_continuity", type=gpiod.LINE_REQ_EV_BOTH_EDGES)
    line_16.set_value(1)

    # Wait a moment for the signal to stabilize    time.sleep(0.1)

    # Check if pin 26 reflects the state of pin 16
    state = line_26.get_value()
    if state == 1:
        print("Continuity detected between pin 16 and pin 26.")
    else:
        print("No continuity detected between pin 16 and pin 26.")

finally:
    # Clean up
    line_16.release()
    line_26.release()
    chip.close()

I get this error:
~ $ sudo python3 contentest.py 
Traceback (most recent call last):
  File "/home/active/contentest.py", line 21, in <module>
    line_16.set_value(1)
PermissionError: [Errno 1] Operation not permitted

brgl, 2024-06-10:
Can you run this script through strace like: `strace -f -o/tmp/output.txt <your script>` and attached the output here?

warthog618, 2024-06-15:
You never request line_16, only line_26.

Add 
```python
line_16.request(consumer="test continuity", type=gpiod.LINE_REQ_DIR_OUT, default_val=0)
```
before trying to set the value.  Note that the line will be set to the default_val (0 in my example) when requested.

Admittedly the error could be better - "PermissionError" when trying to use an unrequested line isn't very helpful.

closed 2024-10-11 by brgl

ActiveTutorial, 2024-10-12:
I totally forgot about me posting about this. @brgl sorry for ignoring you and @warthog618 thanks for the solution.