brgl/libgpiod #83: libgpiod seems to miss edge interrupts

rweickelt, 2024-06-14:
I am using libgpiod 1.6.2 on a RPi 3B with Raspbian Bullseye. I have implemented an interrupt monitor strictly following [this](https://github.com/brgl/libgpiod/blob/v1.6.x/bindings/cxx/examples/gpiomoncxx.cpp) example (triggers on falling edges). Each interrupt is acknowledged by the host before the peripheral can generate a new one. The interrupt frequency is about 100-200 Hz.

My program usually runs for half a minute, sometimes longer. At some point it stops because libgpiod hangs at `gpiod::line::event_wait()` and doesn't recognize the interrupt. I am surprised that this is even possible. How can the interrupt be lost?

I am tempted to implement a simple polling loop instead.

brgl, 2024-06-14:
What do you mean by "doesn't recognize the interrupt"? The library is just a wrapper around the kernel uAPI so it's most likely the linux driver that for some reason doesn't detect the interrupt. But once it stops, no more interrupts are triggered, is that what you mean?

rweickelt, 2024-06-14:
Thanks for the quick reply. 

> What do you mean by "doesn't recognize the interrupt"?

I can see an edge on the GPIO line that should have triggered an event, but I don't get the event in software. I don't know anything about the underlying kernel uAPI and how edge detection is implemented (on the RPi). Is it done in hardware or in software? If it's done in software, then I see how we could miss an edge.

brgl, 2024-06-14:
I don't know, I am not familiar with every single GPIO driver in the kernel. Do you see any warnings in the kernel log?

rweickelt, 2024-06-14:
I switched to the CAPI now. Same behavior.

Maybe I am using libgpiod the wrong way? My program waits on interrupts on 2 different IO pins concurrently in 2 different threads. Is that valid?

brgl, 2024-06-14:
Do you requested them separately?

rweickelt, 2024-06-14:
Yes, everything is completely separated. No objects are shared. 

The kernel log does not contain any warnings btw.

warthog618, 2024-06-15:
You also asked this question on[ Stack Overflow](https://stackoverflow.com/questions/78620397/can-libgpiod-lose-edge-events-on-a-raspberry-pi).  Might've been nice to reference that.

> Each interrupt is acknowledged by the host before the peripheral can generate a new one.

What you are seeing is consistent with the interrupt not being acked at the peripheral, no?
So, how is that done?  How can you be sure that is working correctly?

rweickelt, 2024-06-15:
When the host sees an event (falling edge on IO line), it acknowledges the interrupt with a SPI message. After the message, the peripheral will deassert the IO line and produce a new falling edge after some time. That’s IMO safe and it works well when using a FT4222 as interface or when the host is a bare metal program on a microcontroller. It’s just an RPi that acts up. 

 I’ll check if I can reproduce the problem with a single peripheral as well. So far I’ve only been using two peripherals concurrently. However, it’s probably not a libgpiod problem.

warthog618, 2024-06-16:
It probably isn't a libgpiod problem, but hopefully we can still help you track it down.

The fact that the SPI and interrupt handshaking works on other platforms doesn't prove anything.  New platform, new problems.

To prove it is an interrupt problem with the RPi, you need to monitor the SPI and interrupt line and demonstrate that the interrupt is deasserted, then re-asserted but you don't get an event from libgpiod.  A scope on the SPI clock line and the interrupt line would probably be sufficient to show that.  As I stated earlier, what you are seeing is consistent with the SPI ack failing and, especially given you are driving two concurrently (hopefully on separate buses??), that would seem to me to be the more likely culprit and I would be wanting to eliminate that as a possibility.

rweickelt, 2024-06-17:
Here's a trace of the problem:
[image]

The trace consists of
- 2 peripherals connected to the same SPI bus (SCLK, PICO and POCI)
- distinct CS lines A and B
- distinct IRQ lines A and B
- one debug output signal showing when my program sees the falling edge event on IRQ line A
- the peripherals are handled in 2 different threads - no libgpiod objects are shared between the threads

A successful communication round:
1. Falling edge on IRQ A
2. RPi program sees the lgpiod event and toggles the debug pin
3. RPi reads out data from peripheral A and eventually sends the interrupt ack message. The peripheral A then immediately deasserts the IRQ line

Peripheral B is handled in a different thread and does exactly the same. After another communication round, we see:
4. Falling edge on IRQ A
5. RPi program sees the lgpiod event and toggles the debug pin
6. RPi program acknowledges the interrupt, peripheral A deasserts the IRQ line
7. Peripheral A asserts the IRQ line for the next round, but the host doesn't see the event and hangs `gpiod_line_event_wait()` (I switched to the CAPI now).

I am not able to reproduce this problem with only one peripheral, but only with 2.

warthog618, 2024-06-17:
Interesting.  If it only happens with two peripherals then you've probably found a race somewhere - the problem is finding where.  It would be good to get some debugging from the kernel to see if it is seeing that interrupt. (Hopefully Bart can suggest how best to do that.)  If the kernel is seeing it then we can track down where in the kernel it is getting lost.  If the kernel isn't seeing it then the problem is in the driver or hardware itself.

It would also be good to be able to reproduce the problem without additional hardware - just with pins looped back on the RPi.  Unfortunately I don't have a Pi 3B to play with.

Other background questions:
How do you drive the SPI (including the chip selects) and the debug pin?

Your two peripherals are synchronised so they assert their IRQs at about the same time (within the resolution of the scope image)?
At around 250Hz?  Reminds me of a project I worked on with two SPI ADCs, though that had a much higher data rate to the point it required separate SPI buses and bare metal rather than Linux.

brgl, 2024-06-17:
Any chance that GPIO1's interrupt fires when interrupts are disabled as the one for GPIO2 is being handled?

rweickelt, 2024-06-17:
Today I found [this topic](https://forums.raspberrypi.com/viewtopic.php?p=1224799&hilit=RPi+gpio+lost+interrupt#p1224799) which indicates that this might be hardware bug with a [workaround in place](https://github.com/raspberrypi/linux/pull/5457).

Surprising that any modern MCU would fail on such a basic task.

warthog618, 2024-06-18:
Yep, that could be it.  It would be good to confirm that a patched kernel fixes your problem.

I'm more surprised that it has taken this long to be picked up and fixed.  Even if it is specific to the 3B, that has been out a good while.

golemparts, 2024-06-25:
As a side note, I started working on transitioning to the GPIO v2 API ([here](https://github.com/golemparts/rppal/blob/master/src/gpio/ioctl/v2.rs)) back when it wasn't certain the Pi 5 would still give us direct GPIO register access, but never completed the transition when it was clear we could use `/dev/gpiomem0` instead. The GPIO API, currently v1, is used for interrupt detection. The v2 [documentation](https://docs.kernel.org/next/userspace-api/gpio/gpio-v2-line-event-read.html) mentions this regarding missing events:

> The size of the kernel event buffer is fixed at the time of line request creation, and can be influenced by the request.event_buffer_size. The default size is 16 times the number of lines requested.
> 
> The buffer may overflow if bursts of events occur quicker than they are read by userspace. If an overflow occurs then the oldest buffered event is discarded. Overflow can be detected from userspace by monitoring the event sequence numbers.

So while this doesn't solve the issue, once I get around to moving the interrupt handling to the v2 API, which adds those event sequence numbers, at least I can detect when a event has been skipped, and possibly add that info to the callback.

closed 2024-10-11 by brgl