brgl/libgpiod #85: Wheels for musl don't work

brgl, 2024-06-21:
Hi @vfazio 

I had some time today to test the wheels and while the once for glibc work fine for `x86-64` and `aarch64`, the ones linked against musl give me the following error:

```
Python 3.12.4 (main, Jun  6 2024, 18:26:44) [GCC 14.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.12/site-packages/gpiod/__init__.py", line 10, in <module>
    from . import _ext
ImportError: cannot import name '_ext' from partially initialized module 'gpiod' (most likely due to a circular import) (/usr/lib/python3.12/site-packages/gpiod/__init__.py)
```

After having installed them like that:

```
# pip3 install gpiod
Collecting gpiod
  Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_x86_64.whl.metadata (3.2 kB)
Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_x86_64.whl (96 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.2/96.2 kB 865.1 kB/s eta 0:00:00
Installing collected packages: gpiod
Successfully installed gpiod-2.2.0
```

Could you advise?

brgl, 2024-06-21:
Please also note that this is reported by `ldd`:

```
# ldd /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so 
	ldd (0x7f9f345c7000)
	libc.musl-x86_64.so.1 => ldd (0x7f9f345c7000)
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyDict_Next: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyLong_AsSize_t: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyEval_GetGlobals: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_Free: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_CallFunction: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_Type: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyErr_SetString: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyUnicode_FromString: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyLong_FromLong: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyErr_Occurred: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyModule_AddStringConstant: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyEval_RestoreThread: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: _PyObject_New: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyArg_ParseTupleAndKeywords: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_CallFinalizerFromDealloc: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyModule_AddType: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_GetAttrString: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyList_SetItem: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyLong_AsUnsignedLong: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyModule_AddObject: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyMem_Calloc: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyMem_Free: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyTuple_GetItem: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyUnicode_AsUTF8: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyErr_SetFromErrno: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyList_GetItem: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyList_New: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_CallMethod: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyEval_SaveThread: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_GetIter: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: _Py_Dealloc: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyObject_Size: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyBool_FromLong: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyErr_NoMemory: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyModule_AddIntConstant: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyLong_FromUnsignedLong: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyModule_Create2: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyLong_AsLong: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyDict_GetItemString: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyIter_Next: symbol not found
[... cut ...]
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyType_GenericNew: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyType_GenericNew: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyType_GenericNew: symbol not found
Error relocating /usr/lib/python3.12/site-packages/gpiod/_ext.cpython-312-x86_64-linux-musl.so: PyType_GenericNew: symbol not found
```

vfazio, 2024-06-21:
This is going to be fun..

vfazio, 2024-06-21:
Works on aarch64:

```
localhost:/tmp# python3 -m venv venv
localhost:/tmp# . venv/bin/activate
(venv) localhost:/tmp# pip install gpiod
Collecting gpiod
  Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_aarch64.whl.metadata (3.2 kB)
Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_aarch64.whl (96 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.0/97.0 kB 899.0 kB/s eta 0:00:00
Installing collected packages: gpiod
Successfully installed gpiod-2.2.0

[notice] A new release of pip is available: 24.0 -> 24.1
[notice] To update, run: pip install --upgrade pip
(venv) localhost:/tmp# python3
Python 3.12.3 (main, Apr 18 2024, 07:52:31) [GCC 13.2.1 20240309] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod

>>> gpiod.version.__version__
'2.2.0'

```

vfazio, 2024-06-21:
Note that the symbols are undefined the same way between musl and glibc:

musl
```
(venv) localhost:/tmp# readelf --dyn-syms venv/lib/python3.12/site-packages/gpio
d/_ext.cpython-312-aarch64-linux-musl.so | grep Py | sort -k8
    83: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyArg_ParseTuple
    29: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyArg_ParseTuple[...]
    64: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyBool_FromLong
    77: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyDict_GetItemString
     3: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyDict_Next
    69: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyErr_NoMemory
    23: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyErr_Occurred
    50: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyErr_SetFromErrno
    20: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyErr_SetString
     7: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyEval_GetGlobals
    25: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyEval_RestoreThread
    59: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyEval_SaveThread
    37: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_BrokenPipeError
    47: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_ChildProce[...]
    51: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_FileExistsError
    66: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_FileNotFou[...]
    68: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_Interrupte[...]
    15: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_IsADirecto[...]
    10: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_NotADirect[...]
    74: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_NotImpleme[...]
    79: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_OSError
    81: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_PermissionError
    44: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_TimeoutError
    52: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_TypeError
    49: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyExc_ValueError
   186: 00000000000065d0   280 FUNC    GLOBAL DEFAULT    9 PyInit__ext
    82: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyIter_Next
    54: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyList_GetItem
    55: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyList_New
    34: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyList_SetItem
    76: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyLong_AsLong
     5: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyLong_AsSize_t
    36: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyLong_AsUnsignedLong
    22: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyLong_FromLong
    71: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyLong_FromUnsig[...]
    40: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyMem_Calloc
    41: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyMem_Free
    70: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyModule_AddIntC[...]
    39: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyModule_AddObject
    24: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyModule_AddStri[...]
    31: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyModule_AddType
    72: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyModule_Create2
    30: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_CallFin[...]
    12: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_CallFunction
    58: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_CallMethod
     9: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_Free
    33: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_GetAttr[...]
    60: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_GetIter
    63: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_Size
    17: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_Type
    42: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyTuple_GetItem
    67: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyType_GenericNew
    43: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyUnicode_AsUTF8
    21: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyUnicode_FromString
   145: 0000000000005e90    52 FUNC    GLOBAL DEFAULT    9 Py_gpiod_GetGlob[...]
   120: 00000000000062d0   156 FUNC    GLOBAL DEFAULT    9 Py_gpiod_LineCon[...]
   147: 00000000000064b4   156 FUNC    GLOBAL DEFAULT    9 Py_gpiod_LineSet[...]
   193: 00000000000070a0   260 FUNC    GLOBAL DEFAULT    9 Py_gpiod_MakeReq[...]
   209: 0000000000005ec4    96 FUNC    GLOBAL DEFAULT    9 Py_gpiod_PyLongA[...]
   190: 0000000000005d44   328 FUNC    GLOBAL DEFAULT    9 Py_gpiod_SetErrF[...]
   172: 0000000000005d10    52 FUNC    GLOBAL DEFAULT    9 Py_gpiod_dealloc
    27: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _PyObject_New
    62: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_Dealloc
    53: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_FalseStruct
[... 52 lines cut ...]
    44: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_CallFunction
    12: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_CallMethod
    32: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_Free
    16: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND PyObject_GetAttr[...]
[... cut ...]
   184: 0000000000006280   287 FUNC    GLOBAL DEFAULT   11 Py_gpiod_SetErrF[...]
   167: 0000000000006250    34 FUNC    GLOBAL DEFAULT   11 Py_gpiod_dealloc
    68: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _PyObject_New
    18: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_Dealloc
    42: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_FalseStruct
    59: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_NoneStruct
    62: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND _Py_TrueStruct

```

vfazio, 2024-06-21:
If I add libpython3 as a dependency, the unfound symbol errors go away:

```
(venv) localhost:/tmp# patchelf --add-needed libpython3.so venv/lib/python3.12/s
ite-packages/gpiod/_ext.cpython-312-aarch64-linux-musl.so
(venv) localhost:/tmp# ldd venv/lib/python3.12/site-packages/gpiod/_ext.cpython-
312-aarch64-linux-musl.so
        /lib/ld-musl-aarch64.so.1 (0x7f8b536000)
        libpython3.so => /usr/lib/libpython3.so (0x7f8b4e3000)
        libc.musl-aarch64.so.1 => /lib/ld-musl-aarch64.so.1 (0x7f8b536000)
        libpython3.12.so.1.0 => /usr/lib/libpython3.12.so.1.0 (0x7f8aeb5000)

```

I see we're using Python.h, but it doesn't look like we're linking against libpython. I don't know if this is actually a functional issue. the package does seem to work without the linkage w/ glibc and at least musl aarch64...

vfazio, 2024-06-21:
@brgl Can you try installing gpiod in a virtualenv?

```
cd `mktemp -d`
python3 -m venv venv
. venv/bin/activate
pip install gpiod==2.2.0
python3 -c "import gpiod;print(gpiod.__version__)"
```

vfazio, 2024-06-21:
Apparently, it is recommended to _not_ link against libpython. 

https://peps.python.org/pep-0513/#libpythonx-y-so-1
https://github.com/pypa/manylinux/issues/69

So, what we're doing is appropriate.

brgl, 2024-06-21:
> @brgl Can you try installing gpiod in a virtualenv?
> 
> ```
> cd `mktemp -d`
> python3 -m venv venv
> . venv/bin/activate
> pip install gpiod==2.2.0
> python3 -c "import gpiod;print(gpiod.__version__)"
> ```

I can (on Monday though) but shouldn't it work outside venv anyway? Also: is virtualenv and venv the same thing? Yocto only seems to support virtualenv.

vfazio, 2024-06-21:
> I can (on Monday though) but shouldn't it work outside venv anyway? Also: is virtualenv and venv the same thing? Yocto only seems to support virtualenv.

They should do the same thing, but the command would be `python -m virtualenv venv`

vfazio, 2024-06-21:
Oof, Yocto. You can also try installing from the sdist and not the wheel, but you'll need the required headers on the machine for that to work.

```
pip install gpiod --no-binary :all:
```

brgl, 2024-07-01:
I know I can install from sdist, that's not an issue. We've had a recipe for yocto for a long time. I'm only using yocto as a minimal system with just python3 and pip to see if the wheels work fine.

brgl, 2024-07-01:
Here's with `virtualenv`:

```
root@qemux86-64:~# python3 -m virtualenv venv
created virtual environment CPython3.12.4.final.0-64 in 419ms
  creator CPython3Posix(dest=/root/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==24.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
root@qemux86-64:~# pip3 install gpiod
Collecting gpiod
  Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_x86_64.whl.metadata (3.2 kB)
Downloading gpiod-2.2.0-cp312-cp312-musllinux_1_2_x86_64.whl (96 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.2/96.2 kB 1.1 MB/s eta 0:00:00
Installing collected packages: gpiod
Successfully installed gpiod-2.2.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommev
root@qemux86-64:~# python3 -c "import gpiod;print(gpiod.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.12/site-packages/gpiod/__init__.py", line 10, in <module>
    from . import _ext
ImportError: cannot import name '_ext' from partially initialized module 'gpiod' (most likely due to a circular import) (/usr/lib/python3.12/site)
root@qemux86-64:~# 
```

vfazio, 2024-07-01:
I tested the wheels in an Alpine container, both 3.14 and 3.20, and they seemed to work fine there, which seems to imply either some python distribution nuance for Yocto or some c level library issue preventing the library from being loaded. I guess I could try to debug this at some point if i could get a yocto image with a toolchain and gdb in it

Alpine 3.14:
```
vfazio@vfazio4 /tmp/tmp.pBsqfbBKHz $ docker run -it --rm alpine:3.14 sh
/ # apk add python3 py3-pip py3-setuptools
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/x86_64/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/x86_64/APKINDEX.tar.gz
(1/38) Installing libbz2 (1.0.8-r1)
(2/38) Installing expat (2.5.0-r0)
(3/38) Installing libffi (3.3-r2)
(4/38) Installing gdbm (1.19-r0)
(5/38) Installing xz-libs (5.2.5-r1)
(6/38) Installing libgcc (10.3.1_git20210424-r2)
(7/38) Installing libstdc++ (10.3.1_git20210424-r2)
(8/38) Installing mpdecimal (2.5.1-r1)
(9/38) Installing ncurses-terminfo-base (6.2_p20210612-r1)
(10/38) Installing ncurses-libs (6.2_p20210612-r1)
(11/38) Installing readline (8.1.0-r0)
(12/38) Installing sqlite-libs (3.35.5-r0)
(13/38) Installing python3 (3.9.17-r0)
(14/38) Installing py3-appdirs (1.4.4-r2)
(15/38) Installing py3-chardet (4.0.0-r2)
(16/38) Installing py3-idna (3.2-r0)
(17/38) Installing py3-urllib3 (1.26.5-r0)
(18/38) Installing py3-certifi (2020.12.5-r1)
(19/38) Installing py3-requests (2.25.1-r4)
(20/38) Installing py3-msgpack (1.0.2-r1)
(21/38) Installing py3-lockfile (0.12.2-r4)
(22/38) Installing py3-cachecontrol (0.12.6-r1)
(23/38) Installing py3-colorama (0.4.4-r1)
(24/38) Installing py3-contextlib2 (0.6.0-r1)
(25/38) Installing py3-distlib (0.3.1-r3)
(26/38) Installing py3-distro (1.5.0-r3)
(27/38) Installing py3-six (1.15.0-r1)
(28/38) Installing py3-webencodings (0.5.1-r4)
(29/38) Installing py3-html5lib (1.1-r1)
(30/38) Installing py3-parsing (2.4.7-r2)
(31/38) Installing py3-packaging (20.9-r1)
(32/38) Installing py3-toml (0.10.2-r2)
(33/38) Installing py3-pep517 (0.10.0-r2)
(34/38) Installing py3-progress (1.5-r2)
(35/38) Installing py3-retrying (1.3.3-r1)
(36/38) Installing py3-ordered-set (4.0.2-r1)
(37/38) Installing py3-setuptools (52.0.0-r3)
(38/38) Installing py3-pip (20.3.4-r1)
Executing busybox-1.33.1-r8.trigger
OK: 78 MiB in 52 packages
/ # pip install -U pip
Requirement already satisfied: pip in /usr/lib/python3.9/site-packages (20.3.4)
Collecting pip
  Downloading pip-24.1.1-py3-none-any.whl (1.8 MB)
     |████████████████████████████████| 1.8 MB 3.3 MB/s
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-24.1.1

/ # pip install gpiod
Collecting gpiod
  Downloading gpiod-2.2.0-cp39-cp39-musllinux_1_2_x86_64.whl.metadata (3.2 kB)
Downloading gpiod-2.2.0-cp39-cp39-musllinux_1_2_x86_64.whl (94 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 94.2/94.2 kB 2.6 MB/s eta 0:00:00
Installing collected packages: gpiod
Successfully installed gpiod-2.2.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
/ # python3 -vv -c "import gpiod" 2>&1 | grep gpiod
# trying /gpiod.cpython-39-x86_64-linux-musl.so
[... 295 lines cut ...]
╰─>
    The system-wide python installation should be maintained using the system
    package manager (apk) only.

    If the package in question is not packaged already (and hence installable via
    "apk add py3-somepackage"), please consider installing it inside a virtual
    environment, e.g.:

    python3 -m venv /path/to/venv
    . /path/to/venv/bin/activate
    pip install mypackage

    To exit the virtual environment, run:

    deactivate

    The virtual environment is not deleted, and can be re-entered by re-sourcing
    the activate file.

    To automatically manage virtual environments, consider using pipx (from the
    pipx package).

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

brgl, 2024-07-01:
Ok, I guess this isn't that much of an issue as yocto already has a recipe for python bindings but it would be worth to figure out why this is happening at some point. Let's leave it for now.

vfazio, 2024-07-01:
Ok, we can leave this open with a plan to investigate further at some point

pdcastro, 2024-10-10:
@brgl, if you pip install `gpiod` with the `--no-binary gpiod` option (which forces compilation from source), do you still get the same `ImportError`?
```sh
# pip3 install --no-binary gpiod gpiod
...
# python3 -c "import gpiod;print(gpiod.__version__)"
2.2.1
```

vfazio, 2024-10-10:
That's essentially how Yocto's recipe works, by manually building/installing the wheel from an sdist AFAIK.

My gut says it's a problem with either Python search paths or a libc incompatibility since Yocto builds all of that from scratch.

It would be interesting to see just the output from `python3 -vvv -c "import gpiod"` to see the search paths it's attempting. The last time i saw this was actually as part of a build and it was because the external library wasn't in the correct location.

vfazio, 2024-10-10:
I just noticed you did this in qemu. If you have a yocto config you can share that i can use to reproduce, ill try to give it a shot when i get back from Japan in Nov.

brgl, 2024-10-10:
It's literally the simplest `core-image-minimal` with `python3-pip` in `INSTALL_IMAGE`.

Full config is:

```
MACHINE = "qemux86-64"
DISTRO = "poky"

EXTRA_IMAGE_FEATURES = "debug-tweaks"

INIT_MANAGER = "sysvinit"

IMAGE_INSTALL:append = " \
    python3-pip \
    connman \
"

TCLIBC = "musl"
```

pdcastro, 2024-10-10:
> Sure but what good is the wheel then? :)

If the `ImportError` also happens with `pip3 install --no-binary gpiod gpiod`, this rules out the issue being with the wheel published to PyPI (because the wheel published to PyPI is not used). I didn’t mean that’s a solution to problem, just a debugging step to isolate the problem.