brgl/libgpiod #86: Cannot build libgpiod on a NanoPi NEO3

Sl8rBug, 2024-07-04:
I am having difficulty building libgpiod v2.2.0 from source. 
I have cloned libgpio from https://github.com/brgl/libgpiod onto a new install of debian on a NanoPi NEO3.
This is a minimal install with limited pre-installed packages.

I went to run the command : `./autogen.sh --enable-tools=yes --prefix=<install path> `from the libgpiod-master directory and noticed I need to include an install path.
Where do I install it to?
After some googling (with out much success) and looking at the issues here on github, I came across an issue encountered by JohnWulff (https://github.com/brgl/libgpiod/issues/53) in which he was having similar problems.
From his post I saw that he had the install path as /usr/local

I ran the command:
   ` ./autogen.sh --enable-tools=yes --prefix=/usr/local
`
This came back with the message:

>  ./autogen.sh: 12: autoreconf: not found

I noticed in issue #53, JohnWulff had installed the packages 'autoconf; and 'autoconf-archive'
Ablu mentioned if 'libtool' had been installed.

I then installed these missing packages.
```    sudo apt install autoconf
    sudo apt install autoconf-archive
    sudo apt install libtool
```
After running the command: `./autogen.sh --enable-tools=yes --prefix=/usr/local`
The following message came back:

> Can't exec "aclocal": No such file or directory at /usr/share/autoconf/Autom4te/FileUtils.pm line 274.
> autoreconf: error: aclocal failed with exit status: 2

Googling led me to the information that I needed to install the package 'automake'.

    `sudo apt install automake`

An idea popped into my head, "Let's install 'make' just in case it's missing".

   ` sudo apt install make`

It was missing because the above command installed it.

However, after running the  `./autogen.sh --enable-tools=yes --prefix=/usr/local`  command again, there was some positive looking output until:-

> configure.ac:75: warning: The macro `AC_HEADER_STDC' is obsolete.
> configure.ac:75: You should run autoupdate.
> ./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
> configure.ac:75: the top level
> configure.ac:85: error: possibly undefined macro: AC_CHECK_HEADERS
>       If this token and others are legitimate, please use m4_pattern_allow.
>       See the Autoconf documentation.
> configure.ac:210: error: possibly undefined macro: AC_LANG_PUSH
> configure.ac:213: error: possibly undefined macro: AC_LANG_POP
> autoreconf: error: /usr/bin/autoconf failed with exit status: 1

After clearing some (unnecessary) hurdles, I have still stumbled.
I have no idea what these errors mean, how to fix them or even where to look for information on how to continue.
Am I still missing dependencies?
I have very limited experience in building libraries/programs from source, and then it was on a fully blown desktop computer which had all dependencies installed.

Any help appreciated.
Regards,
John

brgl, 2024-07-04:
> I am having difficulty building libgpiod v2.2.0 from source. I have cloned libgpio from https://github.com/brgl/libgpiod onto a new install of debian on a NanoPi NEO3. This is a minimal install with limited pre-installed packages.
> 

libgpiod v2.2 is not out yet so you're probably referring to the current master branch. You should stick with the latest stable release which is v2.1.2.

> I went to run the command : `./autogen.sh --enable-tools=yes --prefix=<install path> `from the libgpiod-master directory and noticed I need to include an install path. Where do I install it to? After some googling (with out much success) and looking at the issues here on github, I came across an issue encountered by JohnWulff (#53) in which he was having similar problems. From his post I saw that he had the install path as /usr/local
> 
> I ran the command: `./autogen.sh --enable-tools=yes --prefix=/usr/local` This came back with the message:
> 
> > ./autogen.sh: 12: autoreconf: not found
> 
> I noticed in issue #53, JohnWulff had installed the packages 'autoconf; and 'autoconf-archive' Ablu mentioned if 'libtool' had been installed.
> 
> I then installed these missing packages.
> 
> ```
>     sudo apt install autoconf-archive
>     sudo apt install libtool
> ```
> 
> After running the command: `./autogen.sh --enable-tools=yes --prefix=/usr/local` The following message came back:
> 
> > Can't exec "aclocal": No such file or directory at /usr/share/autoconf/Autom4te/FileUtils.pm line 274.
> > autoreconf: error: aclocal failed with exit status: 2
> 
> Googling led me to the information that I needed to install the package 'automake'.
> 
> ```
> `sudo apt install automake`
> ```
> 
> An idea popped into my head, "Let's install 'make' just in case it's missing".
> 
> ` sudo apt install make`
> 
> It was missing because the above command installed it.
> 
> However, after running the `./autogen.sh --enable-tools=yes --prefix=/usr/local` command again, there was some positive looking output until:-
> 
> > configure.ac:75: warning: The macro `AC_HEADER_STDC' is obsolete.
> > configure.ac:75: You should run autoupdate.
> > ./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
> > configure.ac:75: the top level
> > configure.ac:85: error: possibly undefined macro: AC_CHECK_HEADERS
> > If this token and others are legitimate, please use m4_pattern_allow.
> > See the Autoconf documentation.
> > configure.ac:210: error: possibly undefined macro: AC_LANG_PUSH
> > configure.ac:213: error: possibly undefined macro: AC_LANG_POP
> > autoreconf: error: /usr/bin/autoconf failed with exit status: 1
> 
> After clearing some (unnecessary) hurdles, I have still stumbled. I have no idea what these errors mean, how to fix them or even where to look for information on how to continue. Am I still missing dependencies? I have very limited experience in building libraries/programs from source, and then it was on a fully blown desktop computer which had all dependencies installed.
> 
> Any help appreciated. Regards, John

Do you have `pkg-config` installed?

Sl8rBug, 2024-07-04:
brgl, thanks for your quick reply.
No, `pkg-config` was not installed.
After  installing it, `sudo apt install pkg-config` , I ran the autogen.sh ... command and it ran without error.
Although there was a warning:-

> configure.ac:75: warning: The macro `AC_HEADER_STDC' is obsolete.
> configure.ac:75: You should run autoupdate.
> ./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
> configure.ac:75: the top level

I then did `sudo make` and `sudo make install`
Everything built and installed fine. No errors reported.

When I ran the `gpiodetect` command, the following errors came back.
 

> gpiodetect: error while loading shared libraries: libgpiod.so.3: cannot open shared object file: No such file or directory

You may be correct that I am referring to the current master branch. 
This is the link that I used:-
`wget https://github.com/brgl/libgpiod/archive/refs/heads/master.zip`

If this is v2.2.0, could it be the reason for the gpiodetect error?
Where do I find v2.1.2?
Google sends me here.
Regards,
John

brgl, 2024-07-04:
> brgl, thanks for your quick reply. No, `pkg-config` was not installed. After installing it, `sudo apt install pkg-config` , I ran the autogen.sh ... command and it ran without error. Although there [...]

Your system's linker most likely doesn't look for libraries in the directory you installed libgpiod to (with `--prefix`). You can use the `LD_LIBRARY_PATH` environment variable to point it to it.

> You may be correct that I am referring to the current master branch. This is the link that I used:- `wget https://github.com/brgl/libgpiod/archive/refs/heads/master.zip`
> 
> If this is v2.2.0, could it be the reason for the gpiodetect error? Where do I find v2.1.2? Google sends me here. Regards, John

I don't want to go all RTFM on you but if you spent 2 minutes reading the README, then you'd find the link at the bottom: https://mirrors.edge.kernel.org/pub/software/libs/libgpiod/

Sl8rBug, 2024-07-05:
We finally got there.
I entered the command `export LD_LIBRARY_PATH=/usr/local` and then did a rebuild.
Checked with `gpiodetect` and everything looks good.
Now, to update my existing python scripts to reflect the new instruction syntax.

I apologise, I had read to the bottom of the README but didn't realise what I was looking at.
I am just a hobbyist playing with a few SBCs learning how to use them in practical ways.
My wife often tells me I'm a bit thick and need the bleeding obvious to be pointed out to me.

Thank you for your help and patience.
Regards,
John

closed 2024-07-05 by Sl8rBug