brgl/libgpiod #91: Debian version 2.x.x. libgpiod-dev availability

remendes, 2024-08-27:
I'm trying to build a container based on Debian to use C++ with the version 2.x.x of libgpiod. 

I've noticed that the package libgpiod-dev that is in the Debian package manager is currently in version 1.6.3-1.1. Cannot install libgpiod2-dev as it is not available.

How can I develop using the version 2 of libgpiod with the new features? Should I need to download the code from GitHub and compile it myself?

brgl, 2024-08-28:
Yes, you'll need to build the project from sources. I don't deal with distros other than poky so I can't tell when v2 will be available in Debian.

warthog618, 2024-08-31:
You can track the status of libgpiod packages in Debian [here](https://tracker.debian.org/pkg/libgpiod).  As of now, libgpiod v2 has made its way into unstable, but if you are on stable or testing you would need to backport it, so you are probably still better off building from source.

It appears libgpiod-dev will jump to v2 - there will be no libgpiod2-dev.  I guess that makes sense as the header name is not changing.  So if a user wants to develop for v1 they will have to switch to an older version of libgpiod-dev.   That makes sense as v1 is deprecated for new developments which should be using libgpiod v2.  OTOH there will be libgpiod3 so both v1 (libgpiod2) and v2 (libgpiod3) versions of the library can be installed for apps that depend on them.

closed 2024-10-11 by brgl