brgl/libgpiod #94: Add the py.typed marker to Python bindings

vfazio, 2024-09-11:
The Python bindings are currently supplying type annotations, however, without a py.typed marker, type checkers like mypy do not know to check for these annotations.

```
(venv) vfazio@vfazio4 /tmp/tmp.qo91wk2VVE $ cat test.py
import gpiod

chip = gpiod.Chip(1)

(venv) vfazio@vfazio4 /tmp/tmp.qo91wk2VVE $ mypy test.py
test.py:1: error: Skipping analyzing "gpiod": module is installed, but missing library stubs or py.typed marker  [import-untyped]
test.py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
```

We should expect:

```
(venv) vfazio@vfazio4 /tmp/tmp.qo91wk2VVE $ mypy test.py
test.py:3: error: Argument 1 to "Chip" has incompatible type "int"; expected "str"  [arg-type]
Found 1 error in 1 file (checked 1 source file)

```

Adding this support should be pretty easy, so I'll post a patch tomorrow

```diff
vfazio@vfazio4 ~/development/libgpiod $ git diff HEAD
diff --git a/bindings/python/gpiod/py.typed b/bindings/python/gpiod/py.typed
new file mode 100644
index 0000000..e69de29
diff --git a/bindings/python/setup.py b/bindings/python/setup.py
index 9607a28..1f04b99 100644
--- a/bindings/python/setup.py
+++ b/bindings/python/setup.py
@@ -224,6 +224,7 @@ setup(
     name="gpiod",
     url="https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git",
     packages=find_packages(exclude=["tests", "tests.*"]),
+    package_data={"gpiod": ["py.typed"]},
     python_requires=">=3.9.0",
     ext_modules=[gpiod_ext],
     cmdclass={"build_ext": build_ext, "sdist": sdist},

```

vfazio, 2024-09-14:
https://lore.kernel.org/linux-gpio/20240914183302.15768-1-vfazio@gmail.com/T/#u

brgl, 2024-09-16:
I applied both patches. After giving mypy a try, I'm seeing some warnings all around the place. Do you have any plans of addressing them yourself or should start looking into them?

closed 2024-09-16 by brgl

vfazio, 2024-09-16:
> I applied both patches. After giving mypy a try, I'm seeing some warnings all around the place. Do you have any plans of addressing them yourself or should start looking into them?

I can submit a single patch to address the mypy warnings. Im considering specifying some dev dependencies such as mypy and black or ruff just to keep things consistent, but that may come later

brgl, 2024-09-16:
> I can submit a single patch to address the mypy warnings. Im considering specifying some dev dependencies such as mypy and black or ruff just to keep things consistent, but that may come later

Ok, I'll wait for your patch then. Don't you think it would make sense to split them at least by subject?

vfazio, 2024-09-16:
> Ok, I'll wait for your patch then. Don't you think it would make sense to split them at least by subject?

Like, per file? The diff for resolving all of mypy's complaints are all over the subtree.

```
(venv) vfazio@vfazio4 /mnt/development/libgpiod/bindings/python $ git diff --stat
 bindings/python/gpiod/__init__.py      | 19 ++++++++++++++++++-
 bindings/python/gpiod/chip.py          | 12 +++++-------
 bindings/python/gpiod/edge_event.py    |  2 +-
 bindings/python/gpiod/info_event.py    |  2 +-
 bindings/python/gpiod/internal.py      |  3 ++-
 bindings/python/gpiod/line.py          |  2 +-
 bindings/python/gpiod/line_info.py     |  1 -
 bindings/python/gpiod/line_request.py  | 24 +++++++++++++++---------
 bindings/python/gpiod/line_settings.py |  2 +-
 9 files changed, 44 insertions(+), 23 deletions(-)
```

```diff
diff --git a/bindings/python/gpiod/__init__.py b/bindings/python/gpiod/__init__.py
index 9cbb8df..fa2becb 100644
--- a/bindings/python/gpiod/__init__.py
+++ b/bindings/python/gpiod/__init__.py
@@ -7,7 +7,7 @@ Python bindings for libgpiod.
 This module wraps the native C API of libgpiod in a set of python classes.
 """
 
-from . import _ext
+from . import _ext  # type: ignore[attr-defined]
 from . import line
 from .chip import Chip
 from .chip_info import ChipInfo
@@ -20,6 +20,23 @@ from .version import __version__
 
 api_version = _ext.api_version
 
+__all__ = [
+    "_ext",
+    "line",
+    "Chip",
+    "ChipInfo",
+    "EdgeEvent",
+    "ChipClosedError",
+    "RequestReleasedError",
+    "InfoEvent",
+    "LineRequest",
+    "LineSettings",
+    "is_gpiochip_device",
+    "request_lines",
+    "api_version",
+    "__version__",
+]
+
 
 def is_gpiochip_device(path: str) -> bool:
     """
diff --git a/bindings/python/gpiod/chip.py b/bindings/python/gpiod/chip.py
index 7692ba4..e7af9fd 100644
--- a/bindings/python/gpiod/chip.py
+++ b/bindings/python/gpiod/chip.py
@@ -1,7 +1,7 @@
 # SPDX-License-Identifier: LGPL-2.1-or-later
 # SPDX-FileCopyrightText: 2022 Bartosz Golaszewski <brgl@bgdev.pl>
 
-from . import _ext
+from . import _ext  # type: ignore[attr-defined]
 from .chip_info import ChipInfo
 from .exception import ChipClosedError
 from .info_event import InfoEvent
@@ -11,11 +11,9 @@ from .line_info import LineInfo
[... 179 lines cut ...]
@@ -165,7 +171,7 @@ class LineRequest:
                 lines = [lines]
 
             for line in lines:
-                offset = self._name_map[line] if self._check_line_name(line) else line
+                offset = self._name_map[line] if self._check_line_name(line) else line  # type: ignore[index]
                 line_settings[offset] = settings
 
         for offset in self.offsets:
diff --git a/bindings/python/gpiod/line_settings.py b/bindings/python/gpiod/line_settings.py
index 41712cc..dab6634 100644
--- a/bindings/python/gpiod/line_settings.py
+++ b/bindings/python/gpiod/line_settings.py
@@ -1,7 +1,7 @@
 # SPDX-License-Identifier: LGPL-2.1-or-later
 # SPDX-FileCopyrightText: 2022 Bartosz Golaszewski <brgl@bgdev.pl>
 
-from . import _ext
+from . import _ext  # type: ignore[attr-defined]
 from dataclasses import dataclass
 from datetime import timedelta
 from gpiod.line import Direction, Bias, Drive, Edge, Clock, Value
```

I can try to group them by theme, but as this is nothing but internal information, there should be no functional change from this patch (or series of patches). It seems like overkill to submit 8-9 patches.

brgl, 2024-09-16:
Not by file, by theme ideally. It makes it way easier to review. I don't mind several patches.

vfazio, 2024-09-16:
Understood. I'll see what i can do in the next couple of days so split these changes up.