brgl/libgpiod #96: Bindings: Python: type annotations may need tweaks

vfazio, 2024-09-12:
I'm look at the current annotations before I submit a PR for #94.

It looks like `Chip.request_lines` has an `output_values` argument that's defined as `Optional[dict[tuple[Union[int, str]], Value]]`

It looks to be defined similarly to the `config` argument, however I think the handling of `output_values` is assuming it's defined as: `Optional[dict[Union[int, str], Value]]`

```python
        # If we have global output values - map line names to offsets
        if output_values:
            mapped_output_values = {
                self.line_offset_from_id(line): value
                for line, value in output_values.items()
            }
```

`line_offset_from_id` takes a `str` or `int`, which would be fine if it was annotated as `dict[int|str, Value]` and not `dict[tuple, Value]`.

Before I fix this, i want to confirm what interface is expected here, the typed interface or the functional interface?

If i look at `examples/toggle_multiple_line_values.py`, it probably should be the functional interface, in which case i can fix the typing annotation.

There are some other cleanups needed to get the library passing a mypy check, but they don't seem to be functional issues, most are related to `LineRequest` which is an internal object anyway.

brgl, 2024-09-13:
I think you're right. It may be a wrong copy-paste from the config argument on my part. Feel free to change it. Maybe even submit it as a fix. Thanks.

vfazio, 2024-09-14:
https://lore.kernel.org/linux-gpio/20240914183352.16090-1-vfazio@gmail.com/T/#u

closed 2024-09-16 by brgl