brgl/libgpiod #100: python: allow a LineRequest to expose its settings

reidspreiter, 2024-09-19:
Currently, a `LineRequest` exposes line values, but in scenarios where the line's settings are changed often, it would be helpful to expose more.

A new chip object must be opened to access these fields:

```python
class GPIOPin:

    def __init__(self, gpiochip: str, offset: int) -> None:
        self.gpiochip = gpiochip
        self.offset = offset
        with gpiod.Chip(gpiochip) as chip:
            self.line = chip.request_lines({offset: gpiod.LineSettings()})

    def set_status(self, direction: gpiod.line.Direction, active_low: bool, ...) -> None:
        self.line.reconfigure_lines(...)

    @property
    def direction(self) -> gpiod.line.Direction:
        with gpiod.Chip(self.gpiochip) as chip:
            return chip.get_line_info(self.offset).direction

    @property
    def active_low(self) -> bool:
        with gpiod.Chip(self.gpiochip) as chip:
            return chip.get_line_info(self.offset).active_low
```
This could be solved by opening a chip and keeping it open for the duration of the line request, but for many `GPIOPin` instances, many more file descriptors are kept open.

Would something like this be possible to add? Or is there a reason this information can only be retrieved from the chip?
```python
class GPIOPin:

    def __init__(self, gpiochip: str, offset: int) -> None:
        self.offset = offset
        with gpiod.Chip(gpiochip) as chip:
            self.line = chip.request_lines({offset: gpiod.LineSettings()})

    def set_status(self, direction: gpiod.line.Direction, active_low: bool, ...) -> None:
        self.line.reconfigure_lines(...)

    @property
    def direction(self) -> gpiod.line.Direction:
        return self.line.get_settings(self.offset).direction

    @property
    def active_low(self) -> bool:
        return self.line.get_settings(self.offset).active_low
```
Thank you,
Reid S

brgl, 2024-09-20:
> Currently, a `LineRequest` exposes line values, but in scenarios where the line's settings are changed often, it would be helpful to expose more.
>
> A new chip object must be opened to access these [...]

The design goal behind the v2 of the kernel uAPI was to disconnect the life-times of the file descriptors associated with chips and sets of requested lines.

The way it's modeled in the library just follows this principle. I very much prefer it this way (to have LineRequest's and Chip's life-times indepentent), so no: we won't be adding it. 

What is the use-case for that anyway? After all  - you are controlling a requested line - you know its state, don't you?

reidspreiter, 2024-09-20:
That makes sense, thank you for your reply!

My use case allows users to remotely toggle a line's settings for debug purposes, and some interfaces require toggling a line's `direction` during runtime. Lines are also temporarily releasable, allowing other interfaces to request and use them with different settings, before being re-requested by the original requester with the original settings.

Opening the line's chip each time to get the line's `LineInfo` still works great though, and there aren't any noticeable impacts to performance, so I will continue doing this.

closed 2024-09-20 by reidspreiter