brgl/libgpiod #99: Question : gpiod_line_find() equivalent in V2.0 ?

Orel-B, 2024-09-19:
Hello,

I am working on migrating some software using libgpiod v1.6 to V2.0.
I noticed that some useful functions are not developed in V2 such as gpiod_line_find().

I checked on V2 sources how gpioget is working with --by-name and also in : [find_line_by_name.c](https://github.com/brgl/libgpiod/blob/master/examples/find_line_by_name.c)

Why there is no function equivalent to gpiod_line_find() ?
Is an equivalent function planned to be deployed in V2 ?

Thank you,

regards

brgl, 2024-09-19:
Because libgpiod v2 was designed to be minimal and only cover the functionality exposed by the kernel uAPI.

Orel-B, 2024-09-19:
i understand the point, thank you for the answer :)

closed 2024-10-11 by brgl