brgl/libgpiod #101: bindings: python: external module expects return type classes to be in globals()

vfazio, 2024-09-26:
When calling functions like `gpiod.chip.Chip.get_info`, calls are forwarded to the external module to create an object and return it.

Right now, we include `gpiod.chip_info.ChipInfo` as an import in `gpiod.chip`. The module itself does not consume or create instances of `ChipInfo` so it's only purpose is "superficial" and is used for type checking.

However, if i put the import behind a `TYPE_CHECKING` guard so it's only available when actually type checking a la:

```
if TYPE_CHECKING:
    from .chip_info import ChipInfo
```

We get unit test errors:

```
vfazio4 /mnt/development/libgpiod/bindings/python :( # LD_LIBRARY_PATH=/mnt/development/libgpiod/lib/.libs/:/mnt/development/libgpiod/tests/gpiosim/.libs/ ./venv/bin/python -m unittest tests.tests_info_event.WatchingInfoEventWorks.test_watch_line_info_returns_line_info
E
======================================================================
ERROR: test_watch_line_info_returns_line_info (tests.tests_info_event.WatchingInfoEventWorks)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/mnt/development/libgpiod/bindings/python/tests/tests_info_event.py", line 71, in test_watch_line_info_returns_line_info
    info = self.chip.watch_line_info(7)
  File "/mnt/development/libgpiod/bindings/python/gpiod/chip.py", line 191, in watch_line_info
    return self._get_line_info(line, watch=True)
  File "/mnt/development/libgpiod/bindings/python/gpiod/chip.py", line 164, in _get_line_info
    return self._chip.get_line_info(self.line_offset_from_id(line), watch)
  File "/mnt/development/libgpiod/bindings/python/gpiod/chip.py", line 156, in line_offset_from_id
    if offset >= self.get_info().num_lines:
  File "/mnt/development/libgpiod/bindings/python/gpiod/chip.py", line 118, in get_info
    self._info = self._chip.get_info()
SystemError: <method 'get_info' of 'gpiod._ext.Chip' objects> returned NULL without setting an error

----------------------------------------------------------------------
Ran 1 test in 0.001s

FAILED (errors=1)
```

This seems to be because the external module looks for the classes within `globals()`:

```c
type = Py_gpiod_GetGlobalType("ChipInfo");
```

So we have an implicit dependency on `gpiod.chip` importing `gpiod.chip_info` for this (or any similar) lookup to work.

We may be able to avoid this by explicitly importing the class we need:

```diff
diff --git a/bindings/python/gpiod/ext/chip.c b/bindings/python/gpiod/ext/chip.c
index e8eaad8..8826d7e 100644
--- a/bindings/python/gpiod/ext/chip.c
+++ b/bindings/python/gpiod/ext/chip.c
@@ -73,22 +73,25 @@ static PyObject *chip_close(chip_object *self, PyObject *Py_UNUSED(ignored))
 static PyObject *chip_get_info(chip_object *self, PyObject *Py_UNUSED(ignored))
 {
        struct gpiod_chip_info *info;
-       PyObject *type, *ret;
+       PyObject *type, *ret, *module = NULL;
 
-       type = Py_gpiod_GetGlobalType("ChipInfo");
-       if (!type)
-               return NULL;
+       module = PyImport_ImportModule("gpiod.chip_info");
+       if (!module)
+               return PyErr_SetFromErrno(PyExc_ImportError);
 
        info = gpiod_chip_get_info(self->chip);
-       if (!info)
-               return PyErr_SetFromErrno(PyExc_OSError);
-
-        ret = PyObject_CallFunction(type, "ssI",
-                                    gpiod_chip_info_get_name(info),
-                                    gpiod_chip_info_get_label(info),
-                                    gpiod_chip_info_get_num_lines(info));
-        gpiod_chip_info_free(info);
-        return ret;
+       if (!info) {
+               ret = PyErr_SetFromErrno(PyExc_OSError);
+       }
+       else {
+               ret = PyObject_CallMethod(module, "ChipInfo", "ssI",
+                                               gpiod_chip_info_get_name(info),
+                                               gpiod_chip_info_get_label(info),
+                                               gpiod_chip_info_get_num_lines(info));
+               gpiod_chip_info_free(info);
+       }
+       Py_DECREF(module);
+       return ret;
 }
```

```
vfazio4 /mnt/development/libgpiod/bindings/python # LD_LIBRARY_PATH=/mnt/development/libgpiod/lib/.libs/:/mnt/development/libgpiod/tests/gpiosim/.libs/ ./venv/bin/python -m unittest tests.tests_info_event.WatchingInfoEventWorks.test_watch_line_info_returns_line_info
.
----------------------------------------------------------------------
Ran 1 test in 0.001s

OK

```

I don't have much experience with the Python C API so this was just a best guess at getting something working and may not be fully correct from an API perspective (like making sure ref counts are correct).

It's a bit of a weird scenario where gpiod imports _ext but _ext needs gpiod. I don't know if there's a better method to cache the module reference instead of looking it up every function call, but I have a feeling the circular dependency situation may cause problems if not done correctly, which lazy lookups/imports may avoid.

brgl, 2024-09-27:
That's a good catch! Are there more instances like that? The patch looks correct to me wrt reference counting. Do you want to send it to the list?

vfazio, 2024-09-27:
> That's a good catch! Are there more instances like that? The patch looks correct to me wrt reference counting. Do you want to send it to the list?

@brgl there are definitely other places. If you want, I can go through and document them in this issue.

This is one of the nice things about using linters. Running https://docs.astral.sh/ruff/rules/#flake8-type-checking-tch checks helped me identify this.

I thought about including this in the series I plan to submit soon, but it's already 22 patches long and getting rather unwieldy. I will likely just spin up another set of patches for these fixes. After they land, another patch can be made to move the imports behind the `TYPE_CHECKING` guard.

vfazio, 2024-09-27:
The following functions are affected:

```
chip_get_info -> ChipInfo (Chip.get_info)
make_line_info -> LineInfo (Chip.get_line_info, Chip.read_info_event)
chip_read_info_event -> InfoEvent (Chip.read_info_event)
request_get_values -> Value (Request.get_values)
request_read_edge_events -> EdgeEvent (Request.read_edge_events)
```

brgl, 2024-09-27:
Thanks. So are you planning on getting to it eventually or should I start fixing it? Also: the "returned NULL without setting an error" thing is an issue too, we should definitely set an error on every failure in the ext module.

vfazio, 2024-09-27:
@brgl I will probably get to it eventually but may not be as quick as you. 

I did not do an analysis of all places that returned NULL that didn't set an error. It's actually not clear to me what standard Python C API functions set an error or if it's up to the caller to do that by querying PyErr_Occurred?. I see many instances of CPython returning Null without an explicit PyErr_SetString(). Finding and resolving those also seems like it may be a bit more time intensive.

brgl, 2024-09-27:
In that case I'll spend some time cleaning it up. Should I wait for your big series with it? Will it even touch any of the C code?

vfazio, 2024-09-27:
I'm purposefully not touching any of the C code. I'm trying to actually not change any functionality at all with this series. All code coverage and functional tests should be equivalent.

The only thing that may be affected is if you changed any of the return types from the exported module functions since i've added typing for those, but I don't think that's necessary or your intention for these fixes.

vfazio, 2024-10-01:
Apparently `PyImport_ImportModule` will set `ModuleNotFoundError`

```c
	module = PyImport_ImportModule("gpiod.chip_inf");
	if (PyErr_Occurred())
		PyErr_Print();
	if (!module)
		return NULL;
```

```
ModuleNotFoundError: No module named 'gpiod.chip_inf
```

`PyObject_CallMethod` will also set an `AttributeError` if it hits an issue:

```c
	} else {
		ret = PyObject_CallMethod(module, "ChipInf", "ssI",
					  gpiod_chip_info_get_name(info),
					  gpiod_chip_info_get_label(info),
					  gpiod_chip_info_get_num_lines(info));
	 	gpiod_chip_info_free(info);
	}
```
```
AttributeError: module 'gpiod.chip_info' has no attribute 'ChipInf'
```

vfazio, 2024-10-01:
I don't know how you feel about `goto`, but here's a possible implementation:

```c
static PyObject *chip_get_info(chip_object *self, PyObject *Py_UNUSED(ignored))
{
	struct gpiod_chip_info *info;
	PyObject *module = NULL, *ret = NULL, *attr = NULL;

	module = PyImport_ImportModule("gpiod.chip_info");
	if (!module)
		return NULL;

	info = gpiod_chip_get_info(self->chip);
	if (!info) {
		ret = PyErr_SetFromErrno(PyExc_OSError);
		goto out;
	}
	attr = PyObject_GetAttrString(module, "ChipInfo");
	if (!attr)
		goto free_info;

	ret = PyObject_CallFunction(attr, "ssI",
				    gpiod_chip_info_get_name(info),
				    gpiod_chip_info_get_label(info),
				    gpiod_chip_info_get_num_lines(info));
free_info:
	gpiod_chip_info_free(info);
out:	
	Py_DECREF(module);
	return ret;
}
```

I don't know if there's a benefit of calling `PyObject_GetAttrString + PyObject_CallFunction` vs just calling `PyObject_CallMethod`

brgl, 2024-10-01:
> I don't know how you feel about `goto`, but here's a possible implementation:
> 

I'm fine if used in correct context.

> 
> I don't know if there's a benefit of calling `PyObject_GetAttrString + PyObject_CallFunction` vs just calling `PyObject_CallMethod`

I think `PyObject_CallMethod` will end up doing the former internally anyway.

vfazio, 2024-10-01:
indeed

```c
PyObject *
PyObject_CallMethod(PyObject *obj, const char *name, const char *format, ...)
{
    PyThreadState *tstate = _PyThreadState_GET();

    if (obj == NULL || name == NULL) {
        return null_error(tstate);
    }

    PyObject *callable = PyObject_GetAttrString(obj, name);
    if (callable == NULL) {
        return NULL;
    }

    va_list va;
    va_start(va, format);
    PyObject *retval = callmethod(tstate, callable, format, va);
    va_end(va);

    Py_DECREF(callable);
    return retval;
}

static PyObject*
callmethod(PyThreadState *tstate, PyObject* callable, const char *format, va_list va)
{
    assert(callable != NULL);
    if (!PyCallable_Check(callable)) {
        _PyErr_Format(tstate, PyExc_TypeError,
                      "attribute of type '%.200s' is not callable",
                      Py_TYPE(callable)->tp_name);
        return NULL;
    }

    return _PyObject_CallFunctionVa(tstate, callable, format, va);
}
```

vs

```c
PyObject *
PyObject_CallFunction(PyObject *callable, const char *format, ...)
{
    va_list va;
    PyObject *result;
    PyThreadState *tstate = _PyThreadState_GET();

    va_start(va, format);
    result = _PyObject_CallFunctionVa(tstate, callable, format, va);
    va_end(va);

    return result;
}
```
It does at least remember to decrement the callable ref when done, unlike my example.

vfazio, 2024-10-02:
I'm not sure how busy you are or what your plans are for the next version of the python bindings is with 2.2.x-rc2 announced. I can try to submit a patch tomorrow or Friday that builds and passes tests.

brgl, 2024-10-02:
@vfazio python bindings are largely separate on pypi so the v2.2 release doesn't concern them much.

vfazio, 2024-10-03:
rough draft

```diff
From 4c9bff3b19eb7a4443ef227dbf4d3ef82fa42e03 Mon Sep 17 00:00:00 2001
From: Vincent Fazio <vfazio@xes-inc.com>
Date: Thu, 3 Oct 2024 15:21:08 -0500
Subject: [PATCH] bindings: python: explicitly import gpiod classes within
 external module

Previously, the external module implicitly relied on gpiod classes being
present with `globals()` to construct return values back to gpiod.

However, this required callers to the external module methods to have
imported the attributes to populate `globals()` for this to function
correctly. This contract is very opaque and prone to causing issues if
imports within gpiod modules get reworked.

Now, the external module explicitly imports class attributes from gpiod.

There should be no concern about circular imports as the external module
should be completely imported by the time callers call into it.

---
 bindings/python/gpiod/ext/chip.c     | 35 +++++++++++++++++-----------
 bindings/python/gpiod/ext/common.c   | 14 +++++++----
 bindings/python/gpiod/ext/internal.h |  3 ++-
 bindings/python/gpiod/ext/request.c  | 29 +++++++++++++++++------
 4 files changed, 55 insertions(+), 26 deletions(-)

diff --git a/bindings/python/gpiod/ext/chip.c b/bindings/python/gpiod/ext/chip.c
index e8eaad8..b5c3740 100644
--- a/bindings/python/gpiod/ext/chip.c
+++ b/bindings/python/gpiod/ext/chip.c
@@ -75,31 +75,34 @@ static PyObject *chip_get_info(chip_object *self, PyObject *Py_UNUSED(ignored))
        struct gpiod_chip_info *info;
        PyObject *type, *ret;
 
-       type = Py_gpiod_GetGlobalType("ChipInfo");
+       type = Py_gpiod_GetModuleAttrString("gpiod.chip_info", "ChipInfo");
        if (!type)
                return NULL;
 
        info = gpiod_chip_get_info(self->chip);
-       if (!info)
+       if (!info) {
+               Py_DECREF(type);
                return PyErr_SetFromErrno(PyExc_OSError);
+       }
 
-        ret = PyObject_CallFunction(type, "ssI",
-                                    gpiod_chip_info_get_name(info),
-                                    gpiod_chip_info_get_label(info),
-                                    gpiod_chip_info_get_num_lines(info));
-        gpiod_chip_info_free(info);
-        return ret;
+       ret = PyObject_CallFunction(type, "ssI",
+                                   gpiod_chip_info_get_name(info),
+                                   gpiod_chip_info_get_label(info),
+                                   gpiod_chip_info_get_num_lines(info));
+       gpiod_chip_info_free(info);
+       Py_DECREF(type);
+       return ret;
 }
 
 static PyObject *make_line_info(struct gpiod_line_info *info)
 {
-       PyObject *type;
+       PyObject *type, *ret;
 
-       type = Py_gpiod_GetGlobalType("LineInfo");
[... 181 lines cut ...]
                if (!event) {
                        Py_DECREF(events);
+                       Py_DECREF(type);
                        return NULL;
                }
 
@@ -311,6 +324,7 @@ static PyObject *request_read_edge_events(request_object *self, PyObject *args)
                                gpiod_edge_event_get_line_seqno(event));
                if (!event_obj) {
                        Py_DECREF(events);
+                       Py_DECREF(type);
                        return NULL;
                }
 
@@ -318,6 +332,7 @@ static PyObject *request_read_edge_events(request_object *self, PyObject *args)
                if (ret) {
                        Py_DECREF(event_obj);
                        Py_DECREF(events);
+                       Py_DECREF(type);
                        return NULL;
                }
        }
-- 
2.34.1
```

brgl, 2024-10-04:
Looks awesome, does this fix all the instances you listed [here](https://github.com/brgl/libgpiod/issues/101#issuecomment-2379076012)?

vfazio, 2024-10-04:
https://lore.kernel.org/linux-gpio/20241004171438.3066379-1-vfazio@xes-inc.com/T/#u

vfazio, 2024-10-08:
submitted v2 to address comments https://lore.kernel.org/linux-gpio/20241008175139.1198980-1-vfazio@xes-inc.com/T/#u

closed 2024-10-14 by brgl