brgl/libgpiod #103: Segmentation fault in on_edge_event

honzamzr, 2024-09-27:
We are using libgpiod library on our ARM device with musl. When I tried to use gpiocli(DBUS) I found out that the gpiocli can crash on SIGSEGV.

My steps to reproduce it:
Run `gpiocli request --both-edges --input mod1.pin0` to get request.
Run `gpiocli monitor -r 0` to waiting for events.
When the event occurs(on gpio input), gpiocli will crash in musl's strlen.

The reason of this issue are probably used data types(gulong, gint) in `dbus/client/monitor.c:on_edge_event` and `dbus/manager/daemon.c:gpiodbus_daemon_on_edge_event`.
My gulong is 32bit, gint is 32bit too, but "ittt" means int32, uint64, uint64, uint64.

Here is my patch which can fix that issue.

```diff
--- libgpiod.orig/dbus/client/monitor.c 2024-09-27 08:47:01.707408466 +0000
+++ libgpiod/dbus/client/monitor.c      2024-09-27 08:17:59.597359570 +0000
@@ -16,14 +16,14 @@
                          gpointer user_data G_GNUC_UNUSED)
 {
        const char *name = gpiodbus_line_get_name(line);
-       gulong global_seqno, line_seqno;
+       guint64 global_seqno, line_seqno;
        guint64 timestamp;
-       gint edge;
+       gint32 edge;
 
        g_variant_get(args, "(ittt)", &edge, &timestamp,
                      &global_seqno, &line_seqno);
 
-       g_print("%lu %s ", timestamp, edge ? "rising " : "falling");
+       g_print("%llu %s ", timestamp, edge ? "rising " : "falling");
        if (strlen(name))
                g_print("\"%s\"\n", name);
        else
--- libgpiod.orig/dbus/manager/daemon.c 2024-09-27 08:47:01.710741800 +0000
+++ libgpiod/dbus/manager/daemon.c      2024-09-27 08:17:02.220691339 +0000
@@ -521,11 +521,11 @@
 {
        GpiodbusDaemonRequestData *req_data = user_data;
        GpiodbusDaemonLineData *line_data;
-       gulong line_seqno, global_seqno;
+       guint64 line_seqno, global_seqno;
        GpiodglibEdgeEventType edge;
        guint64 timestamp;
        guint offset;
-       gint val;
+       gint32 val;
 
        edge = gpiodglib_edge_event_get_event_type(event);
        offset = gpiodglib_edge_event_get_line_offset(event);
```

brgl, 2024-09-27:
Thanks for the report. I'm building a `qemuarm` image to confirm it.

brgl, 2024-09-27:
@honzamzr I can confirm this happens on any 32-bit system I tried. Do you want to submit a proper fix to linux-gpio yourself?

closed 2024-10-01 by brgl