brgl/libgpiod #104: Impossible to notify on state change of output pin

craigerl, 2024-10-07:
A key feature of /proc/gpio filesystem was that we could use inotify to see when the state of an output pin changed without polling.

This feature seems unimplemented in libgpiod.   If I try to use gpiomon it changes the pin from an output to an input.

Example

   Program A illuminates a LED1
   Program B needs to be notified when LED1 is illuminated.

The libgpiod architecture makes this impossible.

brgl, 2024-10-07:
If a program on the host drives the line then its direction must be output. For a program to watch irqs triggered by an external entity, its direction must be input. What is your setup exactly?

craigerl, 2024-10-07:
In my case, direwolf (C-based radio sound modem) sets pin12 high to trigger a radio transmitter.    Meanwhile a python program (direwatch) watches the output of pin12 to tell the user the radio is transmitting by drawing a red dot on a screen.

With the now-gone /sys/class/gpio filesystem, you simply hung an inotify on /sys/class/gpio/gpio12/value and call a function every time the state of the output changed.   libgpiod doesn't seeem to allow for this, nor gpiozero.  I'm about to backport /sys/class/gpio for my application, so I'm looking for any hints.

brgl, 2024-10-07:
So one process waits for another to set output value of a pin and does it by watching the `value` sysfs attribute? I'd say it's abusing a bad API and the fact that the value even emits an inotify event is purely accidental really. No, libgpiod will not allow you to do this because edge events can only be observed for input pins in character device kernel uAPI which libgpiod simply wraps.

I was about to recommend the new D-Bus GPIO manager but even it won't do what you're expecting.

craigerl, 2024-10-07:
Okay, great, thanks for the confirmation.  So, to be sure, monitoring an output pin is impossible, unless you reinstate the old /sys API and abuse it.  Unfortunately my requirement remains unchanged.   I still have the last kernel that supports the /sys gpio, I'll stick with that until there is a way to simply monitor the state of an output pin in the new/robust libgpiod API.

brgl, 2024-10-07:
To be clear: I have no intention of ever implementing anything like this in the character device API because IMO, it makes no sense.

craigerl, 2024-10-07:
Fully understood, it will remain impossible to monitor the state of an output pin with libgpiod.   I think I mostly wanted to confirm what I already knew, saves me some time, thanks for the followup.

warthog618, 2024-10-09:
Using GPIOs to share state between components is not good design and so not something want to encourage.  The simplest way to reproduce the old sysfs behaviour would be to have direwolf write its "transmit enabled" state to a file that direwatch can watch.

brgl, 2024-10-09:
Or any kind of IPC would be better here to be honest.

craigerl, 2024-10-09:
Yah, I switched to IPC.   I understand the deficiencies of sysfs, however if we force the community switch to something else, we at least have to give them what they had before.  Monitoring the status of an output pin seems like basic functionality to me, and it's gone.   At least I didn't have to teach thousands of users how to solder and connect the output of the pin to an input just so we could observe it.... this time.

warthog618, 2024-10-09:
We do not "force the community" to do anything.  We follow a development and release process that is open to the community, and anyone is welcome to participate.

The feature you were using is accidental - only there as the sysfs interface happened to involve writing to the filesystem.  I was not involved in the initial move away from sysfs, but I was not aware of it being used in that way, and I expect that is also true for the devs that were.  It is difficult to design for features or requirements of which you are unaware.

Such a feature never would've occurred to me, as the need for a feature to tell you the value of an output that you set yourself makes no sense - I know the value that I set it to, and if I did want to share that with another component then I would do it in a way that was not inherently racy.

I will assume the soldering comment was in jest.... this time.
Try teaching users how to design robust concurrent systems.  Fun for the whole community there.

closed 2024-10-11 by brgl