brgl/libgpiod #105: set_edge_detection only works with NONE

Hommus, 2024-10-10:
Running the following cpp code works but if I update `edge::NONE` to `edge::RISING`, `edge::FALLING` or `edge::BOTH`, the program will fail:
```cpp
auto request = gpiod::chip(chip_path_)
    .prepare_request()
    .set_consumer("watch-line-value")
    .add_line_settings(button_offset_, gpiod::line_settings()
        .set_direction(gpiod::line::direction::INPUT)
        .set_edge_detection(gpiod::line::edge::NONE))
    .do_request();
```

Error:
```bash
error requesting GPIO lines: No such device or address
```

Any help would be appreciated.

brgl, 2024-10-10:
This is pretty well tested so should work. Can you run the program through `strace` and post the output?

Hommus, 2024-10-10:
Running the cpp example [watch_line_rising.cpp](https://github.com/brgl/libgpiod/blob/master/bindings/cxx/examples/watch_line_rising.cpp) inside a docker container:

```bash
root@jetson:/ros_ws/src/view# g++ -std=c++17 -o watch_line_rising watch_line_rising.cpp -lgpiodcxx
root@jetson:/ros_ws/src/view# strace ./watch_line_rising
execve("./watch_line_rising", ["./watch_line_rising"], 0xffffef6868b0 /* 24 vars */) = 0
brk(NULL)                               = 0xaaab18039000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff8e9b9000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/atomics", 0xfffff2187a10, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
write(2, "error requesting GPIO lines: No "..., 54error requesting GPIO lines: No such device or address) = 54
write(2, "\n", 1
)                       = 1
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [], 8) = 0
getpid()                                = 160
gettid()                                = 160
tgkill(160, 160, SIGABRT)               = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=160, si_uid=0} ---
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)
```

brgl, 2024-10-10:
> `ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xb4, 0x7, 0x250), 0xfffff2188938) = -1 ENXIO (No such device or address)`

This is an old version of strace that doesn't decode GPIO v2 ioctl()s. Could you build a current strace release from sources and try again?

Hommus, 2024-10-10:
I built strace with  `./configure --disable-mpers` to fix an error, let me know if that will be an issue.

```bash
root@jetson:/ros_ws/src/view# /usr/local/bin/strace ./watch_line_rising
execve("./watch_line_rising", ["./watch_line_rising"], 0xffffc02cc540 /* 24 vars */) = 0
brk(NULL)                               = 0xaaaad9d68000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffff9e0d4000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/aarch64", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/tls", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/aarch64", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/opt/yaml_cpp_vendor/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/aarch64", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/tls", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/aarch64", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/aarch64-linux-gnu", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/atomics", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/opt/ros/foxy/lib/tls/aarch64", 0xfffffbb07790, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/ros/foxy/lib/tls/atomics/libgpiodcxx.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
write(2, "error requesting GPIO lines: No "..., 54error requesting GPIO lines: No such device or address) = 54
write(2, "\n", 1
)                       = 1
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [], 8) = 0
getpid()                                = 114643
gettid()                                = 114643
tgkill(114643, 114643, SIGABRT)         = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=114643, si_uid=0} ---
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)
```

brgl, 2024-10-10:
I see it's most likely an i2c expander. What is the model and are you using an upstream driver? If not, can you point me to the relevant GPIO driver? To me it looks like the driver doesn't support edge detection.

brgl, 2024-10-10:
Do you have the device tree sources for your kernel? You can get that information from there. If not, you can find that information out from the sysfs. Without that, it would be hard to help you more than that, the strace output looks correct.

An alternative is busy polling of course. Just read the value in a loop.

Hommus, 2024-10-10:
```bash
$ cat /sys/class/gpio/gpiochip284/device/modalias
of:Nedgekit-pca6416-mainT(null)Cti,tca6416
```
I'm going to do some more digging tomorrow. Thanks for your help.

brgl, 2024-10-10:
This is pca953x driver, edge detection should work. But the `-ENXIO` errno leads me to believe you're hitting this:

```
1714 static int gpiochip_irq_map(struct irq_domain *d, unsigned int irq,
1715                             irq_hw_number_t hwirq)
1716 {               
1717         struct gpio_chip *gc = d->host_data;
1718         int ret = 0;
1719 
1720         if (!gpiochip_irqchip_irq_valid(gc, hwirq)) 
1721                 return -ENXIO;
```

Do you have `CONFIG_GPIO_PCA953X_IRQ` enabled in your Kconfig?

Hommus, 2024-10-10:
`CONFIG_GPIO_PCA953X_IRQ` is enabled.

brgl, 2024-10-10:
In any case, this is a kernel and not a libgpiod issue. You should probably try to track what's happening in the driver if you can recompile the kernel.

closed 2024-10-11 by brgl