brgl/libgpiod #107: bindings: python: migrate metadata out of setup.py to pyproject.toml

vfazio, 2024-11-08:
PyPA recommends migrating to pyproject.toml where possible: https://packaging.python.org/en/latest/guides/modernize-setup-py-project/

We have one, but it's sparse. We do not explicitly specify the build-backend:

```toml
[build-system]
requires = ["setuptools>=59.0.0", "packaging"] # we need 59.0.0 for https://github.com/pypa/setuptools/commit/974dbb03769a500c5077d37291817f30359a0d7b
build-backend = "setuptools.build_meta"
```

I don't think we need "wheel" as a dependency to generate wheels (testing shows we don't at least).

As part of the migration, we can specify proper Trove classifiers. Newer versions of setuptools support dynamic fields in the toml file.

More configuration will slowly start landing in the toml file as a result of #97, so we may as well make it comprehensive.

brgl, 2024-11-19:
@vfazio  was this achieved by your big series I applied today?

vfazio, 2024-11-19:
> @vfazio was this achieved by your big series I applied today?

It was not. I was trying to keep the typing related stuff in that series and this is outside of that scope.

I think some of this is pretty simple, but i need to make sure that after the migration to pyproject.toml that the wheels and what not still look kosher.

brgl, 2024-11-19:
Got it.

vfazio, 2024-11-20:
@brgl I can certainly work on this if you want me to. Is this something you wanted to close out as part of 2.2.3 or 2.3.x? I'm not 100% sure of the binding versioning scheme.

My availability to work on this stuff is a little limited with the holidays coming up in the US and other travel plans of mine. (also why i haven't responded to the other issue, i'm way behind on other work, I haven't forgotten about it however).

brgl, 2024-11-20:
The next version is going to be v2.3.0. I think the recent rework deserves a minor release.

vfazio, 2024-11-20:
> The next version is going to be v2.3.0. I think the recent rework deserves a minor release.

Are there other fixups or changes you want to include before tagging 2.3.0? I would like to help get those resolved for you if I can.

vfazio, 2024-11-21:
I've got this mostly working. I need to compare the outputs from the migration to previous sdist and wheels to make sure the contents look correct. I'll let you know how that goes.

vfazio, 2024-11-21:
I'm going to post a diff here and then explain some things and then post some evidence just to have a discussion before actually posting the patch to the ML especially since my dev environment is different than yours.

```diff
diff --git a/bindings/python/pyproject.toml b/bindings/python/pyproject.toml
index d6f5f9b..a87055c 100644
--- a/bindings/python/pyproject.toml
+++ b/bindings/python/pyproject.toml
@@ -2,7 +2,48 @@
 # SPDX-FileCopyrightText: 2023 Phil Howard <phil@gadgetoid.com>
 
 [build-system]
-requires = ["setuptools", "wheel", "packaging"]
+requires = ["setuptools>=69.0.0"]
+build-backend = "setuptools.build_meta"
+
+[project]
+name = "gpiod"
+dynamic = ["version"]
+description = "Python bindings for libgpiod"
+readme = "README.md"
+license = {text = "LGPL-2.1"}
+requires-python = ">=3.9.0"
+authors = [
+    { name = "Bartosz Golaszewski", email = "brgl@bgdev.pl" },
+]
+classifiers = [
+  "Development Status :: 5 - Production/Stable",
+  "License :: OSI Approved :: GNU Lesser General Public License v2 (LGPLv2)",
+  "Operating System :: POSIX :: Linux",
+  "Typing :: Typed",
+  "Programming Language :: Python",
+  "Programming Language :: Python :: 3",
+  "Programming Language :: Python :: 3 :: Only",
+  "Programming Language :: Python :: 3.9",
+  "Programming Language :: Python :: 3.10",
+  "Programming Language :: Python :: 3.11",
+  "Programming Language :: Python :: 3.12",
+  "Programming Language :: Python :: 3.13",
+]
+
+[tool.setuptools]
+platforms = ["linux"]
+include-package-data = false
+
+[tool.setuptools.dynamic]
+version = {attr = "gpiod.version.__version__"}
+
+[tool.setuptools.packages.find]
+include = ["gpiod"]
+namespaces = false
+
+[project.urls]
+Homepage = "https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git"
+Issues = "https://github.com/brgl/libgpiod/issues/"
 
 [tool.mypy]
 python_version = "3.9"
diff --git a/bindings/python/setup.py b/bindings/python/setup.py
index 0d518af..1af4709 100644
--- a/bindings/python/setup.py
+++ b/bindings/python/setup.py
@@ -4,7 +4,7 @@
 from os import getenv, path, unlink
 from shutil import copytree, rmtree
 
-from setuptools import Extension, find_packages, setup
+from setuptools import Extension, setup
 from setuptools.command.build_ext import build_ext as orig_build_ext
 from setuptools.command.sdist import log
 from setuptools.command.sdist import sdist as orig_sdist
[... 185 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

(venv) root@rpi-87fa00:/var/tmp/tmp.0emZxixLEE# pkill gpioset

```

The repo's top level default make target continues to work build without error. The python tests continue to build and work as well.

Rationale behind the changes:

* dropping wheel and packaging
  * Setuptools vendors these packages. The pypi artifact script doesn't run into any issues when building sdist or wheels via `python3 -m build`
* Minimal setuptools is required to generate the distributions to ensure the features leveraged in the pyproject.toml work, this includes bundling `py.typed`, `*.pyi`, support for `attr`, etc.
* explicitly include "gpiod" instead of excluding "tests*"
* Excluding namespaces
  * While excluding namespaces isn't strictly required, there are no namespace packages here so better to be explicit
* Excluding package data
  * I was having issues with setuptools thinking gpiod/ext/ should be packaged and complaining when trying to exclude it. If i didn't do this, setuptools would include all of the files in this directory in the wheel, which included the *.c and *.h files
  * https://setuptools.pypa.io/en/latest/userguide/datafiles.html#include-package-data
* Support for the C extensions is left in `setup.py`.
  * Support in pyproject is experimental and I didn't want to risk it getting overhauled and breaking builds
  * see footnote: https://setuptools.pypa.io/en/stable/userguide/ext_modules.html#pyproject-toml
* I added trove classifiers that seemed to make sense, though maybe there are other classifiers you want or maybe you don't want this marked as Production/Stable
  * https://pypi.org/classifiers/

vfazio, 2024-11-21:
it looks like i did LGPL-2.1 and should have done 2.1+ based on the SPDX headers on the files. I will fix that when i actually make the patch. I was going off of the license value formerly in `setup.py`.

vfazio, 2024-11-22:
Sorry to ping you, @brgl does this seem reasonable?

brgl, 2024-11-22:
Yes, awesome. Are tests still packaged in sdist?

vfazio, 2024-11-22:
For good measure, I went ahead and also validated that wheels worked on Alpine too:

```
(venv) localhost:/tmp/tmp.OijHbF# cat /etc/os-release
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.20.3
PRETTY_NAME="Alpine Linux v3.20"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://gitlab.alpinelinux.org/alpine/aports/-/issues"

(venv) localhost:/tmp/tmp.OijHbF# pip install /root/gpiod-2.2.0-cp312-cp312-musl
linux_1_2_aarch64.whl

(venv) localhost:/tmp/tmp.OijHbF# head -n25 venv/lib/python3.12/site-packages/gpiod-2.2.0.dist-info/METADATA
Metadata-Version: 2.1
Name: gpiod
Version: 2.2.0
Summary: Python bindings for libgpiod
Author-email: Bartosz Golaszewski <brgl@bgdev.pl>
License: LGPL-2.1-or-later
Project-URL: Homepage, https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git
Project-URL: Issues, https://github.com/brgl/libgpiod/issues/
Platform: linux
Classifier: Development Status :: 5 - Production/Stable
Classifier: License :: OSI Approved :: GNU Lesser General Public License v2 or later (LGPLv2+)
Classifier: Operating System :: POSIX :: Linux
Classifier: Typing :: Typed
Classifier: Programming Language :: Python
Classifier: Programming Language :: Python :: 3
Classifier: Programming Language :: Python :: 3 :: Only
Classifier: Programming Language :: Python :: 3.9
Classifier: Programming Language :: Python :: 3.10
Classifier: Programming Language :: Python :: 3.11
Classifier: Programming Language :: Python :: 3.12
Classifier: Programming Language :: Python :: 3.13
Requires-Python: >=3.9.0
Description-Content-Type: text/markdown

<!-- SPDX-License-Identifier: CC-BY-SA-4.0 -->

(venv) localhost:/tmp/tmp.OijHbF# python3 example.py
line: 40  type: Rising   event #1
line: 40  type: Rising   event #2
line: 40  type: Rising   event #3
line: 40  type: Rising   event #4
^CTraceback (most recent call last):
  File "/tmp/tmp.OijHbF/example.py", line 24, in <module>
    watch_line_rising("/dev/gpiochip0", 40)
  File "/tmp/tmp.OijHbF/example.py", line 14, in watch_line_rising
    for event in request.read_edge_events():
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/tmp.OijHbF/venv/lib/python3.12/site-packages/gpiod/line_request.py", line 225, in read_edge_events
    return cast(_ext.Request, self._req).read_edge_events(max_events)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

```

I'll post a patch in a bit

vfazio, 2024-11-22:
https://lore.kernel.org/linux-gpio/20241122171027.762311-1-vfazio@xes-inc.com/T/#u

brgl, 2024-11-27:
I applied the patch, awesome work, thank you. Everything I could think of still works.

closed 2024-11-27 by brgl