brgl/libgpiod #110: bindings: python: fixup possibly wrong SPDX licenses

vfazio, 2024-11-21:
There are a few discrepancies in the SPDX headers on files in the bindings:

* gpiod/_internal.py has GPL instead of LGPL like the rest of the files
* tests/tests_line_settings.py has LGPL instead of GPL like the rest of the tests
* tests/tests_line.py has LGPL instead of GPL like the rest of the tests

There may be others or I may be wrong about these being incorrect.

brgl, 2024-11-22:
Thanks. Some of the files were authored by other people so for instance `bindings/python/pyproject.toml` is GPL but should really be CC I think. However I don't think I can just change the license like that. The three files you mentioned should be fine though.

vfazio, 2024-11-22:
the external test module sources and Makefiles look to also use LGPL, and some others are a bit all over the place

```
tests/tests_line_info.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/tests_line_request.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/procname/Makefile:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/procname/ext.c:// SPDX-License-Identifier: LGPL-2.1-or-later
tests/procname/Makefile.am:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/procname/_ext.pyi:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/procname/__init__.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/procname/Makefile.in:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/Makefile:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/tests_chip_info.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/tests_line.py:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/tests_chip.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/Makefile.am:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/tests_info_event.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/tests_module.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/__init__.py:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/Makefile.in:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/tests_line_settings.py:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/__main__.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/gpiosim/Makefile:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/gpiosim/chip.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/gpiosim/ext.c:// SPDX-License-Identifier: LGPL-2.1-or-later
tests/gpiosim/Makefile.am:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/gpiosim/_ext.pyi:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/gpiosim/__init__.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/gpiosim/Makefile.in:# SPDX-License-Identifier: LGPL-2.1-or-later
tests/helpers.py:# SPDX-License-Identifier: GPL-2.0-or-later
tests/tests_edge_event.py:# SPDX-License-Identifier: GPL-2.0-or-later
```

I have to admit, i'm not savvy with licenses so do not know how these should be licensed. I could certainly send a patch, but would have to defer to you to what licenses to use, so maybe it's just easier that you fix them since they have your attribution?

brgl, 2024-11-22:
My logic was that modules that can be imported (and tests are implemented this way) should have LGPL?

vfazio, 2024-11-22:
haha, no idea. i'm not that smart. it sounds logical however

vfazio, 2024-12-03:
I saw you submitted a patch for this and I think I ACK'd. Is this the last piece for the subsequent release? I think we've managed to cram a lot of changes in.

brgl, 2024-12-03:
> I saw you submitted a patch for this and I think I ACK'd. Is this the last piece for the subsequent release? I think we've managed to cram a lot of changes in.
> 

Possibly. Is there anything else you could think of?

> Edit: yea i just screwed up the Subject line... of course https://lore.kernel.org/linux-gpio/PH1P110MB16034D59F26D1C4A54BE81039F28A@PH1P110MB1603.NAMP110.PROD.OUTLOOK.COM/

No worries, I picked it up manually. But your corporate mailer could use some better configuration.

vfazio, 2024-12-03:
> > I saw you submitted a patch for this and I think I ACK'd. Is this the last piece for the subsequent release? I think we've managed to cram a lot of changes in.
> 
> Possibly. Is there anything else you could think of?

Nothing jumps out at me right now

> 
> > Edit: yea i just screwed up the Subject line... of course https://lore.kernel.org/linux-gpio/PH1P110MB16034D59F26D1C4A54BE81039F28A@PH1P110MB1603.NAMP110.PROD.OUTLOOK.COM/
> 
> No worries, I picked it up manually. But your corporate mailer could use some better configuration.

Unfortunately, due to the security clearance and internal policies, there's not much I can do about this. They do not prioritize email/list based development workflows. This is why I often CC my gmail address and do my responses from there.