brgl/libgpiod #112: WARNING: 'aclocal-1.16' is missing on your system

o-alquimista, 2024-12-05:
On Alpine 3.21, when upgrading to libgpiod v2.2, any `make` command I issue on my git clone results in an error.

```
$:~/libgpiod$ make clean
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh '/home/dsilva/libgpiod/autostuff/missing' aclocal-1.16 -I m4
/home/dsilva/libgpiod/autostuff/missing: line 81: aclocal-1.16: not found
WARNING: 'aclocal-1.16' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <https://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/autoconf>
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
make: *** [Makefile:431: aclocal.m4] Error 127
```

It mentions `aclocal-1.16`, which it says it's included in GNU Automake 1.16; but I only have access to Automake 1.17.

brgl, 2024-12-05:
There's no reference to any specific aclocal version in the git repo. Did you try a complete `git clean -fdx`?

o-alquimista, 2024-12-05:
That helped. But now I think I forgot to run `./autogen.sh` before trying `make`. My fault. Thanks.