brgl/libgpiod #113: Pin names for Radxa Zero (Armbian)

way5, 2024-12-05:
Good day everyone!
the results of running `gpioinfo` command must look like this:

```bash
gpioinfo
gpiochip0 - 85 lines:
        line   0:     "PIN_27"       unused   input  active-high
        line   1:     "PIN_28"       unused   input  active-high
        line   2:      "PIN_7"       unused   input  active-high
        line   3:     "PIN_11"       unused   input  active-high
        line   4:     "PIN_13"       unused   input  active-high
        line   5:     "PIN_15"       unused   input  active-high
        line   6:     "PIN_18"       unused   input  active-high
        line   7:     "PIN_40"       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       unused   input  active-high
...
```

including pin names. However running this command on latest version of Armbian linux it doesn't show pin names. I thought it must be defined in Kernel and it is [defined correctly](https://github.com/torvalds/linux/blob/master/include/dt-bindings/gpio/meson-g12a-gpio.h).
Some say it has to be defined in Device Trees, but I cannot really find where it must be placed. 

Where to look to set it up?

Thank you.

brgl, 2024-12-05:
There should be a `gpio-line-names` property of the GPIO controller node in the device tree you're using.

way5, 2024-12-05:
> There should be a `gpio-line-names` property of the GPIO controller node in the device tree you're using.

I'm going to check that again.

> This isn't really a libgpiod problem though.

I am aware of that. This is more Q&A post since I'd like to "make it right".

way5, 2024-12-14:
GPIO names are added to the latest Kernel.

closed 2024-12-14 by way5