brgl/libgpiod #115: Question about gpionotify

Lebensgefahr, 2024-12-06:
OS: Armbian-unofficial 24.2.0-trunk Jammy
Kernel: 6.1.74-current-rockchip64
SOC: rk3328
commit hash: 6d9133a259e64da5e03c7e7784f0f27de7b3e59f

I have a system with gpiochip2 line 16 state 0/1 which I can use to detect power loss or powering off. When switch is turned off state of this pin changing from 0 to 1 and back to 0 if power was turned on. I used it to shutdown system softly by systemd service
There is an old version of gpiod in repository without correct exit codes.
`/bin/bash -c 'gpiomon -n 1 -r -c 2 16 && poweroff'`

I've build and install gpiod from this repo yesterday. 
I can check this by gpiomon like this.
```
mks@mkspi:~/gpiod$ sudo /home/mks/gpiod/bin/gpiomon -c 2 16
133.441027791	falling	gpiochip2 16
133.441195793	falling	gpiochip2 16
133.441281252	falling	gpiochip2 16
133.441286210	falling	gpiochip2 16
133.441314794	falling	gpiochip2 16
133.441355919	rising	gpiochip2 16
133.442866181	falling	gpiochip2 16
133.442901765	rising	gpiochip2 16
134.659500621	falling	gpiochip2 16
^C

```
But when I am using gpionotify like this:
```
mks@mkspi:~/gpiod$ sudo /home/mks/gpiod/bin/gpionotify -c 2 16
^C
```
I see nothing on stdout. Is it ok?

brgl, 2024-12-06:
`gpionotify` displays changes in line states, not edge events. It will tell you when a line gets requested, released or reconfigured.

Lebensgefahr, 2024-12-06:
> `gpionotify` displays changes in line states, not edge events. It will tell you when a line gets requested, released or reconfigured.

I am just trying to find a better way to monitoring the state of the line (with minimal latency as i have a limited time to execute something).
You wrote that gpionotify displays changes in line states. Is changing of the line state from 0 to 1 is not the same?
Miłego dnia

brgl, 2024-12-06:
> > `gpionotify` displays changes in line states, not edge events. It will tell you when a line gets requested, released or reconfigured.
> 
> I am just trying to find a better way to monitoring a state of the line (with minimal latency as i have a limited time to execute something and minimal system loading instead of doing something in a loop like gpioget 2 16|grep 1). You wrote that gpionotify displays changes in line states. Is changing of the line state from 0 to 1 is not the same? I am thinking in 

No, `gpionotify` shows changes in kernel config.

inotify logic way. If file was created than notify. What is the event which cause generating something? Is modern versions of gpiomon exit codes can be recognized as gpio state? For example gpioget 2 16 will return 0 only and only if gpiochip2 line16 is equal to 0 so i can use this value. This question is not described in man page and doc. In old versions gpiomon process can be killed and exit code is 0 the same as it finished with -n 1 -r parameters.
> 
> Miłego dnia

I'm not sure what exactly the question is? `gpiomon` still exists and shows edge events. I don't think we ever considered the return value of gpiomon to mean anything else than errors when non-0. You should rely on the output instead.

closed 2024-12-06 by Lebensgefahr