brgl/libgpiod #117: bindings: python: investigate free threaded builds

vfazio, 2024-12-07:
> 
> Ha! Is there any writeup on this? I would definitely want to make sure the bindings to work fine without the GIL.

https://peps.python.org/pep-0703/

https://docs.python.org/3/whatsnew/3.13.html#free-threaded-cpython

https://docs.python.org/3/c-api/module.html#c.Py_mod_gil

https://py-free-threading.github.io/porting/

_Originally posted by @vfazio in https://github.com/brgl/libgpiod/issues/93#issuecomment-2405210680_

vfazio, 2024-12-08:
I don't know if we should be defensive and throw a compile error if a gil-less build is attempted until we've had a chance to test and validate everything works as expected.

brgl, 2024-12-09:
Does any distro ship GIL-less python 3.13 yet? I'm pretty sure it will make its way into yocto soon, this is when I'll most likely start playing with it. Until then, I don't think we should really do anything about it.

vfazio, 2024-12-10:
@brgl looks like Fedora ships 3.13 as part of f41+: https://docs.fedoraproject.org/en-US/fedora/latest/release-notes/developers/

and it looks like there are build changes and rpms for freethreading, at least for f42, but they're probably also available for f41

https://src.fedoraproject.org/rpms/python3.13/c/84d78b385a57306af12b04922f60d516e6e8be49

https://koji.fedoraproject.org/koji/buildinfo?buildID=2599772

https://koji.fedoraproject.org/koji/rpminfo?rpmID=40917017

vfazio, 2026-03-22:
@brgl if you're really bored, do you want to try testing out https://github.com/vfazio/libgpiod/tree/vfazio-upstream-proposed ?

I've added beta free-threaded support.

I've run the tests under normal python and under a free-threaded build and have not seen any issues when appropriate precautions are taken.