brgl/libgpiod #120: unable to assert gpio when using gpiocli and gpio-manager.

davisRoman, 2024-12-19:
Hello, I'm running a yocto build with libgpiod v2.2, I'm interested in enabling the gpio-manager and gpiocli in order to test out gpio persistence.

Once I boot up, I see that the gpio-manager has started up properly:
[image]

Then I attempt to request my desired gpio which works as expected: 
[image]

However when I attempt to assert the gpio, I get the following:
[image]

and journalctl also shows some additional errors:
[image]

I found this policy file that was included in libgpiod
[image]

It says that root users can do anything and since I'm indeed running as root I'm not quite sure why I'm getting an error.

Any suggestions?

brgl, 2024-12-19:
Hey! Thanks for trying it out!

Just a couple github hints: please don't post images of code, copy the actual console output and put it in code blocks. I edited the second comment for you and put the `gpio-manager` output in codeblocks.

You seem to have requested the line "as-is". Is it possible its direction is still `input`? If so, request it explicitly as output or change its direction with `gpiocli reconfigure` and try again.

davisRoman, 2024-12-20:
Hi Bartosz,

Your suggestion did the trick! 

I just had to add `--output` when invoking `gpiocli request`.

`gpiocli` and `gpio-manager` now work as advertised! 

Thank you for introducing this much needed feature.

closed 2024-12-20 by davisRoman